ttlappalainen/NMEA2000 #122: NMEA 0813 to NMEA 2000 Multiplexer

ryall, 2018-08-26:
I'm looking to build an NMEA 0813 to NMEA 2000 multiplexer with this library. 

From what I understand, I will need one MAX3232 Transceiver for each incoming NMEA 0813 connection, then to read the data from each line and transcode it to the NMEA 2000 bus.

Would this work? Also, for the output, would a single "Multiplexer" device be sufficient on the N2K bus or would it be better to use one device per NMEA 0813 connection?

sarfata, 2018-08-26:
Hi Nat,

You need one form of level conversion for each incoming nmea0183. The max232 family are bi-directional converters but you only need one direction so you can look into other chips like the [DS26LV32](http://www.ti.com/lit/ds/symlink/ds26lv32at.pdf) which is a quad RS422 receiver.

As for software, one device is enough in my experience. The library allows you to appear as multiple devices so you could also do that. Some commercial devices do this, for example my B&G Vulcan7 appears as "Vulcan 7 GPS", "Vulcan 7 Autopilot Control", etc.

If you are interested, my project KBox is open-source and already includes conversion code to receive NMEA, convert it to SignalK internally, and then into another format like NMEA2000. This is all in the 'signalk' folder in the project and you should be able to re-use this easily. I would be happy to help if you have any problem: https://github.com/sarfata/kbox-firmware/tree/master/src/common/signalk

ryall, 2018-08-26:
Thanks a lot @sarfata, especially the receiver chip! I'll take a look at KBox now :)