ttlappalainen/NMEA2000 #124: Custom PGN Question, and acceptable update interval for them

virtuvas, 2018-09-11:
Timo,

working on a new project and in need of communicating from a teensy 3.5 that runs the custom electronics for the fin stabilisers on my boat to a 7inch display (with a Due running it) Tried I2C and after 20+h I gave up... Distance between teensy and Due is around 5m.

So plan now is to use custom PGNs (teensy is already on N2K as it needs speed, rudder and other info to run the fins). Due will be updated to N2K, v.easy:
One PGN to send approx 12double from teensy to the Due.
One PGN for the Due to report back user input to the teensy.

Now for the questions:

A.
Should I just copy your N2kMessages.cpp and .h with another name (like vasN2kMessages ), "built" my two sentences there picking fields/code from yours and include this file to my project?
Would that be the preferred solution? Don't simply want to edit and add my two sentences at the bottom of your files as in the next update I'll have to edit them in again.

B. 
Regarding PGNs, any 6digit name I like can do?  Like can it be PGN150001 and PGN150002?
I haven't found a range of acceptable custom PGNs somewhere, any pointers appreciated.

C.
what is the closest legal interval for updating info on these PGNs? Especially the one from the teensy to the Due has to be updated at least twice a second, preferably 3 times a second to refresh the screen and update fin position, boat pitch and roll angles, etc. Bus is fairly extensive with lots of garmin kit (2plotters, autopilot, 4X4inch displays, VHF, AIS, fishfinder) as well airmar DST800, and two custom black boxes with various other info floating around)

Again many thanks for your excellent work!

V.

ttlappalainen, 2018-09-11:
A. For pure custom PGN, keep the module just on your own lib. Do not add it at all to the NMEA2000 library. Just make similar handling functions for reading and writing, then it is reasonably easy to use.

B. There are proprietary message ranges for manufacturers own messages, which you should use. Messages has to allways start with:
Field 1: "Manufacturer Code", 11 bits (use some unused mfg number for your devices.)
Field 2: "Reserved", 2 bits
Field 3: "Industry Code", 3 bits

So it takes totally 16 bits = 2 bytes and there is 6 bytes left in single frame message for your data. This should be enought for your response. For longer data you can use Fast packet proprietary messages.

Message ranges are:
- proprietary PDU1 (addressed) single-frame range 0EF00 to 0xEFFF (61184 - 61439) messages. Addressed means that you send it to specific node on the bus. This you can easily use for responding, since you know the sender. For sender it is bit more complicate since your device address may change due to address claiming. There is N2kDeviceList module for handling devices on bus and find them by "NAME" (= 64 bit value set by SetDeviceInformation ).

- proprietary PDU2 (non addressed) single-frame range 0xFF00 to 0xFFFF (65280 - 65535). Non addressed means that destination wil be 255 (=broadcast) so any cabable device can handle it.

- proprietary PDU1 (addressed) fast-packet range 0x1EF00 to 0x1EFFF (126720 - 126975)

- proprietary PDU2 (non addressed) fast packet range 0x1FF00 to 0x1FFFF (130816 - 131071)

So as conclusion you can not use PGN150001 or PGN150002. I did not calculate, but they may even be impossible.

C. You can use e.g. 100 ms period = 10 times/sec. Even 50 ms period has been used. If you use Fast Packet, use as slow as possible - e.g. 250 ms period.

See pgn descriptions on https://github.com/canboat/canboat/blob/master/analyzer/pgns.json

virtuvas, 2018-09-21:
Timo, 

thanks for your valuable insight.
Decided to use a proprietary non addressed fast packet PGN, chose PGN131021-025 for testing.
Made a new lib  both .cpp and .h files by copying the N2kMessages.cpp and .h and stripping them out of everything except Engine parameters dynamic which I modified, changed values to what I want and saved with my own PGN and custom name.

Amazingly code on the teensy 3.5 loads, and looks like it's working.

Similarly DataDisplay2 on a Due (which is running the 7inch screen of the system) is also running and reports that my new custom sentence is being pumped by the teensy.

Problem is now debuging the new PGN as the SetN2kPGN131021 works 
but the ParseN2kPGN131021 throws an error
undefined reference to 'ParseN2kPGN131021 in my new .h lib...

cheers

V.

ttlappalainen, 2018-09-21:
I do not prefer to make any modifications to the lib in your case. Keep lib as it is and add your proprietary messages in own .h/.cpp under your own code. In this way you can update the NMEA2000 library at any time.

If you get undefined reference to 'ParseN2kPGN131021, it means that it i not declared at any module. Often this is caused by spelling error, letter case mistake etc.

virtuvas, 2018-09-23:
Timo,

yes, all is placed in a new lib folder under lib with matching names for .h and .cpp
Found the problem one of the 8 parameters of the Parse function was int8_t but defined double in the .cpp file!

So, now I got the teensy to pump PGN127245 (rudder) and my custom PGN131021 data every half a second.
on the due I get both PGNs listed (I modified datadisplay2 by adding the PGN131021 support)
BUT:

rudder data come on the due fine, fixed, constant as sent by teensy

custom data are all wrong:
random values on the first 5 variables (2 int, 3 float) 
not available on the next 3 variables (floats)
and 
-1 on the last 2 variables (int)

I changed the values sent by the teensy, same thing, no change.
Any idea what could I be doing wrong?
Too much data pumped on this PGN? Doubt it as it's less than the PGN127489 I used as a template to do my custom PGN.

cheers

V.

ttlappalainen, 2018-09-24:
One problem is that the library does not know anything about your PGN131021 and since it is fast packet, it should know that it constructs from several frames. So you have to tell about special PGNs for library on the setup before open:
const unsigned long ExtraFastPacketMessages[] PROGMEM={131021,0};
...
vois setup() {
...
  NMEA2000.ExtendFastPacketMessages(ExtraFastPacketMessages);
...
}

virtuvas, 2018-09-25:
thanks again Timo,

after spending some time on the ExtraFastPacketMessages, gave up and organised it differently:

teensy
sends 2 proprietary single frames (four doubles on each frame, enough for my data) with different refresh speeds (one does the v.crucial data with refresh of 100ms, second sends data that can be updated every second) to the Due+TFT

Due+TFT
sends 1 proprietary single frame to the teensy with screen user input (4 int)

Now all works BUT, screen touch input is v.slow if Due+TFT are on the NMEA bus, speed is excellent when unplugged.
Could it be that the NMEA message handler delays the process of giving back control to the loop() ?
Would it make sense to try and use addressed messages?
Remember bus has around 20devices... but both Due and teensy have descreet SetDeviceInformation.

Is there any example of using addressed messages?

again many thanks for you work!

V.

ttlappalainen, 2018-09-26:
What was wrong with ExtraFastPacketMessages? Note that you have to use it on receiving side.

Either I do not understand how you can send four double on single frame? Note that proprietary message has to start with
Field 1: "Manufacturer Code", 11 bits (use some unused mfg number for your devices.)
Field 2: "Reserved", 2 bits
Field 3: "Industry Code", 3 bits

So that is taking 2 bytes. Each float is 4 byte, so 4 floats will take 16 byte or if your accuracy and range is small enough, you can use down to 1 byte double, but then you have only 255 different values. So with very low resolution  1 byte double you can handle 4 double with 6 byte and single frame. But with beter resolution 2 byte double you need totally 10 bytes and so 2 frames. With high resolution 8 byte double you need 4*8+2=34 bytes for 4 doubles.

The same is with ints. This should be simple calculation so what I missed here?

I am not sure about DUE speed. I have it running on system but making conversations and forwarding. I skipped DUE at very beginning, since Teensy is much better. Things to check:
- use due_can lib from my git, since it has been tested.
- if you do not need message forwarding to USB, disable it.
- Check your TFT writing speed. If you do TFT writing fron library callback and the TFT writing is slow (as I have seen with some TFT libraries), it naturally slows down the system. In principle it does not make difference in system speed, if you just collect data on library callback and then have other task for updating data to the TFT, but it makes code more logic and avoid possible loops. Also it is easier to test system by simulating incoming bus data.

Addressed message does not speed up system radically. With addressed message you have to use tN2kDeviceList to collect all devices on the bus and use that for getting your sender address. Or you can make simpler simpler method and listen all ISO address claim messages to just get your device.

virtuvas, 2018-12-03:
closing this issue with two comments:

yes, I just need low resolution floats, data go back and forth intact so I'm happy.

Due problem was because I had some slow running loops on my code delaying control to the due. Changed that using interrupt service routines and now it's much quicker. Next time I'll use a teensy 3.6 for driving a smaller display...

thanks once more Timo!

V.

closed 2018-12-03 by virtuvas