ttlappalainen/NMEA2000 #127: ESP32 - what CAN transiver

sailor1958, 2018-11-02:
Hi

Have now worked some days to get my ESP32 and VP230 (SN65HVD230) CANTRANSCEIVERS, to work. But with out any luck.
Are the any Transceivers you can suggest to use, because you have seen it works ?

I have a MCP2515_CAN unit - but can't see have to force ex. TemperaturMonitor example to use MCP2515 and not the build CAN unit in ESP32

Please advice

mrbubble62, 2018-11-02:
I have found SN65HVD230 is easily killed, maybe because I sourced from eBay. I now use MCP2562 from US vendors. You should not need MCP2515, ESP32 native can works fine.

This message has been scanned for viruses and dangerous content by
E.F.A. Project<http://www.efa-project.org>, and is believed to be clean.
Click here to report this message as spam.<http://efa.mrbubble.net/cgi-bin/learn-msg.cgi?id=8173A260175.A1CE5&token=497c658aa47f1c631faa77dfdc3b88d6>

ttlappalainen, 2018-11-03:
You can force it to use mcp_can by definition
#define USE_N2K_CAN 1
Before #include <NMEA2000_CAN.h>
Remember to add other necessary definitions for mcp_can as documented.

But why not to use ESP32 build in CAN?

sailor1958, 2018-11-03:
> You can force it to use mcp_can by definition
> #define USE_N2K_CAN 1
> Before #include <NMEA2000_CAN.h>
> Remember to add other necessary definitions for mcp_can as documented.
> 
> But why not to use ESP32 build in CAN?

Simple to get it verified is my CAN transreciver VP230 defect :)

Thanks for answer

sailor1958, 2018-11-03:
I found a solutions : 
-->
If the free standing on the bus is not 1.65 V, then use pull-up resistors for this.
CANH > RES 3kOhm > GND (Pull down)
CANL > RES 3kOhm > +3,3 V (Pull up)
<--
It works for me  - I can now run the ESP32 with SN65HVD230 (VP230)

Link to solutions : https://github.com/sandeepmistry/arduino-CAN/issues/6#issuecomment-413456049

Thanks for you input

closed 2018-11-03 by sailor1958