ttlappalainen/NMEA2000 #128: Binary Switch Control

schwinn, 2018-11-14:
Hi Timo,

thanks for the great library!

I want to build a binary switch bank and I'm currently unsure, what's the correct way to control the switches.

Should it be done by a pgn 127502 or by a write or command request group function (pgn 126208) to pgn 127501? I did some research on the internet and I'm still unsure.

I'm aware, that I have to extend the group function code to handle it by a group function for pgn 127501.  

Thanks,
T.

ttlappalainen, 2018-11-14:
To be honest - I am not sure. 127502 name is "Switch Bank Control", so I would expect that is for setting binary bank status. Binary bank itself would then inform its staus with 127501. You logic to use group function is in principle OK, but what is 127502 then for? 

Note that with 127502 you can also set single switch status by setting other to NA.

I think CZone uses those for controlling, spying network with CZone devices would give more information.

Should I implement 127502 support?

schwinn, 2018-11-14:
I found this information about switch control:
https://www.panbo.com/maretron-dcr100-nmea-2000-switching/

The information is older, but I haven't found any more up-to-date information about it.

Empirbus seems to use a Proprietary PGN 65280. It's documented here:
https://www.empirbus.com/wp-content/uploads/Application-Specific-PGN-65280_1.pdf

I think CZone uses also some proprietary PGNs, because there's no support for a dimmer in N2K. Do you have access to a CZone device to do a trace?

ttlappalainen, 2018-11-15:
CZone surely uses proprietary, but it probably uses also available PGN:s.

I thought to by units for spying, they are too expensive just for testing. If you are doing that for your own, use 127501 ans 127502 and if necessary some proprietary for your own. If you think to publish it in future, just have good layers on program so you can easily change the communication way after joining to NMEA organization and getting necessary info.

schwinn, 2018-11-15:
I'll do it this way. Thanks for your support.

newbezz, 2018-12-03:
Hi, 
Schwinn: Do you plan to publish your efforts?
 And to Timo: Now that i'm here I might as well ask you one of the dumb questions. I have installed and run your great work on Arduino like boards.  Now I want to to up in the world, and am wondering what would be your "best" among lager boards? Raspberry (do to its rich environment)
Best
Jorgen

ttlappalainen, 2018-12-03:
What you are looking for? I have made extension board for RPi to connect it directly to NMEA2000. RPi works also with PiCAN2 board. But I have not yet tried to send anything with RPi and have not investigated how they handle sending. The problem is that CAN drivers does not normally handle NMEA 2000 fast packet sending right, so I had to modify all supported CAN drivers to get them work with NMEA 2000. One guy was trying to use library on mBed, but there was the same problem and he could not fix the driver problem.

newbezz, 2018-12-04:
As always very very helpful explanation.  Seems as if RPi should do other stuff and Teensy continue in the NMEA role it does so well.
Thanks
Jorgen

Med venlig hilsen / Best Regards
Jørgen B. Christensen

ttlappalainen, 2018-12-04:
That is sure solution. On the other hand there is no development, if nobody tries anything new and publish it. I would try, but just now for some months I do not have time. And as i mentioned, it depends much, what you are trying to do. Teensy can also act like Actisense NGT-1 and then SignalK can use data from it. Also it can send, but I do not know, does SignalK have sending possibilities.

sprokkie, 2020-01-31:
adding 127502 support would be nice, yachtdevices has a module that works with that pgn.
i configured my B&G according to their website, i cannot get it working.
maybe you can look in to this.
YDCC-04 and YDSC-04  the write: ompatible with other NMEA 2000 digital switching devices managed by standard NMEA 2000 PGNs 127501 and 127502.

ttlappalainen, 2020-02-01:
I have no devices to test 127502, thats why I have not added it. I do not how to exactly to use it. I would expect that if you turn on some switch, you send once 127502 on for that switch single switch only. The the device should send 127501, where you can check did it turn on.

sprokkie, 2020-02-01:
no it did not turn on,
I can also not see a 127502 message beeing transferred over the n2kbus.
Can you made a small script sending 127501 and 127502 ?
then i can test that.

ttlappalainen, 2020-02-01:
```
//*****************************************************************************
void SetN2kPGN127502(tN2kMsg &N2kMsg, unsigned char DeviceBankInstance, tN2kBinaryStatus BankStatus) {
    N2kMsg.SetPGN(127502L);
    N2kMsg.Priority=3;
	BankStatus = (BankStatus << 8) | DeviceBankInstance;
	N2kMsg.AddUInt64(BankStatus);
}

//*****************************************************************************
inline void SetN2kSwitchBankCommand(tN2kMsg &N2kMsg, unsigned char DeviceBankInstance, tN2kBinaryStatus BankStatus) {
  SetN2kPGN127502(N2kMsg,DeviceBankInstance,BankStatus);
}

//*****************************************************************************
void SetSwitch(unsigned char DeviceBankInstance, uint8_t SwitchIndex, bool ItemStatus) {
  tN2kBinaryStatus BankStatus;
  tN2kMsg N2kMsg;

  N2kResetBinaryStatus(BankStatus);
  N2kSetStatusBinaryOnStatus(BankStatus,ItemStatus?N2kOnOff_On:N2kOnOff_Off,SwitchIndex);
  SetN2kSwitchBankCommand(N2kMsg,DeviceBankInstance,BankStatus);
  NMEA2000.SendMsg(N2kMsg);
}
```
Then, when you want to turn on switch 0 on bank instance 0 you call:
SetSwitch(0,0,true);
and off
SetSwitch(0,0,false);

Note that B&G may use 126208 for commanding switches.

sprokkie, 2020-02-01:
put this in my arduino sketch ?

ttlappalainen, 2020-02-04:
Please clarify what you try to do. Do you have YDCC-04 or YDSC-04 or do you try to make one? What devices you have on your test system.

sprokkie, 2020-02-04:
i have a teensy3.1 and a vulcan plotter. found the czone config file of the YDCC-04.
I want to switch outputs with the vulcan plotter on the teensy3.1.

The YDCC-04 uses pgn 127501 and 127502, so my simple understanding:
load the config file in the vulcan plotter, let the teensy send the 127501 and 127502 and see what is going on. When looking at the n2kbus with the actisense reader i see the 127501 and 127502 send from the teensy3.1, but no 127501 and 127502 send from the vulcanplotter to the bus.
In the plotter is the fault message that the YDCC-04 is not present.

ttlappalainen, 2020-02-04:
So you have to send 127501 periodically to bus, where you inform you switch states. Then you have to add listener for 127502. When you receive that PGN, you parse switch states from it, set you switch state according to state. Since you update switch states with 127501, Vulcan should see it. Note that you should also add 127501 to Transmit and 127502 to Receive messages for library.

The problem is that your configuration file seem to contain information about YDCC-04. So it may be that even your device is on the bus, Vulcan will not see it, since it expects to see YDCC-04 there. So test first just send peridically 127501 to inform switch states. If Vulcan does not see your device, then you have to try to set all information (serial number, device info, manufacturer number etc.) to library as on configuration file.

sprokkie, 2020-02-04:
I set all relevant information in the teensy. In the device list there is an YDCC-04 device.
The serial numbers are correct as well.
Regardless of parsing the 127502 message, the vulcan should send that on the network.

ttlappalainen, 2020-02-04:
I think Vulcan (or any device) should send 127502 once only, when you change the switch state. It can not send message continuously, since point of digital switches is that you can control them from different places and they may be disabled to force them off.

You do not need to parse 127502 at first. Just see that Vulcan will see your device and send 127502 on switch state change. If you do not change that switch state to 127501, Vulcan should return switch to your state - maybe give error, that it could not switch.

mrbubble62, 2020-02-04:
The online tool https://www.yachtd.com/products/ds/?czone should create a valid configuration for the Vulcan.

YDCC-04 Manual https://www.yachtd.com/downloads/ydccydsc.pdf

sprokkie, 2020-02-04:
i made de valid configuration with that tool.

it just looks like the vulcan doesnt get a response and thinks the device is offline.
if i use the buttons on the vulcan there is no message send

ttlappalainen, 2020-02-04:
Yes I have seen that. But have you tested to use it? What information fixes configured device to Vulcan? Serialnumber etc. Has anybody tried to find, what vulcan configuration file should contain?

sprokkie, 2020-02-04:
i put that file in my vulcan. i pressed a switch button but no message on the n2knetwork

mrbubble62, 2020-02-04:
As I understand when a new config is detected it will replace the config on ALL czone devices in that group so if any device fails the rest of the network keeps working. If your device is not completing the network initialization correctly perhaps it's being ignored.

sprokkie, 2020-02-04:
ok, that makes sense. now to find a way to let the vulcan think there is a module.
Do you know what message need t be end to the czone config to initialise a module ?

mrbubble62, 2020-02-04:
That's the mystery of the czone proprietary messages, please let me know when you find out. The fact the there is a YDCC indicates it's possible to reverse engineer. I doubt BEP approved this device as it wipes the configuration of any other czone devices on the network! 
From the YDCC manual:
> If you already have CZone equipment installed, you will overwrite the exiting CZone configuration
> with our file and your CZone equipment will not function correctly.

sprokkie, 2020-02-04:
the czone devices are not in the config file that you download from the yachdevice website.

AntonisPapa, 2020-03-25:
I work with pgn127501 from the library.
On the garmin plotter that I have appears the 10 buttons that I would like to have.The only problem that I have now it is how to take the request from the plotter to turn on the switches.i am searching if it is necessary to use the pgn126208 to can take the request for changing the status of the switches.

sprokkie, 2020-03-26:
can you share your work so far ?

AntonisPapa, 2020-03-26:
Of course I will do that.

sprokkie, 2020-03-26:
wunderfull,  i am also curious how you got the buttons on your plotter.

AntonisPapa, 2020-03-26:
#include <Arduino.h>
//#define N2k_CAN_INT_PIN 21
#include "NMEA2000_CAN.h"       
#include "N2kMessages.h"

const unsigned long TransmitMessages[] PROGMEM={127501L,127502L,0};

void setup() {

NMEA2000.SetProductInformation("112233", 100, "Binary Switches", "2.6.3.20 (2020-3-20)","1.1.1.1 (2020-3-15)", 0,0,0,0);                                    

 NMEA2000.SetDeviceInformation(112233,140, 30,333,4,0 ); 
  
                       
 Serial.begin(115200);
 NMEA2000.SetForwardStream(&Serial);
 
 NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly,22);
 NMEA2000.ExtendTransmitMessages(TransmitMessages);
 NMEA2000.Open();
  
  pinMode(2, INPUT);
  pinMode(3, INPUT);
  pinMode(4, INPUT);
  pinMode(5, INPUT);
  pinMode(6, INPUT);
  pinMode(7, INPUT);
  pinMode(8, INPUT);
  pinMode(9, INPUT);
  pinMode(10, INPUT);
  pinMode(11, INPUT);
  
 }

void loop() {
  
 SendN2kBinaryStatus();
 NMEA2000.ParseMessages();
  

}

#define SwitchUpdatePeriod 1000

typedef int64_t BankStatus;

void SendN2kBinaryStatus(){
unsigned char BankInstance=0;
static unsigned long SwitchUpdated=millis();
tN2kMsg N2kMsg;
tN2kOnOff Status1,Status2,Status3,Status4, Status5,Status6,Status7,Status8, Status9,Status10;  
  
    
if ( SwitchUpdated+SwitchUpdatePeriod<millis() ) {
    SwitchUpdated=millis();

     
     if(digitalRead(2)==LOW )
      { Status1=N2kOnOff_On; 
         }
      else if (digitalRead(2)==HIGH )
      { Status1=N2kOnOff_Off; 
        }
        
      if(digitalRead(3)==LOW )
     { Status2=N2kOnOff_On; 
       }
     else if (digitalRead(3)==HIGH )
     { Status2=N2kOnOff_Off; 
       }
       
      if(digitalRead(4)==LOW )
      { Status3=N2kOnOff_On; 
        }
      else if (digitalRead(4)==HIGH )
      { Status3=N2kOnOff_Off; 
        }
        
      if(digitalRead(5)==LOW )
     { Status4=N2kOnOff_On; 
       }
     else if (digitalRead(5)==HIGH )
     { Status4=N2kOnOff_Off; 
       }
      
      if(digitalRead(6)==LOW )
       { Status5=N2kOnOff_On; 
         }
        else if (digitalRead(6)==HIGH )
        { Status5=N2kOnOff_Off; 
         }
 
       if(digitalRead(7)==LOW )
       { Status6=N2kOnOff_On; 
         }
       else if (digitalRead(7)==HIGH )
      { Status6=N2kOnOff_Off; 
        }
        
      if(digitalRead(8)==LOW )
      { Status7=N2kOnOff_On; 
        }
      else if (digitalRead(8)==HIGH )
      { Status7=N2kOnOff_Off; 
        }
        
      if(digitalRead(9)==LOW )
      { Status8=N2kOnOff_On; 
        }
      else if (digitalRead(9)==HIGH )
      { Status8=N2kOnOff_Off; 
        }
 
      if(digitalRead(10)==LOW )
      { Status9=N2kOnOff_On; 
        }
      else if (digitalRead(10)==HIGH )
      { Status9=N2kOnOff_Off; 
        }
        
      if(digitalRead(11)==LOW )
      { Status10=N2kOnOff_On; 
        }
      else if (digitalRead(11)==HIGH )
      { Status10=N2kOnOff_Off; 
        }
      
   SetN2kBinaryStatus(N2kMsg,0,Status1,Status2,Status3,Status4, Status5,Status6,Status7,Status8, 
  Status9,Status10);
   NMEA2000.SendMsg(N2kMsg,0);  
  }
}

AntonisPapa, 2020-03-26:
[image]
[image]
And that is the images.

AntonisPapa, 2020-03-26:
[image]

sprokkie, 2020-03-26:
wow that looks great, hos did you make the switches on your plotter ?
My B&G doesn't have a predefinded button field of possibility

AntonisPapa, 2020-03-26:
I check that when running in the network the pgn127501 information automatically recognized from the plotter.Before I did not have these option on my plotter.

virtuvas, 2020-03-26:
Antoni,

which plotter is this? 7** ?
curious if that will work on my GPSMAP 4008 and 5008 or they are too old!

cheers

V

AntonisPapa, 2020-03-26:
I have the gpsmap1022.I think that if you make the new update 7.80 maybe you can have it,but I am not so sure.I check in the Garmin update but I do not find something there.

ttlappalainen, 2020-03-26:
Have you read the bus how your MFD would handle button press. There are two ways:
it sends 127502 with switch information it wants to change or it uses 126208, which is more complex. So extend your ReceivePGNList with 127502 and then spy bus doesgpsmap send that pgn on switch press.

AntonisPapa, 2020-03-26:
I check the traffic of the nmea 2000 the plotter send pgn127502 and pgn126208 you have right.
My problem for now is to collect the information from the pgn127502 but i could not add on the library to call them then.
The pgn126208 is much difficult to check it and it is neccecary to send and listen all the commands.
If you have any idea please let me know.
thanks

diginautic, 2020-04-16:
Hi,
I just managed to send a 127501 to my Garmin 922xs and all of a sudden switches appered on my screen. When I set one bank switch high in 127501 the corresponding switch in my MFD turns high 😀👍.  Now my problem is to manage to parse 127502 sent from my MFD when pressing a switch. Has someone managed to do that?

best regards 
Niklas

sprokkie, 2020-04-16:
would you share your code ?

AntonisPapa, 2020-04-16:
If you use my code you can see that is working in these way.
I make a lot of tries to take the neccessary phrase from the nmea 2000.
First I change the divice settings to listen and send
NMEA2000.SetMode(tNMEA2000::N2km_ListenAndSend);
And now I am try to separate the phrase that running in the network to take only the 
Pgn127502 and then to change the status of the switch.
The phrase of 127501 and 127502 can manage 28 switches in 1 sid.

ttlappalainen, 2020-04-17:
N2km_ListenAndSend mode is for analysing and test purposes. Device will then not respond to requests. This will cause that it will disappear from the devices list on MFD.

Parsing 127502  is similar as 127501. When you receive 127502, you should change your switch status according received information. Note that you should not touch to any switches, which values on messages is NA. Then you update 127501 as normally.

AntonisPapa, 2020-04-17:
You have right  Timo.
I used this
NMEA2000.SetMode(tNMEA2000::N2km_ListenAndNode,22);
But I could read the phrase of pgn 127502.

diginautic, 2020-04-17:
Hi.

Of course I will share my code. But It might not be all pleasant reading ;)
It partial has some swedish names but without the dots. How should I share
it?

My project is based on a Arduino Mega2560, a CAN-BUS shield with a MCP2515,
a Nextion Enhanced display 7". The system "Diginautic" originally aimed to
collect engine RPM, oil pressure and coolant temperature. But while I got
to know Arduino and that "world" I realized that a touch display and some
relays would easily replace my ordinary switches. And while I was going I
found the ACS712 current sensors and now I have:

Arduino Mega2560
CAN-BUS shield
Nextion Enhanced 7" display
16 relays
13 current sensors
2 op-amps measuring voltage drops over 3 shunt resistors
I2C-bus with 4 16-bits ADC to collect data

Along that in my boat I have Garmin MFD, AIS, VHF, Fusion FM-Radio, Compass
and GFL10 Fluid Level sensor on my NMEA2000-bus. I collect lots of data
from N2k and send quite a lot of data back on N2k.

When I change state on any of my relays I send a 127501 on N2k to my GPS:

When pressing button 1 I do this
N2kSetStatusBinaryOnStatus(SwitchBoard,N2kOnOff_On,1);
SendN2kBinaryStatus(true);

My next step is to add support for 127502 to receive that PGN. The thoughts
I have is to do the opposite of 127501. Parse SwitchBoard to find out what
button was pressed on my GPS. After that I'll send a corresponding 127501
telling my GPS that it actually was turned on.

I'll let you know on my progress
/Niklas

ando274, 2020-06-24:
Hi All,

I have followed this topic with interest, I have been using Timo's excellent and robust libraries and have built a number of projects using the Teensy and Rpi platforms for my yacht. 

Digital switching from my Raymarine Axiom plotter being a most desirable addition, after several failed attempts, it is probably worth relating what I have learnt.

The Axiom supports 3 vendors in digital switching, Empirbus, Yacht Devices, and CZone. All require a custom Qt page, designed and loaded. This isn't so hard as Empirbus and Yacht Devices have tools on there websites that generate the required pages.

Both Empirbus and Yacht Devices both publish some information regarding the N2K PGN's their devices use, mainly dealing with the switching PGN's, so far so good.

The problem is the Axion needs to detect the hardware is on the bus before it enable the switching page or the onscreen switches. 

This in my opinion, during the startup interaction between the device and Axion, and is probably in the Vendor/Device/Serial Number PGN's exchanges between the two devices. 

Having tried several time to simulate these exchanges, I have concluded without an actual vendor device to look at these interactions, it is near impossible.

Regards

ttlappalainen, 2020-06-24:
Have you tried to just set device manufacturer ID to Yacht Devices 717? They use standard messages 127501 and 127502

diginautic, 2020-06-24:
Hi!

On my boat I only use 127501 and 127502 and no configuration files in my
plotter. I have a Garmin 922xs and it works perfectly.

Regards
Niklas

AntonisPapa, 2020-06-24:
Hi guys!

Please if you could can you share with us the code that you use?
The Garmin plotters are recognize direct from the pgns the modules that
have been installed in the network.

The raymarine with empire bus works in different pgns and to appear in the
network it is neccessary to fix an html5 page with the number of the items
icons amd etc.

sprokkie, 2021-01-05:
Hi Timo,

 

Last years i was working on an engine interface, that worked fine at home on a B&G vulcan 5 plotter.

Last winter I installed it on my ship, a bigger n2k network with more sensors and I encounter some difficulties.

After playing for a whole season with settings on the network I finally gave up.

It even prevents me from experimenting with outputs on the N2knetwork

 

 

The device is not visible and then it is  for a while and then it is gone no data on the data fields and there is again in the device list.

Basically it random appears and disappears on the netwok

It look lik It fight’s with the N2k fusion radio , that stops working and need a hard reset,

I changed the network address but that doesn’t do the trick.

Even build in a delay that works sometimes but not consistent

 

Obviously I am doing something wrong can you help me?

I used your example and modify some setting so it works for me.

Perhaps I need to add or remove something.

 

The code that is used initial is below:

 

// Yanmar motor interface, various sensors added to the engine block

// Need about 1 minute starting delay, other wise the plotter wil not recognize the interface,

// i have to sort this out some day........

// Big thank you to Timo Lappalainen for sharing the N2K lib on github.

// 

 //Demo sketch modified to this version by F. Meijlink. 04-2020

//Running on a Teensy 3.2

 

#include <Arduino.h>

#include <NMEA2000_CAN.h>  // This will automatically choose right CAN library and create suitable NMEA2000 object

#include <N2kMessages.h>

#include <N2kMessagesEnumToStr.h>

 

#define Serial SerialUSB

#include <U8g2lib.h>

[... 1152 lines cut ...]
        NMEA2000.SendMsg(N2kMsg);

    //SetN2kCOGSOGRapid(N2kMsg, 1, N2khr_true, DegToRad(115.6), 0.1);

    //NMEA2000.SendMsg(N2kMsg);

  }

}

 

 

 

 

 

 

Many thanks in advanced

Ferry Meijlink

ttlappalainen, 2021-01-06:
Unfortunately you have lot of things wrong on you sw. Lets think you are running laundry. So you will wain on the desk for customer. When customer arrives, you will take the stuff and put them to the washing machine. Next you wait on front of washing machine that it is ready and move them to dryer and wait that to be ready. Finally you packe them all and go to back to desk waiting for customer. Meanwhile there has been coming and going customers, which did not got any service. But luckily there is just arrived one, who you can server and do the round again. This is how your program is doing now. Any delay call means waiting on washing machine that it is ready. Compared to washing machine 2 hour wait delay(100) does not sound long, but for processors it is like that. So within any loop code call to delay is big no except for testing purposes.

So lets start to fix your code.
- first remove all commented code you do not need and can be found again from examples.
- next start to think modularity. Even in Arduino project you can split code to modules simply creating an other .ino file. E.g. just create YamahaPicture.ino under same folder your main ino is and move that there. It should work. The good thing for beginners on Arduino IDE is that you do not need to declare thing first as in normal programming environment. It is enought it exist in some module inside the project. You can also make one module for drawing routines.
- the most complex thing is to start to think tasks. Arduinos and Teensies does not have internal tasking. Means that all code will run sequentially and any delay stops everything running like on waiting washing machine. So in example "wash laundry"  task you simply put thing to machine, wait it ready, move them to dryer, wait it ready, pack laundry and done. You have to chage that to work other way: 
"wash laundry"  task 
- If state is not started set washing state, put laundry to washing machine, exit.
- if state is washing if machine is not ready exit.
- move laundry to dryer, set drying state and exit
- if dryer is not ready exit
- take laundry, pack them and deliver them for waiting customer exit
So with above task you can server several customers in day instead of two and your laundry may survive. It is exactly same with your sw. 

So e.g. your RMP code will change to:
```
void CountRPM() {
#define RPMSamplePeriod 1000
  static unsigned long NextCheck=millis()+RPMSamplePeriod;
  static unsigned long StartTime=millis();

  unsigned long Now=millis();
  
  if ( NextCheck<Now ) {
    NextCheck=Now+RPMSamplePeriod;
   rpm = 60*1000/(Now - StartTime)*rpmcount;
   StartTime=Now;
  }
}
```
Note I did not wate time for stop start interrupt, since reading rpmcount is atomic and can be meshed up. With float you shoud do stat/stop.

See no delay call and it will do the as your old code. Then do the same for all of your tasks and finally your loop will look like:
void loop() {
  NMEA2000.ParseMessages();
  CountRPM();
  ReadTemperatures();
  SendN2kMessages();
  UpdateDisplay();
}

Then more hints:
- RPM measurement code is not very accurate. With 1000 RPM you get 16.6 pulse/s (one pulse per rev). So the reading may see 16 or 17 pulse and RPM will vary 960-1020. So you either need to filter reading or use https://www.pjrc.com/teensy/td_libs_FreqMeasure.html
- DallasTemperature reading also meshes up things since as default it has long (700 ms) delays. You have to change to use it in asynchronous mode and make it similar task as RPM example. See the DallasTemperature library, there should be WaitForConversion2 asynchronous example.
- You have to also check display drawing routines. Sometimes they are orful slow and you have to split them like "wash laundry"
- The goal is that your loop time should never exceed 20 ms. Temperature reading, when it is ready will take about 15 ms, so that will happen time to time. Mostly your sleep time should be <1 ms. You can add loop time checking to your loop:

```
void CheckLoopTime() {
#define LoopTimePeriod 1000
  static unsigned long NextCheck=millis()+LoopTimePeriod ;
  static unsigned long AvgCount=0;
  static float AvgSum=0;
  static unsigned long MaxLoopTime=0;
  static unsigned long StartTime=micros();

  unsigned long UsedTime=micros()-StartTime;
  if ( UsedTime>MaxLoopTime ) MaxLoopTime=UsedTime;
  AvgCount++;
  AvgSum+=UsedTime;

  if ( NextCheck<millis() ) {
    NextCheck=millis()+LoopTimePeriod;
    Serial.print("- Loop times max:"); 
    Serial.print(MaxLoopTime); 
    Serial.print(" us, avg:"); 
    Serial.print(AvgSum/AvgCount); 
    Serial.println(" us");
    MaxLoopTime=0;
    AvgSum=0;
    AvgCount=0;
  }

  StartTime=micros();
}
```

Happy programming!

sprokkie, 2021-01-06:
Hi Timo,

 

Many thankd for youre answer.

 

What i don’t understand that is working perfectly on my desk.

The n2k network on my ship is has more n2k members, I cannot see how delaying some things inside the void loop has influence on the behavior of the teensy in the network.

I expect that the data is not send regulair to the network but missing the whole device for a while I don’t understand.

 

I will read your answer a couple of times and see if I can understand your answer, the part about splitting is new for me don’t know how to do that I have to google more ill gues.

 

Kind regards,

Ferry Meijlink

ttlappalainen, 2021-01-06:
In my laundry example delay in loop is the same as you miss customers, when you are just looking washing machine going around. When other devices requests something, library can not server them, since you just waiting other task to finish. When they does not get served, they remove your device from the list. And then you again send something and they will request who you are.

Just believe me - you have to do those improvement I listed. Specially no delay calls and loop time avg<1ms.

Please do not copy previous responce, since it is available above. If you need to highlight some sentence take only that.

MrClowny666, 2021-02-10:
Hello Boss,

Me again, will you be merging the 127502 code into the main source library?

ttlappalainen, 2021-02-11:
Have you found document, how it should be used? I expect that one should send only bits one would like to change. Others should be marked as NA.

One problem is that 127502 is deprecated and should be replaced with 126208. But if device does not support 127502, how other device can know that it can do switch control?

ttlappalainen, 2021-02-21:
Please send your code for review

christianolsman, 2021-02-28:
Hi all, has anyone got the Yachtdevices screen to work on an Axiom plotter? My YachtDevices custom page is saying "no configuration available". I am sending 127501 which  I can see in the plotter's diagnostics page.

christianolsman, 2021-03-11:
Hi again, I have got some more information out of Yacht Devices about the YDCC-04N. When the Axiom plotter i starting it is asking units to report back and first if it finds a device matching the Zcone configuration it will enable the digital switching screen, I listened to the traffic of my Axiom at startup and I could see that it was, by PGN 059904 ISO request, asking devices to reply with PGN 126464 ( PGN-list Tansmit PGNs group function). I don't think I can come further without having a YDCC-04N to listen to. Anyone who has one and can listen to what it is reporting back on PGN 126464, or any other PGN for that matter?

ttlappalainen, 2021-03-12:
That is normal operation for MFD and other devices, which will have any cross communication on N2k bus. Also my library handles tha automatically. If you tell library which PGNs your device support with ExtendTransmitMessages / ExtendReceiveMessages, it will automatically tell others on request for either 059904 or 126208 all supported PGNs. Library user does not need to care of that functionality - just tell library supported PGNs. Instead if you e.g. support 127501, you should add handler for ISO request (PGN 59904) with SetISORqstHandler and send requested PGN and return true or false, if you do not transmit requested PGN. And if you go futher, you should inherit tN2kGroupFunctionHandler for your PGNs and register those handlers.

I can understand that MFD will enable control buttons, if your deveice informs that you support 127501 as tx and 127502 as rx. But as far as I know 127502 has been marked deprecated, so I do not know what would be the replacement for that.

christianolsman, 2021-03-13:
Thanks, Timo. I added the transmitted and received pgns listed in the manual of tge YDCC-04N and I can see them being received in the plotter. Still no switches though. The Yachtd support mentioned that the serial number of the device and the uploaded czone config must be identical so I am suspecting that I somehow is not corretly set. No other PGNs are transmitted   from the MFD as part of the initial phase. However , After 5-10 minutes of runtime I could see a PGN within the proprietary PGN range being sent from the MFD.  

I use the properties below. 

NMEA2000.SetProductInformation("717   ", // Manufacturer's Model serial code
                                 8212, // Manufacturer's product code (The one Yachdevices uses)
                                 "YDCC-04",  // Manufacturer's Model ID
                                 "1.03",  // Manufacturer's Software version code
                                 "Circuit Control / YACHTD.COM", // Manufacturer's Model version
                                 00260001,
                                 2100
                                );

  NMEA2000.SetDeviceInformation(00260001, // Unique number. Use e.g. Serial number.
                                150, // Device function=Analog to NMEA 2000 Gateway. See codes on http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                30, // Device class=Inter/Intranetwork Device. See codes on  http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                206 // Just choosen free from code list on http://www.nmea.org/Assets/20121020%20nmea%202000%20registration%20list.pdf
                               );

[54 more replies]