ttlappalainen/NMEA2000 #129: “Engine Discrete Status is abnormal”  PGN127489

tmsio, 2018-12-05:
Hi, i have a simple sketch send two engine flow rate this is part of code:
[code]
double Engine0Flow() {
  return lh1;
}

double Engine1Flow() {
  return lh2;
}

void SendN2kFlow() {
  tN2kMsg N2kMsg;
  SetN2kEngineDynamicParam(N2kMsg, 0, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, Engine0Flow(), N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kInt8NA, N2kInt8NA, true);
  NMEA2000.SendMsg(N2kMsg);
  if ((ver == 0001) | (ver == 0002) | (ver == 0003))
  {
    SetN2kEngineDynamicParam(N2kMsg, 1, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, Engine1Flow(), N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kInt8NA, N2kInt8NA, true);
    NMEA2000.SendMsg(N2kMsg);
  }
[/code]
No problem on garmin or simrad, but on Furuno TZTL12F i recive the warning:

“Engine Discrete Status is abnormal”
I describe the issue to Furuno company and they answer me:

【Answer】

PGN127489 is indicated by the engine instance field on NMEA2000.

The Error message means “Engine Discrete Status is abnormal”.

When Engine Discrete Status is normal, the Error message disappears.

(TZTL12F/15F can’t ignore the error information.) Please refer to the manual for engine.

Can someone help me?

Thank you

Regars

tmsio, 2018-12-07:
> You have set flagCheckEngine to true, so doesn't that mean that you have problem with engine? Was that your meaning? If yes doesn't Furuno then work right - you should check you engine and turn it off before it smokes. Have you tried how Furuno says, if you have flagCheckEngine false, but flagOverTemp true `SetN2kEngineDynamicParam(..., N2kInt8NA, false, true);`
> 
> I do not have NMEA 2000 standard information about using discrete status. Normally NMEA 2000 fields has 2 state - data and "not available". Discrete status only has only one bit for each state, so it can not have "not available" status. So each bit should show 0, if status is not active. Still I have seen devices sending all 1 (FF FF) on discrete status. So does that mean that it does not provide information or does it mean that you are in big trouble with engine? There one thing you could test for N2kMessages.cpp:
> On line 237 `int engineStatus2 = 0;` Change that to `int engineStatus2 = 0xff00;`
> The upper byte of second discrete status is reserved, so should it be 1:s or 0:s is open for me. What if comittee decides to add new status bit, then old systems sends it as active. They should have use inverse status.

Hello @

tmsio, 2018-12-07:
Hi, @ttlappalainen i buy a Actisense Nmea Reader for debg Nmea sentences.
I found solution tu recive 0 from Engine Discrete Status 1
I simple put a comment to N2kMessages.cpp:On line 238
//if (flagCheckEngine) engineStatus1P1      |= BIT(0);

Now i Recive 0 (see attached file)

Could this change create some problems?

Thank You
Regards

[image]

ttlappalainen, 2018-12-07:
No. As I mentioned you have set flagCheckEngine to true on your call:
`SetN2kEngineDynamicParam(N2kMsg, 0, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, Engine0Flow(), N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kInt8NA, N2kInt8NA, true);`

So simply do not touch the library code and set flagCheckEngine to false, if you not want to inform to the bus that engine has failure.
`SetN2kEngineDynamicParam(N2kMsg, 0, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, Engine0Flow(), N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kInt8NA, N2kInt8NA, false);`

I thought that you wanted to inform that engine has failure and thats why you set flagCheckEngine true.

ttlappalainen, 2018-12-07:
No problem. So now you know that you can use those flags and get warn to MFD, if you have suitable sensors. You could e.g. wire temp or oil pressure warn switch to your device.