ttlappalainen/NMEA2000 #131: Send failed

sinise, 2018-12-22:
I  have a teensy 3.2 conector to a raymarine axiom via a SN65HVD230D transiver.

Im using you version of the flex can library.

I have still not been able to get any messages send to the raymarine.

When I run the examples and look at the serial monitor I look like messages are being send. But after around 30 seconds it fails on each message it's trying to send. The error message is "send failed"

When running the listener examples I don't se any  data.

Can the send failed error be triggered if the can transiver is not working? And why does it then not failed on the first messages it tries to send.

Do you have a clue what could be wrong. 
I guess I should probably order a mcp 2562 as this is what I understand should be working

Thanks for creating this library

ttlappalainen, 2018-12-23:
Library has internal buffer, where unsent messages will be buffered. You get sent failed, when buffer gets full. 

The main reason is that CAN cotroller can not communicate with bus. Often reason is missing terminator, switched wireing (L->H, H->L) and tranceiver problem. Also for some readon people have problems with SN65HVD230D - I do not unerstand why, since it should be compatible. Aftern changing to MCP2562 everything has been OK. How you have connected RS pin?

> RS: SN65HVD230 and SN65HVD231: Mode select pin: strong pull down to GND = high speed mode, strong
> pull up to VCC = low power mode, 10kΩ to 100kΩ pull down to GND = slope control mode

usauerbrey, 2018-12-23:
Hi
Just to add, I am using https://www.tindie.com/products/Fusion/dual-can-bus-adapter-for-teensy-35-36/ It uses SN65HVD230D transceivers. Together with high speed mode (strong pull down to GND) it works fine.

ttlappalainen, 2018-12-23:
With that module you had changed default CAN pin setting or use only second CAN. I wrote decription how to do that, but do not remember where. Could you add it here just for remainder.

sinise, 2018-12-23:
I missed the part about that I need to add a pull down/up resistor. I will do that and try again. I gues it should be a pull down so that I activate high speed mode.? Thanks for the fast reply

usauerbrey, 2018-12-23:
Hi Timo

You wrote:
----------------------------------------------------------------------------------------------------
There are few things. If you want to use my library, it is in your case best to fix it to Teensy. So instead of having:
#include "NMEA2000_CAN.h"
use
#include <N2kMsg.h>
#include <NMEA2000.h>
#include <FlexCAN.h>
#include <NMEA2000_teensy.h>
Then you can define:
tNMEA2000_teensy NMEA2000_1();
tNMEA2000_teensy NMEA2000_2(4,1);
Then you should have two library objects, which you can initialize. The only problem is that I have never tested two objects at same system. But I know that it is possible to use second port.
The last thing is that your adapter uses alternate pins for CAN0. As default Teensy 3.5/3.6 has CAN0 on pins 3/4. So you have to change Teensy pins to get CAN0 working. I have not populated that functionality to NMEA2000_Teensy. You could try does it work, if you change pins on setup():
CORE_PIN29_CONFIG = PORT_PCR_MUX(2);
CORE_PIN30_CONFIG = PORT_PCR_MUX(2);
See FlexCAN.cpp void FlexCAN::setPins
And please use FlexCAN under my git. Delete all other FlexCAN installations possibly installed with Teensyduino.
-----------------------------------------------------------------------------------------------------------

So I tried several changes including hard coded changes in https://github.com/collin80/FlexCAN_Library, but with no success.
So finally I use the board connected to the original pins used by you and only as a single channel transceiver ;-((

sinise, 2018-12-23:
It was the missing pull down resistor. Thank you for your fast reply and thanks for sharing this library. happy Xmas to you

closed 2019-01-18 by ttlappalainen