ttlappalainen/NMEA2000 #133: setProductInformation values conflicting with Maretron DSM250

virtuvas, 2019-01-26:
Timo,

had a strange situation yesterday and looks like I solved it but I don't exactly know why, so reporting it to you and hopefully get some ideas back...

Right, boat N2K bus has:
9Garmin
1Airmar (DST800)
1NoLand Engineering (engine interface)
1 teensy
2 due

all was fine datawise, although occasionally on the NMEA2000 list of devices, one or two of mine would flash as unknown and eventually stabilise with name, serial, etc. But all data were broadcasted fine.

Now, added an extra device, a Meretron DSM250. Tee was there, just added the device. 
No change in the whole bus. 
Length of main backbone must be around 20m, most tees have 1m cables, 3-4 are 2m long.

On adding the Maretron, one of my two dues started disappearing/reappearing on the bus and eventually settled with a merge of the device name with the Maretron, and the serial number of the due...
So alone each device would show up in the list as:

Maretron DSM250 (serial 1320713)
VosperFin Controller DASH (serial 00212127)

now when both in the bus (with all the other stuff on of course) Id settle (after a minute or so) to:

Maretron VosperFin Controller DASH (serial 00212127)

after lots of head scratching I altered the SetProductInformation settings on the due, same behaviour (even after turning off the bus for a few mins or so)
Now this morning went back again, refitted it, tested it, worked...

The original setup part was:

`// Setup the NMEA
  Serial.begin(115200);
  OutputStream=&Serial;

//  NMEA2000.SetN2kCANSendFrameBufSize(100);
  // Set Product information
  NMEA2000.SetProductInformation("00212127", // Manufacturer's Model serial code
                                 210, // Manufacturer's product code
                                 "VosperFin Controller DASH",  // Manufacturer's Model ID
                                 "1.2.60",  // Manufacturer's Software version code
                                 "1.2.60, Oct 2018", // Manufacturer's Model version
                                 0xff, // load equivalency - use default
                                 0xffff, // NMEA 2000 version - use default
                                 0xff // Certification level - use default
                                 );                    
  //VosperFin Interactive
  NMEA2000.SetDeviceInformation(00212127, // Unique number. Use e.g. Serial number.
                                180, // Device function. See codes on http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                40, // Device class. See codes on  http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                210, // Just choosen free from code list on http://www.nmea.org/Assets/20121020%20nmea%202000%20registration%20list.pdf                               
                                4 // Marine
                                );

  NMEA2000.SetForwardStream(&Serial);
  NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly,15);
  NMEA2000.EnableForward(false);
  NMEA2000.ExtendTransmitMessages(FinSenderINTRTransmitMessages,0);

  NMEA2000.SetMsgHandler(HandleNMEA2000Msg);`

the corrected one is:

`  Serial.begin(115200);
  OutputStream=&Serial;

//  NMEA2000.SetN2kCANSendFrameBufSize(100);
  // Set Product information
  NMEA2000.SetProductInformation("0100212127", // Manufacturer's Model serial code
                                 210, // Manufacturer's product code
                                 "VosperFin DASH",  // Manufacturer's Model ID
                                 "1.2.65 (Jan 2019)",  // Manufacturer's Software version code
                                 "1.2.00 (started Oct 2018)", // Manufacturer's Model version
                                 0x09, // load equivalency - use default
                                 0xffff, // NMEA 2000 version - use default
                                 0xff // Certification level - use default
                                 );                    
  //VosperFin Interactive
  NMEA2000.SetDeviceInformation(0100212127, // Unique number. Use e.g. Serial number.
                                180, // Device function. See codes on http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                40, // Device class. See codes on  http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                210, // Just choosen free from code list on http://www.nmea.org/Assets/20121020%20nmea%202000%20registration%20list.pdf                               
                                4 // Marine
                                );

  NMEA2000.SetForwardStream(&Serial);
  NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly,15);
  NMEA2000.EnableForward(false);
  NMEA2000.ExtendTransmitMessages(FinSenderINTRTransmitMessages,0);

  NMEA2000.SetMsgHandler(HandleNMEA2000Msg);`

wondering if you have any idea about all that or if you find anything wrong in my code. This problematic due is using an MCP2551 (not 2562, sorry fitted it before you mentioned it!) but is working perfectly, communicating with the teensy and instructing it, as well as getting data to present in the screen from the teensy. All's fine using both custom sentences and a few proper ones for speed, temp, etc

Note I did change the load equivalency to 9 from default as this device runs a 7inch screen (using the N2K bus power, maybe not such a great idea, I'll probably redesign it to get external power)

I also wonder what is the point of declaring the load, is there some internal bus function related to it that would explain why the system eventually worked after I declared a larger value?

confused, but happy that everything works :)

cheers

Vassilis

virtuvas, 2019-01-26:
Timo, just noticed that on the other Due that works fine, I don't have:

>  NMEA2000.SetMsgHandler(HandleNMEA2000Msg);

in the setup.

And I don't have this message handling routine either:

//NMEA 2000 message handler
>void HandleNMEA2000Msg(const tN2kMsg &N2kMsg) {
>  int iHandler;
>  
>  // Find handler
>  for (iHandler=0; NMEA2000Handlers[iHandler].PGN!=0 && !(N2kMsg.PGN==NMEA2000Handlers[iHandler].PGN); iHandler++);
>  
>  if (NMEA2000Handlers[iHandler].PGN!=0) {
>    NMEA2000Handlers[iHandler].Handler(N2kMsg); 
>  }
>}

I understand that it's needed as it's also picking up things from the bus, whereas the other Due is only sending things on the bus...
Could there be any relation?

cheers

V.

ttlappalainen, 2019-01-27:
No good ideas yet.

Some things:
- Load is informative value for analyzers or some MFD:s so that you can see total load of the bus power. It should not effect anything.
- On SetDeviceInformation(0100212127... only 21 bits will be used, so it will be masked to 1 1001 0001 1101 1001 1111=1645983 This anyway setups unique name on the bus and it is only important that you do not have other device with same settings on the bus. Device information like serial number is normally picked from product information.

I would first change tranceiver. How do you drive it, 5V or 3.3V? If you use 5V, you overload DUE input. For 2551 documents says Vdd 4.5-5.5, so it may not work well with 3.3V.

In principle every device product information goes through the library, it will some time also get string "Maretron". But I expect that there would be a lot more issues open, if library could concatenate that string successfully to your product information.

One problem on your information:
Maretron DSM250 (serial 1320713)
VosperFin Controller DASH (serial 00212127)
is that I do not know where Maretron name comes. There are two possibilities:
1. It is concatenated by <Manufacturer> <Manufacturer's Model ID> Manufacturer will be then get from device information mfgcode (137 for maretron). In this case it does not show anything for you, since 210 is unknown. Anyway MFD:s often shows "unknown manufacturer" for unknown.
2. It is withing Maretrons <Manufacturer's Model ID>

Above you can see, if you read bus with NMEA Reader, which shows you all information separated. This is important to get some glue, what is going wrong. In first case it means that it is reading mfgcode 137 for your device.

For your second note about message handler I would say that it either should not have effect. Naturally you have to remember that within handler you are not polling bus so if you use any delay(xx) function within handler or actually anywere on your program, you may loose important packets like product information request and then your device will definetely popup and popoff on the MFD.

virtuvas, 2019-04-10:
Timo,

added another two "black boxes" on the bus, one for each engine doing various EGT (with the problems discussed in another issue), gearbox oil pressure and temp, engine oil temp and coolant pressure.

Same/similar things happened with them, merging names between the Maretron and one of them.
BTW, boxes are based on ESP32S chips.

Took me a few hours to solve the issue by simply changing the product information to a 6digit number and changing the first number from 1 to 9... Took me three goes to sort it.
However checking with the method you pointed above, I can see that all the numbers I've used are different to the Maretron serial as displayed in the GMI10 screen (NMEA2K list of devices)

Confused but happy that it works...
Any ideas?

V.

ttlappalainen, 2019-04-10:
Just noticed that you have been using 210 for manufacturer code. On the list it shows free, but you could also test something else like 1999.

closed 2020-01-20 by virtuvas