ttlappalainen/NMEA2000 #134: Materon DST 100 sensor Arduino Mega interfacing

pushkalkatara, 2019-01-28:
Hi, I'm trying to interface Materon DST100 with Arduino Mega using this example - 
https://github.com/ttlappalainen/NMEA2000/blob/master/Examples/ArduinoGateway/ArduinoGateway.ino 
and following this document for connections https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/ArduinoMega_CAN_with_MCP2515_MCP2551.pdf

1. Where can I find the product information to fill in arduino code?

ttlappalainen, 2019-01-29:
I do not understand what do you mean. ArduinoGateway.ino is just replacement for making Actisense "compatible" device to e.g. read data from bus with Actisense compatible applications on PC.

So please explain what you try to do.

pushkalkatara, 2019-01-30:
Hi,
I am trying to integrate Materon DST100 using Arduino Mega, I want to receive the data on arduino serial and print on Arduino Serial monitor.
I tried using the ArduinoGateway.ino but not getting the data from the sensor.

ttlappalainen, 2019-01-31:
Try also with ActisenseReader.ino. If that does not work, it is best to check hardware. Do you have temination resistos on the bus? What clock you CAN shield uses - 8MHz or 16MHz - 16MHz is default on library. Are CANL/CANH right. and so on.

pushkalkatara, 2019-02-04:
Hi. sorry for the late reply. The clock we have on CAN shield is 8MHz, how can we change the default clock from 16MHz to 8MHz in the library

jiauka, 2019-02-04:
#define USE_MCP_CAN_CLOCK_SET 8
 before including NMEA2000 library.

ttlappalainen, 2019-02-04:
More precicely use
#define USE_MCP_CAN_CLOCK_SET 8
#include <NMEA2000_CAN.h>

Options are described on NMEA2000_CAN.h.
I would not enable any debugs on library. I get very messy.

pushkalkatara, 2019-02-05:
Hi, I tried using the following 
```
#define USE_MCP_CAN_CLOCK_SET 8
#include <NMEA2000_CAN.h>
```
Still no data on the serial monitor. What might be the issue? I can hear a ticking sound from the DST100 sensor.

ttlappalainen, 2019-02-07:
Did you build up you system according to my drawings from chips or are you using some ready shield? In my drawings there is 16 MHz chrystal. If you made it as in my drawing, you should not use USE_MCP_CAN_CLOCK_SET 8. If you use some shield ready on market, you have to check what clock it has. With ready shields you have to also check the interrupt signal pin.

And what about termination resistors? On NEMA 2000 bus you must have two 120 ohm resistors between CANH and CANL at  longest end of the bus. If you are testing this on your table, one is enough.

pushkalkatara, 2019-02-07:
Thanks for the reply, we are using a ready sheild, it has a 8Mhz clock and a 120 ohm terminal resistor. I am using the directive USE_MCP_CAN_CLOCK_SET 8, still there is no data on serial monitor.
Here is the link to the shield I'm using - https://robu.in/product/mcp2515-can-module-tja1050-receiver-spi-51-single-chip-program-routine-arduino/

ttlappalainen, 2019-02-07:
You could also try with example MessageSender.ino. Remember to add clock set definition and uncomment the line `NMEA2000.SetForwardType(tNMEA2000::fwdt_Text);`
If that starts to report "PGN xxx send failed", you have definetely hw problem.

pushkalkatara, 2019-02-08:
What might be the issue. I am getting the same error
```
18:09:22.176 -> 67668 : Pri:3 PGN:127488 Source:22 Dest:255 Len:8 Data:0,F8,43,FF,FF,7F,FF,FF
18:09:22.176 -> PGN 129026 send failed
18:09:22.176 -> 67674 : Pri:3 PGN:129026 Source:22 Dest:255 Len:8 Data:1,FC,D0,4E,A,0,FF,FF
PGN 127488 send failed
18:09:22.342 -> 67836 : Pri:3 PGN:127488 Source:22 Dest:255 Len:8 Data:0,F8,43,FF,FF,7F,FF,FF
18:09:22.342 -> PGN 129026 send failed
18:09:22.342 -> 67842 : Pri:3 PGN:129026 Source:22 Dest:255 Len:8 Data:1,FC,D0,4E,A,0,FF,FF
PGN 127488 send failed
18:09:22.508 -> 68004 : Pri:3 PGN:127488 Source:22 Dest:255 Len:8 Data:0,F8,43,FF,FF,7F,FF,FF
18:09:22.541 -> PGN 129026 send failed
18:09:22.541 -> 68009 : Pri:3 PGN:129026 Source:22 Dest:255 Len:8 Data:1,FC,D0,4E,A,0,FF,FF
PGN 127488 send failed
18:09:22.673 -> 68172 : Pri:3 PGN:127488 Source:22 Dest:255 Len:8 Data:0,F8,43,FF,FF,7F,FF,FF
18:09:22.706 -> PGN 129026 send failed
18:09:22.706 -> 68177 : Pri:3 PGN:129026 Source:22 Dest:255 Len:8 Data:1,FC,D0,4E,A,0,FF,FF
PGN 127508 send failed
18:09:22.839 -> 68313 : Pri:6 PGN:127508 Source:22 Dest:255 Len:8 Data:1,F8,4,FF,7F,FF,FF,1
18:09:22.839 -> PGN 127508 send failed
18:09:22.839 -> 68318 : Pri:6 PGN:127508 Source:22 Dest:255 Len:8 Data:0,64,5,A,0,FF,FF,43
18:09:22.839 -> PGN 130316 send failed
18:09:22.839 -> 68326 : Pri:5 PGN:130316 Source:22 Dest:255 Len:8 Data:1,1,4,74,7D,4,84,B
18:09:22.839 -> PGN 130312 send failed
18:09:22.839 -> 68334 : Pri:5 PGN:130312 Source:22 Dest:255 Len:8 Data:1,1,4,F2,72,23,73,FF
18:09:22.883 -> PGN 130311 send failed
18:09:22.883 -> 68343 : Pri:5 PGN:130311 Source:22 Dest:255 Len:8 Data:1,4,F2,72,B6,35,F6,3
18:09:22.883 -> PGN 130310 send failed
18:09:22.883 -> 68352 : Pri:5 PGN:130310 Source:22 Dest:255 Len:8 Data:1,C1,70,95,74,F6,3,FF
18:09:22.883 -> PGN 127513 send failed

```

ttlappalainen, 2019-02-08:
This means that you hardware can not communicate with bus. The buffers on chip and then on library get full, since no packet goes to the bus and library finally reports error. Since the library reports "send failed", it has managed to open communication between MCP2515 chip, but data does not move forward from chip. TemperatureMonitor does not define interrupt for mcp, so it will run in non interrupt mode as default. If you like you can test does it help anything to add also
#define N2k_CAN_INT_PIN xx
where xx is pin where you interrupt line is connected. That should no help anything, since I expect you have problem MCP2515 chip - tranceiver broker, wrong wireing, no terminator. Hopefully you do not have jumper on J3.

pushkalkatara, 2019-02-08:
Thanks for the prompt response, I'll debug with the wiring and the transceiver.