ttlappalainen/NMEA2000 #135: missing function in lib; no member CANRaw::getNumRxBoxes()

OlliHenri, 2019-02-04:
I always run in problems compiling example: DataDisplay2.ino or any function receiving NMEA2000.

I fixed it as follows:

There is a missing function / error in actual GITHUB versions of librarys (2019 02 03) :

NMEA2000_due.cpp

and

due_can.cpp

in setup() we call open() that will, besides other things setup the filters but it can´t bcs there is no member CANRaw::getNumRxBoxes() in due_can.cpp !!

I tried to write one by using CANRaw::findFreeRXMailbox() as example,

/**
* \brief Find unused RX mailbox adapted to NMEA2000_due and return its number
* without this function Arduino says:
* error: 'class CANRaw' has no member named 'getNumRxBoxes'
*/
int CANRaw::getNumRxBoxes() {
    for (int c = 0; c < 8; c++) {
        if (mailbox_get_mode(c) == CAN_MB_RX_MODE) {
            if (mailbox_get_id(c) == 0) {
                return c;
            }
        }
    }
    return -1;
}

but my version is not really working, but at least the code compiles now.

I solved it by manually setup the filters by after open() calling :

 //By default there are 7 mailboxes for each device that are RX boxes
  //This sets each mailbox to have an open filter that will accept extended
  //or standard frames
  int filter;
  //extended
  for (filter = 0; filter < 3; filter++) {
  Can0.setRXFilter(filter, 0, 0, true);
  Can1.setRXFilter(filter, 0, 0, true);
  }

Now it works!

- Oliver -

ttlappalainen, 2019-02-04:
getNumRxBoxes is on line 276 on due_can.h. Did you download it from my github or somewhere else.

Tur-Thure, 2021-09-24:
I try to compile example ActisenseListenerSender in NMEA2000 downloaded from Github. And I get the following error below. What am I doing wrong?

C:\Users\ULundqvist\Documents\Arduino\libraries\NMEA2000_due\NMEA2000_due.cpp: In member function 'virtual bool tNMEA2000_due::CANOpen()':
C:\Users\ULundqvist\Documents\Arduino\libraries\NMEA2000_due\NMEA2000_due.cpp:87:39: error: 'class CANRaw' has no member named 'getNumRxBoxes'
   for (mailbox = 0; mailbox < CANbus->getNumRxBoxes()-1; mailbox++) {
                                       ^
C:\Users\ULundqvist\Documents\Arduino\libraries\NMEA2000_due\NMEA2000_due.cpp:91:28: error: 'class CANRaw' has no member named 'getNumRxBoxes'
   for (; mailbox < CANbus->getNumRxBoxes(); mailbox++) {
                            ^
exit status 1
Error compiling for board Arduino Due (Native USB Port).