ttlappalainen/NMEA2000 #138: errors building ActisenseListner.

yNosGR, 2019-04-23:
Please forgive me if this is an environment issue, I'm still learning.
I've been beating on this for a few days and am unable to make headway. I am getting these errors when I try to build any of the examples:
`
/ynos/Arduino/libraries/NMEA2000_mcp-master/NMEA2000_mcp.cpp: In member function 'virtual bool tNMEA2000_mcp::CANSendFrame(long unsigned int, unsigned char, const unsigned char*, bool)':
/ynos/Arduino/libraries/NMEA2000_mcp-master/NMEA2000_mcp.cpp:98:24: error: 'SREG' was not declared in this scope
     uint8_t SaveSREG = SREG;   // save interrupt flag
                        ^
/ynos/Arduino/libraries/NMEA2000_mcp-master/NMEA2000_mcp.cpp:101:11: error: 'cli' was not declared in this scope
       cli();   // disable interrupts
           ^
/ynos/Arduino/libraries/NMEA2000_mcp-master/NMEA2000_mcp.cpp: In member function 'virtual bool tNMEA2000_mcp::CANGetFrame(long unsigned int&, unsigned char&, unsigned char*)':
/root/Arduino/libraries/NMEA2000_mcp-master/NMEA2000_mcp.cpp:167:26: error: 'SREG' was not declared in this scope
       uint8_t SaveSREG = SREG;   // save interrupt flag
                          ^
/ynos/Arduino/libraries/NMEA2000_mcp-master/NMEA2000_mcp.cpp:168:11: error: 'cli' was not declared in this scope
       cli();   // disable interrupts
           ^
exit status 1
Error compiling for board Generic STM32F103C series.`

Here are the options I've added... `
#define N2K_SOURCE 1
#define N2k_SPI_CS_PIN PB0
#define N2k_CAN_INT_PIN PB1
#define USE_MCP_CAN_CLOCK_SET 8
`
I am able to build the examples in your CAN_BUS_Shield repo.
Thank you for your time.

ttlappalainen, 2019-04-23:
Please download NMEA2000_mcp library again. I have forgot to update fix to GitHub.

yNosGR, 2019-04-23:
After finding [this](https://github.com/ttlappalainen/NMEA2000/issues/28), I did insert 
```
class SREGemulation
{
  public:
    operator int () const attribute((always_inline)) {
      uint32_t primask;
      asm volatile("mrs %0, primask\n" : "=r" (primask)::);
      if (primask) return 0;
      return (1<<7);
    }
    inline SREGemulation & operator = (int val) attribute((always_inline)) {
      if (val & (1<<7)) {
        interrupts();
      } else {
        noInterrupts();
      }
      return *this;
    }
} 
extern SREGemulation SREG;
```
But that created this wall of text that I'm trying to work through.
```
In file included from /ynos/Arduino/libraries/NMEA2000-master/src/NMEA2000_CAN.h:131:0,
                 from /tmp/arduino_modified_sketch_14634/ActisenseListener.ino:20:
/ynos/Arduino/libraries/NMEA2000-master/src/MapleIntCompatibility.h:4:17: error: expected ';' at end of member declaration
 operator int () const attribute((always_inline)) {
                 ^
/ynos/Arduino/libraries/NMEA2000-master/src/MapleIntCompatibility.h:4:33: error: expected identifier before '(' token
 operator int () const attribute((always_inline)) {
                                 ^
/ynos/Arduino/libraries/NMEA2000-master/src/MapleIntCompatibility.h:4:48: error: ISO C++ forbids declaration of 'attribute' with no type [-fpermissive]
 operator int () const attribute((always_inline)) {
                                                ^
/ynos/Arduino/libraries/NMEA2000-master/src/MapleIntCompatibility.h:10:43: error: expected ';' at end of member declaration
 inline SREGemulation & operator = (int val) attribute((always_inline)) {
                                           ^
/ynos/Arduino/libraries/NMEA2000-master/src/MapleIntCompatibility.h:10:55: error: expected identifier before '(' token
 inline SREGemulation & operator = (int val) attribute((always_inline)) {
                                                       ^
/ynos/Arduino/libraries/NMEA2000-master/src/MapleIntCompatibility.h:10:70: error: ISO C++ forbids declaration of 'attribute' with no type [-fpermissive]
 inline SREGemulation & operator = (int val) attribute((always_inline)) {
                                                                      ^
/ynos/Arduino/libraries/NMEA2000-master/src/MapleIntCompatibility.h:10:45: error: 'int SREGemulation::attribute(int)' cannot be overloaded
 inline SREGemulation & operator = (int val) attribute((always_inline)) {
                                             ^
/ynos/Arduino/libraries/NMEA2000-master/src/MapleIntCompatibility.h:4:23: error: with 'int SREGemulation::attribute(int)'
 operator int () const attribute((always_inline)) {
                       ^
/ynos/Arduino/libraries/NMEA2000-master/src/MapleIntCompatibility.h: In member function 'int SREGemulation::attribute(int)':
/ynos/Arduino/libraries/NMEA2000-master/src/MapleIntCompatibility.h:11:5: error: 'val' was not declared in this scope
 if (val & (1<<7)) {
     ^
exit status 1
Error compiling for board Generic STM32F103C series.
```

yNosGR, 2019-04-23:
That looks like it took care of the issue!
I still need a blank MapleIntCompatibility.h to compile, but other than that it is fine.
Thank you for your quick response.

closed 2019-04-24 by yNosGR