ttlappalainen/NMEA2000 #139: Multiple classes/functions and missing/stuck data on the bus

virtuvas, 2019-04-26:
hello Timo, another long post from me, apologies!

Testing and finalising configuration on my 2 new engine blackboxes running ESP32S and MCP2562. Each Device sends:

PGN127493, gearbox oil pressure, state (R N F) and gearbox oil temperature
PGN127489, engine coolant pressure, alternator Voltage (and oil temp which I've not implemented yet)
PGN130312, engine EGT (wrongly anyway with the limitations discussed in #137 )

All that lot is now "broken" into three devices:

Transmission: 	Class 50, Function 165
Engine:		Class 50, Function 160
EGT:			Class 75, Function 130

Following the setproductinfo/setdeviceinfo I have:

  NMEA2000.SetForwardStream(&Serial);
  NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly, 15);
  NMEA2000.EnableForward(false); // Disable all msg forwarding to USB (=Serial)

and as normal on the loop I have: 

  NMEA2000.ParseMessage();

Now there are two identical such devices, one for each engine, each has 3 unique serial codes on SetProductInformation and SetDeviceInformation, that's the way I've built my other black boxes anyway. 
Further on the bus I do have a NoLandEngineering RS11 engine gateway pumping the following:
  Engine RPM
  Engine coolant Temp
  Engine oil Pressure
  Engine Boost

 Device appears on the bus as one single device.
  
Also on the bus are various Garmin kit (2XGPSMAP plotters, 3XGMI10, autopilot, vhf, ais, radar, etc) 
There's also a Maretron DSM250.
  
When I boot up the N2K bus, it's gamble if both the above engine black boxes will run, 
often on first boot I only get the stbrd engine one (NEVER the port one btw!). After a couple of reboots both are working, everything sees everything else, all's good - well almost!

1.
If GPSMAP is running (it's powered separately) then no matter what I do, the port engine black box data will not appear. If I turn off the GPSMAP, then on next N2K bus reboot, the port engine black box will be alive.
2.  
Port engine engine coolant pressure and alternator Voltage created by my blackbox values work properly on the GMIs and GPSMAPs BUT on the Maretron the values are stuck solid to the values they picked on system boot, no matter what I do the values wont change. Stbrd engine/black box values are fine and updated regularly.
However, if the NoLandEngineering RS11 is disconnected from the bus on boot all's fine and the Maretron values are updated properly for both engines.
  
Easy answer is that RS11 implementation is wrong, but I somehow doubt it.
Value merging in PGN127489 seems fine as Garmin sees all things correct
Maretron are rather big and serious to have done something silly, so I wonder if I'm doing something wrong here :D
Backbone is fairly long maybe 20m, all devices on proper garmin tees, drop cables short, garmin terminators. FWIW, stbrd engine is the last device on the bus, port engine is just before it on that side of the bus.
  
  Qs:
  
  A.
Is the split up of Class/Function okay for the PGNs sent? That's a grey area for me, do we really need to split all that up into devices, or could I have one device (appearing on the bus for each engine) that sends all 3 PGNs??? If so, what would be the Class/Function of it???
Or should I split that even more (hope not!)
  
  B. 
I've seen sentences with SetMode with a 22 or something? what defines the number and is it important as from my understanding the bus autoassigns these values to the devices. I have both my engine boxes sent this: 
  NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly, 15);
  should I change and give different values to them?

My bus now with this splitting has 19 virtual devices (14 physical) on it if it makes any difference...

ideas welcomed!

cheers

V.

ttlappalainen, 2019-04-28:
Did I understood that you use library multi device feature?

A. Split is OK. On the other hand there is no need to do splitting. Device function/class is part of the device "NAME". That effects priority during address claiming. Also lower source have higher priority. Those has meaning only on very heavy traffic bus.

I have not seen that device function/class would have been handled any other that informative way. MFD:s will still handle all messages from device even function/class is wrong. In your case I think Class 50, Function 160.

Just remembered that I have Alba Combi multi sensor system on my boat and it uses single claa/function code even it sends tank, temperature, engine etc. data.

You can still keep your splitting, if it works. But you can try does it make any difference without splitting.

B. SetMode with parameter 22 tells that it will first try to use address 22. If there is no other using 22, your device will keep it. If there are, they will start to handle (address claiming), who can keep it.

So you could try to start every box with totally different address to avoid address claiming. Does this make any difference?

That parameter is important, since NMEA 2000 requirements says that you should use last address on new boot. This means that you should track your device address changes and save address to e.g. EEPROM and restore it on start. Something like this:
```
uint8_t Source:
...
setup() () {
 ...
  ReadSourceFromEEPROM(Source);
...
  SetMode(...,Source);
}

loop() {
...
  unsigned char src=NMEA2000.GetN2kSource();
  if ( src!=Source ) {
    Source=src;
    SaveSourceToEEPROM(Source);
  }
}

```

If you have multi device in use, you will naturally need to save/restore address for each device. This also may solve your problem, since once they have have static address on your system, devices will not do address claiming on startup. Anyway this interest me and I would like to know will system work right in single device mode or multi device mode with different start - so no address claiming.

virtuvas, 2019-04-29:
> Did I understood that you use library multi device feature?

Timo, thanks for the comments,
yes, I use the multi device feature, have used it in another black box before sending Wind, rudder, tank, temps etc successfully. So no problems there.

> B. SetMode with parameter 22 tells that it will first try to use address 22. If there is no other using 22, your device will keep it. If there are, they will start to handle (address claiming), who [...]

Right, didn't have enough time to mess about many things, so I just managed to go and change address to my 4 devices now on the bus (a teensy3.5, a due and the two ESP32S with the problem)

Glad to report that with addresses from 120-130 (just used the last 3 digits of the serial numbers...) for all four, on firing up the N2K bus, all devices worked straight away.
So address claiming is simply sorted by making sure you have at least on your devices discreet numbers.

I'll do a test with a reader routine to check what the Garmins, the NoLand and the Maretron report/claim when I'm on the boat next on the weekend.

However, the RS11 conflict with the Maretron is still on. Maretron values sent by the ESP32S are frozen on the initial values, whereas values sent by the RS11 on the same PGN show up and change as they should on the Maretron.
GMI10 shows all values and value changes from both devices just fine...
No idea what may be wrong there, but it's v. annoying!

cheers

V.

ttlappalainen, 2019-04-29:
> If GPSMAP is running (it's powered separately) then no matter what I do, the port engine black box data will not appear. If I turn off the GPSMAP, then on next N2K bus reboot, the port engine black box will be alive.
Have you checked e.g. with NMEA Reader that will port engine black box sent data to the bus or will it hang somehow?

> So address claiming is simply sorted by making sure you have at least on your devices discreet numbers.
What do you mean with above?

What are your transmission periods? In the newest NMEA2000.cpp I have listed default priorities and periods for messages. See e.g. line 207 for PGN127489 you should update it in every 500 ms. Also you should take care it will be send in average with this period. It means that if you have some timer value, you should increment it by 500 not set it to millis()+500.
```
NextSentTime=millis()+500; // wrong, delays will cumulate
NextSentTime+=500; // right, next may be sent after 498 ms, if your code has spent lot of time.
```

virtuvas, 2019-04-29:
> Have you checked e.g. with NMEA Reader that will port engine black box sent data to the bus or will it hang somehow?
No, sorry as I've said, didn't have enough time to do proper testing.

> > So address claiming is simply sorted by making sure you have at least on your devices discreet numbers.
> 
> What do you mean with above?
> 

I changed the address of my 4 boxes to:
```
   NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly, 131); //port engine box
   NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly, 132); //stbrd engine box
```
and both these engine boxes report their data and register on the bus straight away. No need to repower the bus or turn off the GPSMAP. So I guess this has sorted this problem.

> What are your transmission periods? In the newest NMEA2000.cpp I have listed default priorities and periods for messages. See e.g. line 207 for PGN127489 you should update it in every 500 ms. Also [...]

right, didn't realise that, just checked:
PGN127489 is 400ms
PGN127493 is 600ms
and
PGN130312 is 500ms

The engineparams routine looks like this:

```
#define EngineParamsUpdatePeriod 400
void EngineParams() {

  static unsigned long EngineParamsUpdated = millis();
  tN2kMsg N2kMsg;

  //oil temperature
  engoilsensor.requestTemperatures();
  float EngineOilTemp = engoilsensor.getTempC(0);

  //Alternator Input, Voltage Divider dropping output to <10mA
  AltVolt_ACDFilter.Filter(analogRead(AltVoltPin));
  AltVoltValue = AltVolt_ACDFilter.Current();
  AltVolt = map(AltVoltValue, 150, 1200, 450, 2400);  //ACCURATE ONE with UNI-T multimeter at home       
  AltVolt = AltVolt * 0.01;                //convert to 2 decimals
  
  //HUBA Control0-4bar, 0.5-3.5V output seawater pressure sender includes
  SeawaterPress_ACDFilter.Filter(analogRead(SeawaterPressurePin));
  SeawaterPressValue = SeawaterPress_ACDFilter.Current();
  SeawaterPressure = map(SeawaterPressValue, 450, 2000, 0, 3770);         
  SeawaterPressure = constrain(SeawaterPressure, 0, 4000);                //4000 is 4bar

  if ( EngineParamsUpdated + EngineParamsUpdatePeriod < millis() ) {
    SetN2kEngineDynamicParam(N2kMsg, EngineInst, N2kDoubleNA, CToKelvin(EngineOilTemp), N2kDoubleNA, AltVolt, N2kDoubleNA, N2kDoubleNA,
                             mBarToPascal(SeawaterPressure), N2kDoubleNA, N2kInt8NA, N2kInt8NA);
    NMEA2000.SendMsg(N2kMsg);

    EngineParamsUpdated = millis();
  }
}
```

I'll try again with 500 on this PGN, what about the other two, transmission and temps, leave them at 1000 or something?

BTW, you must have misstyped something on line 206 of NMEA2000.cpp as:

>                                       case 127237L: // Heading/Track control, pri=2, period=250
>                                       case 127489L: // Engine parameters dynamic, pri=2, period=500

127237 should either be pri=1 and period=250
or pri=2 and period=500 (like the 127489 below it), right?

Do you think that this 400 vs 500 may be causing the issue with the stuck figures on the Maretron? I very much doubt it as the GMI10 happily reports all changes in values without problems

cheers

V.

ttlappalainen, 2019-04-29:
In code I prefer to use:
```
  if ( EngineParamsNextUpdate < millis() ) {
    EngineParamsNextUpdate += EngineParamsUpdatePeriod ;
...

```
Then you avoid one unnecessary sum function on every loop.

> BTW, you must have misstyped something on line 206 of NMEA2000.cpp as:

No I did not. Priority 1 has not been used. Those comments are right. Note that priority has been set by library on call e.g. SetN2kEngineDynamicParam(...)

I thought that if you use slower period like 1-2 sec, it could be the reason. Anyway I prefer to set right default periods - who knows if it helps. If GMI20 is showing all the values, it means that devices are sending them. The strange thing it also that stbd works (instance=1) but port (instance=0) does not.

virtuvas, 2019-05-06:
Timo good morning, plot thickens...

tried to test fit on the boat bus a Bennett trimtab NMEA2000 kit I had bought a long time ago and never bothered to install.
It doesn't work with Garmin or Maretron kit, which is not nice, values flashing 0 and then 100 and loop again, but that's not the main issue (other than I should just buy the sensors and wire them to one of my blackboxes...)

Anyway, the device is identified as:
Lowrance Bennett Trim Tabs
ver. 1.6.0
and WITHOUT a serial number, NOTHING entry is empty!

Funny thing is that it does appear in the bus and even funnier that within 20secs of fitting it in the bus the two Engine black boxes (ESP32S) disappear from the bus and values are emptied from Garmin and Maretron (even the known stuck ones I reported before) They are simply not there.
My other black boxes (teensy and due) are OK, listed on the bus and transmit/receive data as normal as all the other devices on the bus.

Any ideas?

This evening I'll go there and spent the whole evening with a reader logging all sorts of info for the bus devices hoping it will help solving this.

cheers

V.

ttlappalainen, 2019-05-07:
Serial number is on text field and can be empty. It is not necessarily same as unique number on "NAME".

One question is that will ESP32 boxes stop sending. I have not tested ESP32 much. I have been mostly using Teensy series and FlexCan.

Please make simple list of devices you have on the bus. Add transmit/receive PGNs for you own devices.

virtuvas, 2019-05-07:
[MiToS_NMEA2K_Topology.pdf](https://github.com/ttlappalainen/NMEA2000/files/3151412/MiToS_NMEA2K_Topology.pdf)

This is the topology with all the info you requested. I'm at work now, will hopefully check the bus with a reader in the evening and report back.
Note that on my testing the GVD HELM Due is not connected (currently at home working on it, but will add it in the bus today to have a fuller view)
Note2 the Diesel fuel monitoring fuel consumption are still a work in progress with the developer, values reported are OK, don't have any conflicts, shouldn't be causing issues.

thanks for all your help!

V.

virtuvas, 2019-05-10:
Timo,

some logs from yesterday and today. I'm more confused now!

First of all
Source: 225 is GVD FINCONTROL, Teensy 3.5, MCP2562
             Sending custom PGNs: 65521, 65522, 65524
Source: 227 is GVD FINDASH, Due R3 MCP2551 (I know to be replaced with MCP2562)
             Sending custom PGN: 65523
above have to do with Fin stabilisers so need to pass back and forth heal, roll, fin position for port and stbrd fin etc.

Source: 231 is GVD ECU-P TRANS, ESP32S, MCP2562 (Port engine Black Box)
             Sending PGNs: 127493, 127489 and 130312
Source: 232 is GVD ECU-S TRANS same for starboard engine.
the above two are combined single devices reported as Class 50, function 155 (in an attempt to avoid conflict with RS11-didn't work)

Source: 38 is NoLand Engineering RS11
             Sending PGNs: 127488 and 127489

Take things one by one.

My custom PGN65521 says:

GVD.cpp file
```
// Vosper Fin Sender MAIN params
// 
void SetN2kPGN65521(tN2kMsg &N2kMsg,  int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle) {
    N2kMsg.SetPGN(65521L);
    N2kMsg.Priority=6;
    N2kMsg.Add2ByteInt(PortFinPosition);
    N2kMsg.Add2ByteInt(StbrdFinPosition);
    N2kMsg.Add2ByteDouble(PitchAngle, 0.001);
    N2kMsg.Add2ByteDouble(RollAngle, 0.01);

}
```
the matching section on GVD.h says:

```
// *****************************************************************************
// Vosper Fin Sender MAIN params
// Input:
//  - PortFinPosition	   in 8bit
//  - StbrdFinPosition	   in 8bit
//  - PitchAngle	       in radians
//  - RollAngle		       in radians
// Output:
//  - N2kMsg NMEA2000 message ready to be send.

void SetN2kPGN65521(tN2kMsg &N2kMsg,  int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle);

inline void SetN2kFinSenderMAIN(tN2kMsg &N2kMsg,  int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle) {
  SetN2kPGN65521(N2kMsg, PortFinPosition, StbrdFinPosition, PitchAngle, RollAngle);
}

bool ParseN2kPGN65521(const tN2kMsg &N2kMsg,  int8_t &PortFinPosition,  int8_t &StbrdFinPosition, double &PitchAngle, double &RollAngle);
inline bool ParseN2kFinSenderMAIN(const tN2kMsg &N2kMsg,  int8_t &PortFinPosition,  int8_t &StbrdFinPosition, double &PitchAngle, double &RollAngle) {
  return ParseN2kPGN65521(N2kMsg, PortFinPosition, StbrdFinPosition, PitchAngle, RollAngle);
}
```

Now the EBL Reader says:

```
Raw BST Data:
0x93,0x13,0x06,0xF1,0xFF,0x00,0xFF,0xE1,0x55,0xE6,0x0C,0x00,0x08,0xDF,0xFF,0x02,0x00,0x32,0x00,0x70,0x03,0xB0,
Basic Message
PGN:  65521 (0x0FFF1), SRC = 225, DST = 255, Priority = 6, Len = 8, Data = DF FF 02 00 32 00 70 03 

Full decoded Message
NMEA 2000 PGN:  65521 (0x0FFF1)
  Name: Manu. Proprietary single-frame non-addressed 
  Source = 225, Destination = 255
  Priority = 6, Length = 8
  Number Of Fields = 4
  Field 1: NMEA 2000 Manufacturer Code = 2015
  Field 2: Reserved field
  Field 3: Industry Group = 7
  Field 4: Manufacturer defined field(s) = [02.00.32.00.70.03]
```

does that make any sense???
Shouldn't it mention the names I gave to these four fields in my .h and .cpp files?
The fin controlling and displaying works fine, just the sentence doesn't look right to me!

Based on that I today removed the fin controller and dash display (Sources: 225 and 227) from the buss just in case there was some conflict there, nope, same as before. Only way to get reliably both engine black boxes working on the Maretron is to remove the RS11 from the bus.

I'm sending three EBL files:
one from yesterday with fin control/display connected on the bus while the system was running
**alldevicesconnected.ebl**

and two today without the fin stuff on the bus and by starting the logging before firing up the power on the N2K bus.
One with RS11
**nofins_portECUnotworking.ebl**

one without RS11 and both ESP32S working.
**nofins_noRS11_bothECUwork-EGTdoesnt.ebl**

On the testing, port EGT sensor was connected and theoretically working fine, stbrd was disconnected for testing in port so in theory port ECU should respond something like 2  (I'm dividing by 10 to avoid the limit) and stbrd ECU should respond n/a or empty something

Any light you can throw into this mess would be highly appreciated!

cheers

V.

[LoGs.zip](https://github.com/ttlappalainen/NMEA2000/files/3166876/LoGs.zip)

ttlappalainen, 2019-05-10:
For the proprietary messages you have to fill up two first bytes with manufacturer code and industry code. Then you have remaining 6 bytes for your own use. That is for all proprietary messages - also on fastpacket messages.
```
void SetN2kPGN65521(tN2kMsg &N2kMsg,  uitn16_t MfgCode, int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle) {
    N2kMsg.SetPGN(65521L);
    N2kMsg.Priority=6;
    MfgCode&=0x7ff; // Just for sure filter over 11 bits away.
    MfgCode|=0x9800; // Add industry group =4 and 2 reserved bits.
    N2kMsg.Add2ByteUInt(MfgCode);
    N2kMsg.AddByte(PortFinPosition);
    N2kMsg.AddByte(StbrdFinPosition);
    N2kMsg.Add2ByteDouble(PitchAngle, 0.001);
    N2kMsg.Add2ByteDouble(RollAngle, 0.01);
}
```
Note that in your reader handler it should filter out all PGN65521 messages, which does not match to MfgCode you use and industry group = 4

virtuvas, 2019-05-13:
thanks Timo for the suggestion, but not being a programmer I've not managed to get this thing with the MfgCode working as adding this to the .cpp file: 
`void SetN2kPGN65521(tN2kMsg &N2kMsg, uint16_t MfgCode, int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle)`
is not enough, program wont compile. I assume I have to also add it on the 
`bool ParseN2kPGN65521(const tN2kMsg &N2kMsg, uint16_t &MfgCode, int8_t &PortFinPosition,  int8_t &StbrdFinPosition, double &PitchAngle, double &RollAngle)`

But then, do I also need to edit the .h file and add this MfgCode in there as well, if so where? Tried various combinations none worked!
Arduino IDE complains that it's wrong number of arguments and all sorts of other things...

Anyway, furthermore I have some info to add to the general problem of the two ESP32S appearing and registering properly, bear with me:

Devices connected SRC:
3 -> GPS19x (antenna)
4 -> GHC10 (autopilot)
8 -> AIS600
12, 13, 14 -> GMI10s (14 is the only GMI10 I'm using on the lower helm)
23 -> MaretronDSM250
35 -> Airmar DST200 (actually 800 but reports as 200…)
222 -> PC gateway on Due R3
225 -> teensy 3.5 FINCONTROLLER
231 -> ECU port (ESP32S)
232 -> ECU starboard (ESP32S)

All working, BUT:
Port ESP32S  SRC: 231 is flashing in and out of the list of devices on GMI10 as “unknown device”.
NMEA Reader says:
ISO Acknowledgement PGN59392 for:
35->23 0.98interval, PGN of Requested information: 127501
222->23 0.89interval, PGN of Req info: 130826
225->23 1.22interval, PGN of Req info: 130826
231->14 no interval value, PGN of req info: **12780016 ???**
232->23, 0.89interval, PGN of Req info: 130826

what's that? any ideas why it's messing up the PGN there?

also, what is the ISO Address Claim (PGN60928) Unique Number: ISO Identity Number and why the Garmins have various numbers and mine have almost consecutive ones:
225 1523517
231 1523523
232 1523524

final issue:  ISO Address Claim is sent by all of my devices (231 and 232 is similar nothing odd) BUT ISO Request PGN59904 is only sent by the GMI10 (SRC14) to the 231 (port one the one that is on and off the bus) not seen a ISO Request by the Maretron or the GMI to the 232 at all! Which matches with the Network View tab not having any info or 232, just the Transmission (165) Function and D.Instance of 0 as everything else.
Have to stress that all that time all data are sent to the devices without problems so both Engines data are presented on all GMIs and the DSM250, of course RS11 is not connected to the bus at all in these tests.
I'd like to sort out the ESP32S port engine not registering properly first before continuing.

cheers

V

ttlappalainen, 2019-05-14:
Congratulations! You are not programmer and gone this far!

The .h files describes functions for other modules. So the function descriptions in .h and .cpp must match. You mentioned you have PGN 65521 in your own module GVD.h/.cpp. So you have to have function headers same in both .h and .cpp. Note also that you do not necessarily need both
```
void SetN2kPGN65521(tN2kMsg &N2kMsg,  int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle);

inline void SetN2kFinSenderMAIN(tN2kMsg &N2kMsg,  int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle) {
  SetN2kPGN65521(N2kMsg, PortFinPosition, StbrdFinPosition, PitchAngle, RollAngle);
}
```
I used that way in N2kMessages.h, since someone may like to use functions named with PGNs and others with more explanatory name, which I prefer. So you could just have 
void SetN2kFin(tN2kMsg &N2kMsg...
bool ParseN2kFin(const tN2kMsg &N2kMsg...
in GVD.h/.cpp and forget inlines. The idea is to have function names descriptive.

Since GVD.h/.cpp is your own module, you do not actually need to deliver MfgCode in function call. You can handle it inside the GVD.cpp module to make call simpler. So define at the beginning after includes on CVD.cpp
`uint16_t MfgCode=2046; // set here same code as you use on SetDeviceInformation _ManufacturerCode
`
Then your functions are: 
```
void SetN2kFin(tN2kMsg &N2kMsg,  int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle) {
    N2kMsg.SetPGN(65521L);
    N2kMsg.Priority=6;
    MfgCode&=0x7ff; // Just for sure filter over 11 bits away.
    MfgCode|=0x9800; // Add industry group =4 and 2 reserved bits.
    N2kMsg.Add2ByteUInt(MfgCode);
...
}

bool ParseN2kFin(const tN2kMsg &N2kMsg,  int8_t &PortFinPosition,  int8_t &StbrdFinPosition, double &PitchAngle, double &RollAngle) {
  if (N2kMsg.PGN != 65521L) return false;

  int Index = 0;
  uint16_t MsgMfgCode;

  MsgMfgCode=N2kMsg.Get2ByteInt(Index); 
  if ( MsgMfgCode & 0x7ff !=MfgCode ) return false; // Not from Vosper device
  if ( MsgMfgCode & 0xe000 !=0x8000 ) return false; // Check industry code is 4
  ...
}

```

Normally the reason for some own device flashin in and out is caused because NMEA2000.ParseMessages has not been called often enough. Since simple Arduinos does not have multitasking and building all handlers within interrupt was also risky I made it in that way for public library. But then the requirement is that you should not have any long tasks in your loop. E.g. default 1-wire library read function will not work with my library, since it delays loop 700 ms for reading temperatures. During that time other devices on bus may request N2k device information and when they do not get response, the drop it from the list. Then time to time they get and pop device back. So you have to carefully read, what you do on your loop. It should not have any delay calls! If you are not sure, how long other functions uses time you have to measure it:
```
void loop () {
  static unsigned long MaxTime=0;
  static unsigned long NextPrintMaxTime=millis();
  unsigned long StartMeasure=micros();

  // Your other calls here

  unsigned long Elapsed=micros()-StartMeasure;
  if ( MaxTime<Elapsed ) MaxTime=Elapsed;

  if ( NextPrintMaxTime<=millis() ) {
    NextPrintMaxTime+=1000;
    Serial.printf("%lu, Max. off time: %lu\n",millis(),MaxTime);
  }
  NMEA2000.ParseMessages();
}

```

PGN59392 is response for other device requesting some PGN with ISO Request PGN 059904. So you should also see those PGN 059904 requests from source 23 to others. If you do not add your own PGN 059904 handler, library will as default respond NACK for other than internal PGNs. I do not know why Materton is continuously requesting you  130826, which is proprietary PGN. Have you defined that in transmit messages?

ISO Address Claim (PGN60928) Unique Number: ISO Identity Number should be the same as you have set on call SetDeviceInformation. On other devices it can be what ever.

virtuvas, 2019-05-17:
> Congratulations! You are not programmer and gone this far!

well, if someone needs to be congratulated, it's definitely not me but you for helping the rest of us go along the N2K route :)

> Since GVD.h/.cpp is your own module, you do not actually need to deliver MfgCode in function call. You can handle it inside the GVD.cpp module to make call simpler. 

thanks Timo, this worked and revealed a couple of other issues as in a two of these PGNs I had something like 7 or even 8 bytes, so had to revise and reduce variables down to 6bytes (2 doubles 2 int is fine). Now all's working OK DASH and CONTROLLER communicate and present data properly and PGNs are reported correctly on the Actisense Reader. So one hurdle done.
> 
> Normally the reason for some own device flashing in and out is caused because NMEA2000.ParseMessages has not been called often enough. Since simple Arduinos does not have multitasking and building all handlers within interrupt was also risky I made it in that way for public library. But then the requirement is that you should not have any long tasks in your loop. E.g. default 1-wire library read function will not work with my library, since it delays loop 700 ms for reading temperatures. During that time other devices on bus may request N2k device information and when they do not get response, the drop it from the list. Then time to time they get and pop device back. So you have to carefully read, what you do on your loop. It should not have any delay calls! 

Considering that the PGNs sent by the two ECU ESP32S boxes are:
PGN127493 and PGN127489 accurately sent every 500ms (according to Actisense Reader)
and PGN130312 every 560something ms (not sure why that's the EGT)
I assume I'm fine.
That's with one wire lib disabled and fixed values for both DS18B20 sensors, no delays anywhere in the code and generally careful checking, no Serial.print(), etc.
I do however have MAX31855 for the EGT sensor running using Software SPI but I don't think that's the problem:

```
#define MAXDO   GPIO_NUM_12  //DO  on MAX31855
#define MAXCS   GPIO_NUM_14  //CS  on MAX31855
#define MAXCLK  GPIO_NUM_27  //CLK on MAX31855
Adafruit_MAX31855 thermocouple(MAXCLK, MAXCS, MAXDO); //init thermocouple
int val;
double thermoTempC;
```

and the routine is:

```
#define EGTUpdatePeriod 500

void EGTParams() {

  static unsigned long EGTUpdated = millis();
  tN2kMsg N2kMsg;
  
  unsigned char SID = 0;
  unsigned char TempInst = EngineInst;
  tN2kTempSource TempSource = N2kts_ExhaustGasTemperature;
  
  if ( EGTUpdated + EGTUpdatePeriod < millis() ) {

    thermoTempC = thermocouple.readCelsius();
    SetN2kTemperature(N2kMsg, EngineInst, TempInst, TempSource, CToKelvin(0.1*thermoTempC), N2kDoubleNA);
    NMEA2000.SendMsg(N2kMsg);
    // if EGT over a limit (to be set!) engine check light on!
    if (thermoTempC > 48) {
      SetN2kEngineDynamicParam(N2kMsg, EngineInst, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA,
                               N2kDoubleNA, N2kDoubleNA, N2kInt8NA, N2kInt8NA, 1);
      NMEA2000.SendMsg(N2kMsg);
    }
    EGTUpdated = millis();
  }
}
```

This flashing issue is getting on my nerves tbh! I wonder if it relates to:

- EPS32S being a crap chip/implementation/etc It's A2D converter is really awful, have to use 11bit as in 12bit it's impossible to use in a large range of values
- Something else is messing with the bus and not letting the PGNs from my ESPs through
- related to the fact that the two ESPs are at the end of the bus (but only 2 and 4m from the bulk of the devices which are at the lower helm (a dozen tees there)

Now, if I leave the bus off power for a few hours (or overnight) next morning on first power up all devices work, confirmed on the last 2-3 days) If I start working, uploading code, plugging things in and out, chances are that turning N2K power off and on again one or both my ESPs will be out as in:

- not appearing at all in the list of devices, no PGNs sent as I'm monitoring the Actisense Reader and obviously no data on the GMI/Maretron
- flash in and out, some (not necessary all!) PGNs visible on Actisense Reader, some data on GMI/Maretron

BTW, on a clean boot that everything works, Maretron displayed data from PGN127489 (which is the only sentence being sent by both my ESP32S and the No-Land Engineering RS11) come up as:

- RS11 display and update normally
- ESP32S display the initial power on values and are locked there, wont move
- 

Note the GMIs at the same time show and update all values without problems.

If I unplug and replug the Maretron, all data (even the ESP32S ones) will show values that update and match the GMI ones, so problem solved by having the Maretron appear on the N2K bus AFTER all the others boot, rings a bell? anything I could do to "speed up" the boot process of the ESPs?

The problem is that Actisense Reader is so much easier to check and follow compared to ESP Reader reading the log, so can only spent time thinking/testing it on the boat only.

> PGN59392 is response for other device requesting some PGN with ISO Request PGN 059904. So you should also see those PGN 059904 requests from source 23 to others. If you do not add your own PGN [...]

No I have not and I'll check if I can disable the Maretron from sending it (doubt it)
BTW, what would be an appropriate propitiatory PGN handler? Any example please?

thanks!

V

virtuvas, 2019-05-21:
hi,

another update, took the maretron at home to test/check a NMEA2000 trim tab indicator.
so system features the rest of the devices (and the RS11)

Now, tried it in 3 consecutive visits to the boat (one each day) and everytime the starboard ESP32S is visible on the bus and the data are visible on the GMI, but the port one is not visible on the list of devices and obviously no data on the GMI from port engine.

If I physically unplug the port ESP32S and plug it again after a minute (or less), then it's visible on the bus and data are visible on the GMI.

Is there another way that I can _FORCE_ the ESP32S to broadcast itself on the bus?
Should I introduce a delay in the Setup() routine before the 
NMEA2000.SetProductInformation ( ... )
NMEA2000.SetDeviceInformation (...)
and the rest of the NMEA2000.  calls?

for reference, my setup routine looks like this:

```
void setup() {

//  ESP32-S is not v.good at ADC at 12-bit, so use 11-bit for smoother output, 
//  check this: https://esp32.com/viewtopic.php?f=12&t=1045
  analogReadResolution(11);  
  Serial.begin(115200);

  if (EngineInst==0) {
   NMEA2000.SetProductInformation("9912131",                      
                                 1210,                             
                                 "GVD ECU-PORT",          
                                 SWVER,
                                 "NodeMCU ESP-32S",      
                                 0x02,                            
                                 0xffff,                         
                                 0xff                            
                                );
   NMEA2000.SetDeviceInformation(9912131, 
                                160,     
                                50,      
                                1210,     
                                4       
                               );  
  } 
  else {
   NMEA2000.SetProductInformation("9912136",       
                                 1210,                                 
                                 "GVD ECU-STBRD",           
                                 SWVER,
                                 "NodeMCU ESP-32S",       
                                 0x02,                                
                                 0xffff,                               
                                 0xff                                  
                                );

   NMEA2000.SetDeviceInformation(9912136, 
                                160,      
                                50,       
                                1210,     
                                4         
                               ); 

  }

  if (EngineInst==0)
    NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly, 231);
  else
    NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly, 236);
  NMEA2000.EnableForward(false); 
  NMEA2000.ExtendTransmitMessages(TransmitMessages);
  NMEA2000.Open();
}
```

and my loop is v.simple:

```
void loop() {
  TransmissionParams();
  EngineParams();
  EGTParams();
  NMEA2000.ParseMessages();
}
```

EngineInst and SWVER is defined above the Setup() to avoid having to edit all that each time I compile for port or stbrd engine:

```
// Engine Instance 0 port, 1 stbrd
unsigned char EngineInst = 0;
const char*  SWVER = "1.51 - 05.2019";
```

Any suggestions (other than sticking a delay(4000000); on the setup :)

V.

ttlappalainen, 2019-05-22:
You can try to NMEA2000.SendDeviceInformation(); in every 30 s. Devices should automatically request information for devices, they receive data and they do not know yet. I had some device, which required NMEA2000.SendDeviceInformation(); every 2 sec to keep my device alive. You can also test NMEA2000.SendProductInformation();

Have you tested to switch instance between two devices? Does problem then move to other device?

This should be also tested so that you start recording for the bus and turn on the devices. Then you should see all devices to inform them with 60928. All MFD:s should the request pdroduct information from others and you should alse see all responses.

Note also that unique number has only 21 bit. Your 9912136 will be cutted to 1523523. So please use unique number as max. 2097150

virtuvas, 2019-05-22:
> You can try to NMEA2000.SendDeviceInformation(); in every 30 s. Devices should automatically request information for devices, they receive data and they do not know yet. I had some device, which [...]

Timo,

NMEA2000.SendDeviceInformation();
fails with: 

`'class tNMEA2000' has no member named 'SendDeviceInformation'
`
tried NMEA2000.SendProductInformation();
which compiles (I added it to the main loop) but then every 3-4secs I had an error message that GPS receiver lost connection (or something along these lines)
Re upped code with this line off and back to normal. 
Am I missing something on the way you use them?

> Have you tested to switch instance between two devices? Does problem then move to other device?
> 
> This should be also tested so that you start recording for the bus and turn on the devices. Then you should see all devices to inform them with 60928. All MFD:s should the request product information from others and you should alse see all responses.

Right, more testing and checking with NMEA Reader, even before uploading new code to the devices:
NONE of my 4 custom boxes seems to be queried by the GMIs (GPSMAP was off btw)
On going into a GMI menu, N2K device list, it immediatelly made the call and the boxes responded.
Must be something wrong by not having these values sent properly in the first place?
Mind all ESP32S, Teensy and the Due had no request before.

> Note also that unique number has only 21 bit. Your 9912136 will be cutted to 1523523. So please use unique number as max. 2097150
thanks, yes changed that to 12121** so under the 21bit limit, uploaded to all my devices, shouldn't have any issue with that.

ttlappalainen, 2019-05-23:
Sorry, the command was NMEA2000.SendIsoAddressClaim();

Note that you have to have timer for both SendIsoAddressClaim or SendProductInformation. If you sent them in every 30 sec, it should not cause any harm in every 3-4 se. Are you sure you did not send it on every loop and fill up the bus?

virtuvas, 2019-05-26:
doh!
you're right, I did some testing adding NMEA2000.SendProductInformation(); in the main loop()...
OK, fixed it by adding the: 
NMEA2000.SendIsoAddressClaim();  
with a 30sec timer.
Seems that my 4devices are working now, although occasionally the port ECU one will be in the flashing in and out mode for hours to no end.

Didn't have time to do extensive logging, will do this week and report. Also need to bring back onboard the Maretron DSM250 and see how the combination performs. 
I think I'll need some type of delay booting when I fire up the NMEA2000 bus from the breaker, or I'll have to try a separate power supply for the ESP32S on the engines but that's going to be very messy and I'd rather avoid it.

BTW, you believe it's acceptable to have both these lines:

```
NMEA2000.SendIsoAddressClaim();  
NMEA2000.SendProductInformation();
```

every 30 seconds for ever? Any side effects?

cheers

V.

virtuvas, 2020-04-22:
closing issue since I've scrapped the ESP32S and I'm back to using teensy3.5 on a custom board. Everything works, all devices show up everytime.

cheers

V.

closed 2020-04-22 by virtuvas