ttlappalainen/NMEA2000 #140: coverter NMEA0183 to NMEA2000

vikont-s, 2019-05-22:
I'm trying to make coverter NMEA0183 to NMEA2000, transfer the CAN bus to Linux PC and see the time and coordinates on GPSD. 
GPSD receives the N2kGNSS packet (PGN 129029: GNSS Positition Data) and even parses it correctly (correct coordinates and time and the STATUS_FIX flag), but cgps only sees time, there are neither coordinates nor STATUS_FIX. What is missing in the example of Demo: NMEA0183 library. NMEA0183 -> NMEA2000 for GPSD to see a normal GPS receiver?

ttlappalainen, 2019-05-23:
What device you used to connect NMEA2000 bus?

I have to test that. I used example on RPi and read data with OpenCPN.

vikont-s, 2019-05-23:
to connect to a computer, I use my project based on STM32F446 - USBtoCAN , on Linux I have ttyACM and use CAN_UTILS to convert to SocketCAN - and then this is the standard interface that everyone understands. The GPSD program has the test_nmea2000 utility and there is a nmea2000 operation log which can be submitted to the GPSD input. There are many more types of messages (among them, of course, there are PGN 129029: GNSS Positition Data), but it alone is not enough.

vikont-s, 2019-05-23:
Indeed, if GPSD is OpenCPN as a GPSD client, then GPSD sends data in NMEA 0183 format and there are coordinates
gpsd:IO: got data:127258:  6: 01x 03x 119x 70x 00x 00x
gpsd:IO: got data:129029: 47: 01x 119x 70x 40x 143x 86x 16x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x
gpsd:IO:                    : 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 01x 00x 00x 00x 00x 00x 00x 00x
gpsd:IO:                    : 00x 00x 01x 00x 00x 00x 00x
gpsd:PROG: Changed mask: {ONLINE|TIME|LATLON|ALTITUDE|STATUS|MODE|DOP|PACKET|REPORT} with unreliable cycle detection
gpsd:PROG: time to report a fix
gpsd:IO: <= GPS (binary tpv) nmea2000://slcan0: $GPRMC,073650,V,0000.0000,S,00000.0000,W,0.0000,0.000,230519,,*3E\x0d\x0a$GPGSA,A,0,,,,,,,,,,,,,0.0,0.0,*1F\x0d\x0a
gpsd:CLIENT: => client(0): $GPRMC,073650,V,0000.0000,S,00000.0000,W,0.0000,0.000,230519,,*3E\x0d\x0a$GPGSA,A,0,,,,,,,,,,,,,0.0,0.0,*1F\x0d\x0a

ttlappalainen, 2019-05-29:
Sorry, I missed your notifications.

For some reason you do not get 129019 frames right. After first frame the rest just shows 0. The first frame contains date and time and the rest other data. That is why you see just 0 on lat/lon.

vikont-s, 2019-05-29:
there are certain positive results
1. In order for CGPS to start displaying coordinates and states, you must at least add a function SetN2kGNSSDOPData

      SetN2kGNSSDOPData(N2kMsg, 1, N2kGNSSdm_3D, N2kGNSSdm_3D, pBD->HDOP, 0, 0);
      pNMEA2000->SendMsg(N2kMsg);

      SetN2kMagneticVariation(N2kMsg, 1, N2kmagvar_Calc, pBD->DaysSince1970, pBD->Variation);
      pNMEA2000->SendMsg(N2kMsg);

there are also recommendations

      //RMC - Recommended Minimum Specific GNSS Data -
      //126992 // System Time
      //127250* Vessel Heading
      //127258 Magnetic Variation
      //129025 Position Rapid update
      //129026 COG & SOG, Rapid update
      //129029 GNSS Position Data
      //129033 Local Time Offset

      //GGA - Global Positioning System Fix Data -
      //126992 System Time
      //129025 Position Rapid update
      //129029 GNSS Position Data
      //129033 Local Time Offset
      //129539 129539 GNSS DOPs

vikont-s, 2019-05-29:
the second
about the NTP
in GPSD driver_nmea0183.c, after receiving a message with time, the function returns the flag retval | = NTPTIME_IS; and the timestamp is updated for NTP
GPSTP driver_nmea2000.c NTPTIME_IS is not encountered anywhere
This leads to the fact that the time for NTP is not updated and there is no possibility to synchronize time.

adding this flag to the message parsing function

static gps_mask_t hnd_129029 (unsigned char * bu, int len, PGN * pgn, struct gps_device_t * session)

session-> newdata.time = getleu16 (bu, 1) * 24 * 60 * 60 + getleu32 (bu, 3) / 1e4;
     mask | = TIME_SET | NTPTIME_IS;

launches time update process for NTP

vikont-s, 2019-05-29:
and about the message with zero coordinates

gpsd:IO: got data:129029: 47: 01x 119x 70x 40x 143x 86x 16x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x
gpsd:IO: : 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 00x 01x 00x 00x 00x 00x 00x 00x 00x
gpsd:IO: : 00x 00x 01x 00x 00x 00x 00x

GPS lay in the room and accordingly lacked coordinates - there is the main thing that I finally saw that the coordinates go somewhere further and this is already good

:-)