ttlappalainen/NMEA2000 #141: Understanding multi arch nmea2000 library

MENIER, 2019-06-07:
I'm trying to use the N2K library but there is no pinning hardware esp32 connecting in document directory.
and just TXD an RXD define
#define ESP32_CAN_TX_PIN GPIO_NUM_5
#define ESP32_CAN_RX_PIN GPIO_NUM_4
1)I'm asking about the interrupt PIN(#define N2k_CAN_INT_PIN 21) for ESP32, it is the same as arduino due no needs?
2)The code it is independent on HARDWARE transceiver chip (or inversed level)?
3) It is good arch of the nmea2000 library?:
`//+----------+----------------------------------------------------+-----------------------+
//| SOFTWARE |                                                    | HARDWARE Transceiver  |
//|          |                  HARDWARE/SOFTWARE                 |         |∖                        |
//|          |          CAN DRIVER ALL ARDUINO COMPATIBLE         |     TXD_| ∖___CANH    |
//|          |                                                    |         |  ∖                    |
//|          |                                                    |     RXD_|  /__CANL    |
//|          |                                                    |         | /           |
//|          |                                                    |         |/            |
//|          |                                                    |  CAN LEVEL ADAPTATOR  |
//+----------+----------------------------------------------------+-----------------------+
//|          |        |               | ESP32(VDD=3.3V)ESP32      | MCP2551(VDD=5V)       |
//|          | C      |               | driver:CAN Thomas Barth   | OR                    |
//+ N        + A      +               +---------------------------+ MCP2562(VDD=5v=>3.3V) +
//| M        |        |    INTERNAL   | DUE(VDD=3.3V)             | OR                    |
//| E        | N      |               | driver:due_can            | sn65hvd23x(VDD=3.3V)  |
//| A        |        |               | TXD:A14 RXD:A15           | OR                    |
//+ 2        +        +               +---------------------------+ sn65hvd34 etc...      +
//| K        | D      |               | Teensy 3.1/3.2 or 3.5/3.6 |                       |
//|          | R      |               | driver:FlexCAN            |                       |
//+          + I      +---------------+---------------------------+                       +
//|          | V      |    EXTERNAL   | ALL arduino with MCP2515  |                       |
//+          + E      +               + SPI (VDD=5v=>.3.3V)       +                       +
//|          | R      |               | driver:NMEA2000_mcp       |                       |
//+----------+--------+---------------+---------------------------+-----------------------+`

MENIER, 2019-06-07:
3) 3) It is good arch of the nmea2000 library?: 
https://github.com/MENIER/nmea2k/blob/master/nmea2kdoc.h

ttlappalainen, 2019-06-08:
On ESP32 you can use any I/O-pin for CAN. Naturally it is bet to select ones, which are reserved for something else you may need like ADC or UART. Just connect selected pins to tranceiver chip and on sw define pins before including NEMA2000_CAN.h as below:
#define ESP32_CAN_TX_PIN GPIO_NUM_18
#define ESP32_CAN_RX_PIN GPIO_NUM_19
#include <NMEA2000_CAN.h>

1) ESP32 has internal CAN so you do not need any interrupt pin definition. It is same with Arduino Due. You need to define interrupt pin, if you use external CAN controller like MCP2515 with e.g. Arduino Mega.

2) Yes. The tranceiver is "passive" chip and does not require any program. 

3) Pretty much like that.

virtuvas, 2019-06-08:
what board have you got?
here is what worked on my NodeMCU ones. Mine didn't work using Timo's suggestion, don't know why, probably done something silly...

https://github.com/ttlappalainen/NMEA2000_esp32/issues/3#issuecomment-443166348