ttlappalainen/NMEA2000 #142: PGN127497  specs?

virtuvas, 2019-06-17:
Timo,

any idea how this PGN is laid out other than the OpenSkipper xml file?
Garmin are happy and show fuel flow using PGN127489 but Maretron DSM250 wants PGN127497 :(
So planning to read the one PGN and pump to the bus the other :)

Any reason you've not added it to your lib? I'm wondering as it looks a bit odd with 9byte info compared to the typical max 8byte of the other PGNs. I wonder how you formulate it!
Have you got anything on that or plans to add to your lib?

cheers

V.

ttlappalainen, 2019-06-17:
It is just, because no-one has not needed it before. Test code by adding below to specified files. Report me, if message is OK so I can update the lib. I tested only that it translates.

N2kMessages.h (below line 611 to have them in order):
```
//*****************************************************************************
// Trip Parameters, Engine
// Input:
//  - EngineInstance           Engine instance.
//  - TripFuelUsed             in litres
//  - FuelRateAverage          in litres/hour
//  - FuelRateEconomy          in litres/hour
//  - InstantaneousFuelEconomy in litres/hour
// Output:
//  - N2kMsg                NMEA2000 message ready to be send.
void SetN2kPGN127497(tN2kMsg &N2kMsg, unsigned char EngineInstance, double TripFuelUsed,
                     double FuelRateAverage,
                     double FuelRateEconomy=N2kDoubleNA, double InstantaneousFuelEconomy=N2kDoubleNA);

inline void SetN2kEngineTripParameters(tN2kMsg &N2kMsg, unsigned char EngineInstance, double TripFuelUsed,
                     double FuelRateAverage,
                     double FuelRateEconomy=N2kDoubleNA, double InstantaneousFuelEconomy=N2kDoubleNA) {
  SetN2kPGN127497(N2kMsg,EngineInstance,TripFuelUsed,FuelRateAverage,FuelRateEconomy,InstantaneousFuelEconomy);
}

bool ParseN2kPGN127497(const tN2kMsg &N2kMsg, unsigned char &EngineInstance, double &TripFuelUsed,
                     double &FuelRateAverage,
                     double &FuelRateEconomy, double &InstantaneousFuelEconomy);
inline bool ParseN2kEngineTripParameters(const tN2kMsg &N2kMsg, unsigned char &EngineInstance, double &TripFuelUsed,
                     double &FuelRateAverage,
                     double &FuelRateEconomy, double &InstantaneousFuelEconomy) {
  return ParseN2kPGN127497(N2kMsg,EngineInstance,TripFuelUsed,FuelRateAverage,FuelRateEconomy, InstantaneousFuelEconomy);
}
```

N2kMessages.cpp (below line 325 to have them in order):
```
//*****************************************************************************
// Trip Parameters, Engine
void SetN2kPGN127497(tN2kMsg &N2kMsg, unsigned char EngineInstance, double TripFuelUsed,
                     double FuelRateAverage,
                     double FuelRateEconomy, double InstantaneousFuelEconomy) {
  N2kMsg.SetPGN(127497L);
  N2kMsg.Priority=2;
  N2kMsg.AddByte(EngineInstance);
  N2kMsg.Add2ByteUDouble(TripFuelUsed,1);
  N2kMsg.Add2ByteDouble(FuelRateAverage, 0.1);
  N2kMsg.Add2ByteDouble(FuelRateEconomy, 0.1);
  N2kMsg.Add2ByteDouble(InstantaneousFuelEconomy, 0.1);
}

bool ParseN2kPGN127497(const tN2kMsg &N2kMsg, unsigned char &EngineInstance, double &TripFuelUsed,
                     double &FuelRateAverage,
                     double &FuelRateEconomy, double &InstantaneousFuelEconomy) {
  if (N2kMsg.PGN!=127497L) return false;

  int Index=0;

  EngineInstance=N2kMsg.GetByte(Index);
  TripFuelUsed=N2kMsg.Get2ByteUDouble(1,Index);
  FuelRateAverage=N2kMsg.Get2ByteDouble(0.1,Index);
  FuelRateEconomy=N2kMsg.Get2ByteDouble(0.1,Index);
  InstantaneousFuelEconomy=N2kMsg.Get2ByteDouble(0.1,Index);

  return true;
}
```

NMEA2000.cpp to line 208 (between case 127489L: and case 127506L:)
`                                      case 127497L: // Trip Parameters, Engine, pri=5, period=1000
`

virtuvas, 2019-06-17:
impressively quick as always Timo 👍 

works, rather compiles fine no problems, added it to my code, Maretron is not impressed, shows nothing in all available options:
Total Fuel Economy, Total fuel Rate, Total Fuel Consumption as well as the Flow Rate :(

Struggling with time at the moment, will check and make sure NMEA Reader picks up the right PGNs during the week and report back.

cheers

V.