ttlappalainen/NMEA2000 #146: USE

LoetLuemmel, 2019-07-23:
Dear all I have a CAN-Transceiver board with 8MHz crystal connected to an Arduino Mega.

There are recommendations to set the crystal frequency with the following setting:

#define USE_MCP_CAN_CLOCK_SET 8

Does this also apply to this project? At first glance I couldn‘t find refernce to such setting.
On http://shelvin.de/das-nmea2000-netzwerk-etwas-zum-einstieg/ I‘ve learned that the crystal needs to be changed to 16MHz?

Thanks,
-Peter

ttlappalainen, 2019-07-24:
For mcp_can library defaults to 16 MHz setting, so for tranceiver boards with 16 MHz chrystal, you do not need to use any setting. For 8 MHz chrystal use:
#define USE_MCP_CAN_CLOCK_SET 8
#include <NMEA2000_CAN.h"

LoetLuemmel, 2019-07-29:
Thank you very much!
Runs like a charm now.
It failed on Windows based Arduino, but on a Mac live is good!

Fair winds!

closed 2019-07-29 by LoetLuemmel