ttlappalainen/NMEA2000 #148: Simrad Rudder reference PGN 127245

mrbubble62, 2019-07-27:
In case anyone is still looking for Navico (Simrad) rudder reference PGN 127245
Simrad RF25N
Rudder Instance = 255
Direction Order = 7 !
Angle Order = Not Available

Example:
SetN2kRudder(N2kMsg, position * DEG2RAD, 255, N2kRDO_Navico, N2kDoubleNA);

N2KMessages.h - line 176
enum tN2kRudderDirectionOrder {
                            N2kRDO_NoDirectionOrder=0,
                            N2kRDO_MoveToStarboard=1,
                            N2kRDO_MoveToPort=2,
                            N2kRDO_Navico=7
                          };

ttlappalainen, 2019-07-28:
In generally 1 in all bits means "Not Available". As far as I have understood, in direction order there should be 0=N2kRDO_NoDirectionOrder instead of NA, but seems that all does not do that. Anyway your N2kRDO_Navico=7 should be N2kRDO_Unavailable=7.  I have that on the coming version.

mrbubble62, 2019-07-28:
Great, thanks for the explanation.

closed 2019-07-28 by mrbubble62