ttlappalainen/NMEA2000 #149: Using Command Group Function

rmobst, 2019-08-05:
Hi Timo,
I'm using an Arduino Mega with CANdiy-Shield (MCP 2515 and 2562 ) on the NMEA2000-Bus to connect my old analog VDO-Windinstruments (to get AWA and AWS) with the bus and all works fine (using NMEA2000-master from 2016). Now I want to reset the Triplog of an Aimar DST800 Log and the manual told me to use a Command Group Function to set Field #4 to 0 using PGN128275. But I don't know how to use Group Functions (implemented in NMEA2000-master 2019)- is there any example?

ttlappalainen, 2019-08-05:
There is not implementation to use Group function for commanding other devices. The implementation adds basic group functions for responding, if others commands your device. The implemented ones are the minimal requirements for NMEA 2000 certification. They also responds default response like "not supported", in case you do not add your own implementation by inheriting group basic functions. E.g. each PGN, which has  some instance value should provide possibility for other device to modify instance by group function. Unfortunately in that case you should also save changes to keep them after restart.

There is either ready function to build group function command, since it depends of destination. To reset Triplog you use Command Group Function, which meand that for 126208 first field should be set to 1. Next field should be PGN commanded, so here 128275. Next field priority setting is 8=do not change. Field 4 is reserved and has to be set 0xf. Field 5 is number of pairs. Field 6 should be field number for first pais, so here 4. Field 7 is value for 128275 field 4. Here starts the problems, since values depends of destination field size, it is not possible to make common Command Group Function without having full cross reference of all PGNs - well I have some ideas, but not implemented yet. Anyway for resetting Triplog, command build should be something like.
void SetResetTripLog(tN2kMsg &N2kMsg, uint8_t destination) {
  N2kMsg.SetPGN(126208UL);
  N2kMsg.Priority=3;
  N2kMsg.Destination=destination;
  N2kMsg.AddByte(1); // Field 1, Command
  N2kMsg.Add3ByteInt(128275UL); // Field 2, commanded PGN
  N2kMsg.AddByte(0xf0 | 0x8 ); // Field 3, Do not change priority and field 4, reserved
  N2kMsg.AddByte(1); // Field 5, 1 field/value pair
  N2kMsg.AddByte(4); // Field 6, field 4 to be commanded
  N2kMsg.Add4ByteUInt(0); // Field 7, value for first pair =0 for reset distance log.
}

Note that you have to know your triplog source address, which you use as destination on function. And to make it 100% sure you should use tDeviceList and use triplog "Name" to match it and use "Name" for requesting source, since source may change due to address claiming. Source remains same as far as you do not add any devices to the bus.

You can find description of Group Function from https://www.nmea.org/Assets/20140109%20nmea-2000-corrigendum-tc201401031%20pgn%20126208.pdf

Note also that you test use of group function with my NMEA Simulator. Select Tools-NMEA 2000 Commands-Group Function-Command

rmobst, 2019-08-06:
Hi Timo, many thanks for your quick and detailed answer - I'll try the command group message next time entering my boat (I'm just some 500 km far away) and wonder about the answer of the log.

closed 2019-08-06 by rmobst