ttlappalainen/NMEA2000 #150: Minimum environment just for e.g. a battery sensor

LoetLuemmel, 2019-09-08:
Is there a possibility to establish a pretty small sensor using just an Arduino Nano for transmitting e.g. just the battery voltage?

I understood that library space requirements won't comfortably even fit on a UNO.
But if I don't need to evaluate all traffic, but just sending messages, there should be a smaller library support necessary.

Are there viable possibilities or do I have to think towards a MEGA 2560 as bare necessity supporting NMEA2k?

Thanks for helping!     J

-Pit

ttlappalainen, 2019-09-08:
There is no sense to maintain library, which does not full fill N2k requirements.

Instead of using Arduino Mega or any Arduino I prefer to go to Teensy 3.2. That is faster, has more memory and takes less power than Mega. You can compile Teensy 3.2 for 24 MHz and it still handles N2k and uses only 13 mA on 12.5 V. Check e.g. http://skpang.co.uk/catalog/teensy-canbus-breakout-board-include-teensy-32-p-1507.html

LoetLuemmel, 2019-09-11:
13 milli amps on a Teensy 3.2, that would do the Trick. -
Thank you Timo!

ttlappalainen, 2019-09-11:
I have not checked what power supply that has, but must have switching PSU to go down to 13 mA. Alsu you have to select 24 MHz for compiling. I used Recom R-785.0-1.0 DC-DC converter and simply added MCP2562 tranceiver for CAN. There is shcema on https://github.com/ttlappalainen/NMEA2000/blob/master/Examples/TeensyActisenseListenerSender/Documents/Teensy_Actisense_listener_sender_schematics.pdf , just leave extra serial-USB converter away. Also remember to cut VUSB-Vin trace on Teensy so that you do not feed it from two sources, when programming. In schema there is jumper, so that I can select power source.

olesenrl, 2019-09-11:
body{font-family:Helvetica,Arial;font-size:13px}Folks,I am sorry of this is off topic.I am looking to incorporate NMEA2000 reception in an IOS app. Is this the right library to consider? Can anyone point to a IOS solution using this or a different solution?Thanks a lot.Regards,Bob

ttlappalainen, 2019-09-12:
Yes or no. For NMEA2000 connection you need some format converter device. There are commercial devices, which connects to N2k bus and converts data to e.g. SeaSmart format. There is also SeaSmart converter example on library for ESP32. E.g. iNavX uses SeaSmart.

IOS can not connect to N2k bus so you have to decide, which app you se and then buy or make suitable converter e.g. with ESP32.