ttlappalainen/NMEA2000 #151: Issues compiling on RPi3 b+

simonebusoe, 2019-09-17:
Hi everybody, 
i'm new here and i'd like to make a huge thanks to @ttlappalainen and @thomasonw  for all the work done.
I'm using N2k library on Arduino uno, due, ESP32 without problems. I've just started to work on RPi 3 and PiCan 2, no problem to make configurations and to candump from Terminal but a lot of problmes trying to compile with c++. 

I've tried to compile on NeatBeans IDE (remote with ssh) but i find problems on including .h files (i'm working on it), so i started with Code::Blocks IDE on Rasbian.

On C::B, no problems to include .h files, but when i try to build the example i've found on NMEA2000_SocketCan repo (or NMEA2000 docs), i found:
- undefined reference to tNMEA2000::tNMEA2000() //(on NMEA2000_SocketCAN.h)
-undefined reference to tNMEA2000::Open()
-undefined reference to tNMEA2000::ParseMessages()

Could you suggest some changes to the code?
`
#include <cstdlib>
#include <stdio.h>
#include <iostream>
#include </NMEA2000.h>

#include <NMEA2000_SocketCAN.h>       // https://github.com/thomasonw/NMEA2000_socketCAN
tNMEA2000_SocketCAN NMEA2000;
tSocketStream serStream;

#include <N2kMsg.h>

using namespace std;
int main(void)
{
    cout << "Starting CAN watching" << endl;

    setvbuf (stdout, NULL, _IONBF, 0);                                          // No buffering on stdout, just send chars as they come.

    NMEA2000.SetForwardStream(&serStream);                                      // Connect bridge function for streaming output.
    NMEA2000.SetForwardType(tNMEA2000::fwdt_Text);                              // Show in clear text (for now)

    if (!NMEA2000.Open()) {
       cout << "Failed to open CAN port" << endl;
       return 1;
   }

    cout  << endl << "CAN started, going to watch it now" << endl;

     while(1) {
         NMEA2000.ParseMessages();                                               // Will send out CAN messages in open text
    }

    return 0;
}
`

Thank you in advance.
Simone

ttlappalainen, 2019-09-17:
You have not loaded NMEA2000 library to the project lib folder. See on https://github.com/ttlappalainen/NMEA2000/blob/master/Examples/NMEA2000ToNMEA0183/N2kToNMEA0183.cbp where all libraries should be loaded.

simonebusoe, 2019-09-17:
Thank you Timo, now it's working

closed 2019-12-04 by simonebusoe