ttlappalainen/NMEA2000 #158: Initialize buffers problem on due3-teensy3.5 combination

virtuvas, 2020-01-21:
good morning Timo,

slowly sorting code, adding features and re-building my 4 custom boxes came up with the following problem when adding a new box to control the generator (adding new code/features)

Due3+7in display (called DASH)
teensy3.5 controlling fin stabilisers
teensy3.5 controlling generator engine (new device!)
all devices, setup properly and working with MCP2562 sending and receiving data from the bus.
Test bus at home has a GMI10 to check and display info (as well as check that the boxes register properly on the bus.

All normal PGNs sent and received fine from all
series of custom sentences going back and forth:
PGN65521, PGN65522, PGN65524, sent from finstab to DASH
PGN65523 sent from DASH to finstab

PGN65525 sent from gencontr to DASH
PGN65526 sent from DASH to gencontr

Now, on the DASH gencontr combination:
If I only have PGN65526 sent (and PGN65525 disabled/commented out) then both devices, work fine, gencontr reads the data (2 int8_t) fine and updates them as per the dash touchscreen actions/button presses.
If however, I try to sent PGN65525 I get on the serial monitor:

A. from the DASH (receiver of this PGN) a few Failed to parse PGN: 65525, then a "Initialize buffers" and THEN ONCE the output from reading the PGN (it's a test comment I've done, same as all other PGN reading routines, just cut/copied) and then for every loop a Failed to parse:

`Failed to parse PGN: 65525
Failed to parse PGN: 65525
Initialize buffers
GenMAIN instr from Generator: 3, Action: 2
Failed to parse PGN: 65525
Failed to parse PGN: 65525
`

B. OK, I can accept that there's something wrong, but at the same time the already working PGN65526 on the teensy now stops working and reports on every loop:
`Failed to parse PGN: 65526
`
searching seems that the only place there's such a Initialize buffers is on can_due.cpp

BTW, to complicate things a bit more, if instead of sending PGN65525 from the teensy I sent say SetN2kTemperature, two way communication is re-established. However if I try another custom sentence, it wont work either.
Tried A LOT editing and testing values (even tried NMEA2000.SetN2kCANSendFrameBufSize(100) ) but haven't found a solution.

I'm sure the teensy and due boxes are fine hardware wise, if I load the fincontroller code to the teensy, the two way communication with custom PGNs works fine between it and the due, there's something odd I'm missing on the generator to DASH communication.

Not much point sending all the code yet, just waiting for any clever idea you may have. I'll try over the course of the day to strip down the problem on two small files I can sent over for checking/testing (I hope you have a due and some time to spare!)

cheers

V.

ttlappalainen, 2020-01-21:
Which code writes "Failed to parse PGN: 65525"? I do not have that on my library.

Have you recorded data on bus and does it look right?

Do you include MfgCode and Industry code on proprietary messages?

And you parse and check them before future handling? Could some other device start to send those messages naturally with its own MfgCode?

virtuvas, 2020-01-21:
Failed to parse was in the parsing routine which I assumed I picked up from your lib, doubt I'd have written that myself :)

`    if (ParseN2kGenSendrMAIN(N2kMsg,GenStatus,GenAction, GenBits)) {
      
        Serial.println("GenMAIN instr from Generator: " + String(GenStatus) ); 
      }
    }
    else {
      OutputStream->print("Failed to parse PGN: "); OutputStream->println(N2kMsg.PGN);
    }
`

After two evenings of messing with code, I grabbed my custom PGNs lib that was on my notebook at the boat, and noticed a minute difference in the way I was checking for my MfgCode on .cpp file Parsing code.
Changed it back and all's working now!

Interesting that if ONLY one custom PGN was on the bus, things worked, once TWO were on the bus, all hell broke loose and nothing was working. Anyway, sorted it now, works fine, thanks once more!

cheers

V.

closed 2020-01-21 by virtuvas