ttlappalainen/NMEA2000 #159: PGN127489  bool flags behaviour Q

virtuvas, 2020-01-27:
Timo,

not sure you can help on this one, but don't know who to ask :)

Finishing my generator motor control box which is measuring oil pressure, coolant temp, egt, rpm, etc.

I want when the coolant temp goes over a limit, or oil pressure under a limit to raise a flag so that I get a warning on all my N2K displays (which are a few...)
OK, I can do that by raising one of the bool say:
`
SetN2kEngineDynamicParam(N2kMsg, EngineInst, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kInt8NA, N2kInt8NA, 0, 1);
NMEA2000.SendMsg(N2kMsg);
`
the problem is that now ALL my displays are beeping like crazy and pressing OK or cancel (according to display) will only momentarily remove the pop up and the sound which will start almost immediately as the loop runs again.

My options seem to be to run that warning/flag for a limited amount of time and live with the noise, or find a way to manually disable that by hitting the right key on each device.
Such a system works for low depth, or drag or high wind warnings on my Garmins GMI10s.
Doesn't work on this case though.

Am I right in thinking that there should be some feedback routine where the GMI10 sends a message to the device to acknowledge the event so that the event stops been broadcasted on the bus? Is there such a thing, if so how?

Else it looks like I have to simply disable this code and never issue warnings as they will only disappear when the reason ceases. Means I can live this warning for EGT which drops v.quickly when you slow down, but not for other events!

cheers

V.

ttlappalainen, 2020-01-28:
I am sorry, but I can not help.

As far as I know bit fields should indicate the current state. So the bit must be on, while you e.g. have high temp on your engine. There is no ack command and it would not make sense, since your motor still has high temp and you are supposed to send your engine status. So the ack status should be handled on MFD. I mean that if you ack warning on MFD, it should keep it as ack until status drops off and rises back again. If status does not change, ack on MFD should keep warning off. If not, then contact your MFD manufacturer to claim unproper behaviour.

It is not right way to send flag only limited time. Some MFD:s may have delay for warning, so they would miss that, if your time is too short. You should send current status! You would not sent RPM once and then turn it off or unavailable.

 Before claiming manufacturer:
- check that you are sending 127498 every 500 ms and not missing any in your logic. Actisense listeners shows send period and it should be 500 ms. If you have timer for message sending use NextSendTime+=500; instead of NextSendTime=millis()+500; Last may have cumulative error.
- Check that you dor not drop bit even for one message. You can test ack by sending hardcoded warning bit and check does ack then work. If it does, then you logic may drop bit for one message causing ack to be cacelled on MFD and new warning.

virtuvas, 2020-01-28:
> I am sorry, but I can not help.
> 
well you did help Timo!
> Before claiming manufacturer:
>
> * check that you are sending 127498 every 500 ms and not missing any in your logic. Actisense listeners shows send period and it should be 500 ms. If you have timer [...]

I made the silly mistake of having ONE routine sending the PGN with all the data collected from sensors, and now on another routine I'd sent once more (again every 500ms) with all fields N2kDoubleNA and just the bits at the end. 
Merged them all by doing the temp/press tests in the original routine and now it works as it should, one hit and it disappears!

thanks yet once more

V.

closed 2020-01-28 by virtuvas