ttlappalainen/NMEA2000 #160: PGN spesification

haavardholm, 2020-02-20:
Hello, I am trying to get into the NMEA PGN format. 

I find from nmea.org or other places the content of pgn, like PGN 129029 :
field 1Sequence ID
field 2Position date
field 3Position time
field 4Latitude
field 5Longitude
field 6Altitude
....

I do however not find more detailed info : How many bits are used to specify each field and how are each field encoded.

Any hints of where to find that info ?

Best regards
Håvard

ttlappalainen, 2020-02-20:
There is ready message builder SetN2kGNSS (or SetN2kPGN129029) in library N2kMessages module.

haavardholm, 2020-02-20:
The routine look good, but how did you figure out that ? Did you have some samples sentences and figured out by testing and trying or was it documented somewhere ?

ttlappalainen, 2020-02-20:
Most messages will be found on OpenSkipper project or canboat project pgn.h https://github.com/canboat/canboat/blob/master/analyzer/pgn.h OpenSkipper N2k data descriptions has been originally inherited from canboat.

It does not explain all what I am doing with tested code on N2kMessages