ttlappalainen/NMEA2000 #161: raymarine autopilot with b&g plotter

joshoreefe, 2020-02-23:
Hi,
I am trying to create a two can adapter bridge between b&g plotter and rm autopilot. Both speak nmea2000 but a different dialect. Most functions do not work. I have in mind to read the plotter commands from one can socket, reformulate the sentences in rm understandable form and pass over to the other socket.
My problem is, I do not know where to start. I've read the big parts of the code, but am still uncertain of the correct approach. I was hoping you might come up with a suitable vision?

Of course this is not an software issue. Sorry, if I am talking on a wrong channel.
Thanks in advance, Josh

ttlappalainen, 2020-02-23:
Yes this is a bit wrong channel. Better would be NMEA2000 shield forum on Arduino side.

Unfortunately I am afraid that even that would not help. Your problem is behind proprietary messages which does not have description available, if no one has done reverse engineering for them. Actually also NMEA2000 is closed protocol and mostly opened to public originally by Kees.

So if no one has yet opened your messages you can do it by yourself, but it requires much work. You need some device to listen network e.g. Teensy with my library example. Also I prefer to have one device for using my Simulator. Then you can send known exact data (like GPS pos) to the bus. Then you need all actual devices so that you can record data, make test which action sends which command etc. Problem is that things happens in bit level, since at beginning you have no idea where some data is.

I did once some hacking for Raymarine pilot and found how to set auto on/off and change cource. Unfortunatelly I do not remember where I put the descriptions.

You do not need two channels. If you find necessary messages, your device can fool to be B&G autopilot and Raymarin controller on same bus. Then B&G plotter would think to communicate with B&G pilot and send commands to it etc.

Anyway the biggest problem is that for doing this hacking you need all devices.

joshoreefe, 2020-02-23:
Thanks for the input Timo. I also thought it would be possible to have them talk on just one bus, but since I happen to have a two channel board I thought it would be perhaps simpler to make the bridge.  I am close to having all the devices needed for this, so I guess it's just a matter of digging deep...

closed 2020-02-23 by joshoreefe