ttlappalainen/NMEA2000 #162: PGN 126208 and tN2kGroupFunctionHandler

tmcadam, 2020-03-04:
Hi

I am interested to use PGN 126208 to change basic settings on my sensors (like read rate or instance id) from another device on the N2K bus. Do you have an example or some some pseudo-code of how to use tN2kGroupFunctionHandler?

ttlappalainen, 2020-03-05:
126208 is also called  "Complex Request Group Function" - I like the name Complex...

e.g Library module N2kGroupFunctionDefaultHandlers has method tN2kGroupFunctionHandlerForPGN126998::HandleCommand, which handles required change for installation description. So your should do the same and inherit handler for 126208 as in that module tN2kGroupFunctionHandlerForPGN126998. Then you register handler with
NMEA2000.AddGroupFunctionHandler(<pointer to your class>) as it does internally for e.g tN2kGroupFunctionHandlerForPGN126996.

As example attached is handler for binary status message, which I have not yet published, since I have not fully validated it yet, but I use it on one system. There is instance change handler, which you can study.

On module N2kGroupFunctionDefaultHandlers tN2kGroupFunctionHandlerForPGN126993::HandleRequest there is sample for changing period, which actually should not be done for 126993, but certification requires it.

[N2kGroupFunctionBinaryStatus.zip](https://github.com/ttlappalainen/NMEA2000/files/4293445/N2kGroupFunctionBinaryStatus.zip)

With NMEA Simulator you can test settings manually. Go to Tools-NMEA2000 Commands...-Group Function-Request or Command to send data. Not that since group function takes fields data, but the fiedl sizes has are specific for each PGN, you have to set on simulator each field size to "Bytes" and its multiplier to "Multiplier". Instance fields has multiplier 1, but e.g. PGN 128267 field 3 "depth offset" has multiplier 0.001.

You can find document for 126208 from https://www.nmea.org/Assets/20140109%20nmea-2000-corrigendum-tc201401031%20pgn%20126208.pdf

atestani, 2021-09-02:
How can I send a PGN126208 to request another device's configuration?

ttlappalainen, 2021-09-02:
What do you mean with configuration? To request PGNs with PGN126208 you simply construct request with required PGN and sent it to device. See document https://www.nmea.org/Assets/20140109%20nmea-2000-corrigendum-tc201401031%20pgn%20126208.pdf

One simple way is to use library tDeviceList in DeviceList module. That gets automatically devices NAME, product information, configuration information and send/receive PGN list. It also tracks device source addresses and you can request if there has been changes. If you listen some device PGNs, right way to do it is to lock it to NAME and request source from device list. Then you check from device list has there been any source changes. If there is, you request your interest device source again with NAME and start to use that.

atestani, 2021-09-02:
I am using the ActisenseListenerSender example in a tester for an N2K
product. NMEAReader is immediately showing the periodic messages being sent
by the unit under test (UUT) but the address claim and
configuration data that can be seen when clicking the Properties tab in the
right pane of NMEAReader takes 30+ seconds to appear after the UUT is
started. I need to verify the Unique ID which is part of the configuration
screen and that is much too long in a production environment

What I am looking for is the SetPGN126908 method in the library or the
equivalent with another name to "force" the configuration information to
appear quickly. I have searched and cannot find it. I'm looking for the Set
method for PGN126208 to send the Configuration request so NMEAReader will
show the configuration data if it exists.

ttlappalainen, 2021-09-02:
With NMEAReader you should use NGT-1 device. ActisenseListenerSender is passive and does not request device information/configuration. If you add DeviceList functionality for it, it will do the requests, but uses ISO Request PGN not 126208.

You still did not specify what do you mean with configuration? PGN 126998?

126208 has too many options to make general function call. Read the manual and build calls for your needs. 

You can also use function SetN2kPGNISORequest to request PGNs.

atestani, 2021-09-02:
OK... thanks

I know I should use an NGT-1 but I cannot, as part of the tester.  This is
why I put a Teensy and MCP2562 into the tester to act like an NGT-1.  It
all works except the long delay until NMEAReader actually displays my
address claim and the configuration data it has requested. My original
(misguided) thought was to have my tester's "NGT-1" send a 126908 to the
UUT and get the configuration data sooner.  I need to experiment which is
the best approach. I may just send my configuration data directly from the
UUT either periodically or right after the address claim rather than wait
for NMEAReader to request it.  This is probably not conforming to the N2K
standard but it is in a tester so it doesn't matter.

Thanks for your clarification and suggestion on this.

ttlappalainen, 2021-09-02:
You could do those things with my NMEA Simulator. There are ready buttons to request NAME, product information and configuration information and selected PGN. Or you can configure it automatically request information for all devices it sees on the bus. Or you can write any message as hex to be sent. It works with ActisenseListenerSender, but does not show messages parsed. For that you need commercial box version from https://acrossoceansystems.com/index.php/nmea2000-nmea0183-simulator/.