ttlappalainen/NMEA2000 #163: PGN 127751 DC Voltage/Current

tmcadam, 2020-03-04:
Hi, I'd like to use this PGN to monitor voltage but can't find any definition for it here, CANBoat or Openskipper. Do you know the details of the PGN? I don't have any sample data. I know the fields but not the field lengths/data types.

1: Sequence ID 
2: Connection Number
3: DC Voltage
4: DC Current
5: NMEA Reserved

ttlappalainen, 2020-03-05:
I have to dig. I have it somewhere noted, but did not add it to library, since it is rarely needed. Why do you need this PGN? Doesn't 127508 work for you?

tmcadam, 2020-03-05:
Thanks for your reply. 

I have a solar charge controller with data output. I'm using 127508 for the battery voltage and load, but I also want see the voltage the voltage and current coming from the solar panel. There is a PGN for dc charging equipment but it doesn't fit well in this situation, so I thought of using 127508 and 127751 together. 

I'm open to suggestions of a better way to represent a solar controller like this on the bus.

ttlappalainen, 2020-03-05:
Which charger? I would like to get ready code for Victron. It can be collected, but have no time.

Combination 127508 for battery and 127751 for solar charger should be OK. With 127751 there can be used PGN 127750 Converter (Inverter/Charger) Status or PGN 127510 Charger Configuration Status

tmcadam, 2020-05-29:
Hi Timo 

Thanks for your response, and sorry for the delay replying. I have an Epever/Epsolar charge controller. It has Modbus/RS485 port for reading data. I have used Victron charge controllers as well and have integrated them with other micro-controllers. It is a very simple integration, if you need sample data from  the Victron I can send you some?

Tom

ttlappalainen, 2020-05-30:
I am going to make reader for multiple Victron MPPT controller. As default it sends current statur - voltage,current,power. If you have code to read some history status, that would be interesting.

tmcadam, 2020-06-02:
I don't know if you can read history from the devices we were using, I think they were called Victron Bluesolar MPPT. It just outputs current values every second as serial data, with the values separated by newline.

virtuvas, 2020-06-02:
> I don't know if you can read history from the devices we were using, I think they were called Victron Bluesolar MPPT. It just outputs current values every second as serial data, with the values separated by newline.

hi,

I do have a Victron MPPT 100/30 and a bluetooth dongle so I can see all the data in my mobile/tablet. For sure the system has a history log of 30days for various data. Not sure if these would be of any use in N2K though...

V.

ttlappalainen, 2020-06-03:
I also have one with internal BT and one external adapter for testing. As virtuvas mentioned, there is 30 days history. History must be on MPPT, since if I move aapter to other device, it shows that ones info. On NMEA2000 there is no PGNs for that. Anyway I would like to collect all data from my 4 MPPT devices with one ESP32 to be viewable in boats WiFi and also make total power available on NMEA2000. Total power can be simply read with serial, but history requires some communication.

mrbubble62, 2020-06-03:
you would probably have more luck using VE.Direct api https://www.sv-zanshin.com/r/manuals/victron-ve-direct-protocol.pdf

ttlappalainen, 2020-06-03:
That describes only basic protocol. You can make deeper connect with Blue Solar hex protocol. With that it should be possible to read history data and also set device settings. http://www.therollinghome.co.uk/wp-content/plugins/download-attachments/includes/download.php?id=267