ttlappalainen/NMEA2000 #164: Alarms and RPM

lescollier, 2020-03-13:
Hello Everyone
                     I am very new to Nema 2000 and slowly finding my way around, it looks like some excellent work has already been achieved.
I am hoping to put a Arduino into my engine compartment to monitor the engine , batteries and exhaust pipe temperature. However I cannot find a PGN with engine revs in it and similarly exhaust temperature, PGN 127508 should be ok for the batteries. Do you know are these missing items present in the Nema2000 spec ??

This where I have been looking 

[image]

 Any help gratefully received

LesC

currentleak, 2020-03-13:
you can look in N2kMessages.cpp
maybe 127488 and 127489

lescollier, 2020-03-13:
ah thank you will do...................

Is there a reference somewhere I have missed which lists which PGN's objects are in which files ??

regards

LesC

lescollier, 2020-03-13:
ah  got it its referred to as Engine speed not engine rpm