ttlappalainen/NMEA2000 #165: Two Nodes on one Network

TyphoonSB, 2020-03-19:
Hello Together,

I have a Simrad GO7 XSR Chart Plotter..... and I have two nodes created by my own on an ESP8266....

1st is sending a Heading Signal
2nd is sending Humanity and Teperaturs

Strange is that the GO7 is not showing both on the correct way if both nodes are connected to the bus (see pictures) Name of Node is most like --------- and both nodes are shown together ....

Code1:
`#include <Arduino.h>

#include <EEPROM.h>
#define USE_MCP_CAN_CLOCK_SET 8
#define N2k_SPI_CS_PIN 15
#include <NMEA2000_CAN.h>
#include <N2kMessages.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#define ONE_WIRE_BUS 2
#include <OneWire.h> 
#include <DallasTemperature.h>
#include <Wire.h>
#include "DFRobot_SHT20.h"

DFRobot_SHT20    sht20;

#ifndef STASSID
#define STASSID "ssid"
#define STAPSK  "password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

const int SPI_CS_PIN = 15;

OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);

#define RapidDataUpdatePeriod 2000 //167 // Some strange periot to cause Slow and rapid to run unsync.

// List here messages your device will transmit.
const unsigned long TemperatureTransmitMessages[] PROGMEM={130316L,0};
const unsigned long HumidityTransmitMessages[] PROGMEM={130313L,0};

#define DEV_TEMP 0
#define DEV_HUMI 1

void setup() {
  
 Serial.begin(115200);
 EEPROM.begin(512);
 
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  int count;
  EEPROM.get(0 ,count);
  Serial.print("Boot Counter: ");
    Serial.println(count);
  
  while (WiFi.waitForConnectResult() != WL_CONNECTED && count <= 10) {
    Serial.println("Connection Failed! Rebooting...");
    count++;
    EEPROM.put(0, count);
    EEPROM.commit();
[... 354 lines cut ...]
  tN2kMsg N2kMsg;

  if ( IsTimeToUpdate(RapidDataUpdated) ) {
    SetNextUpdate(RapidDataUpdated,RapidDataUpdatePeriod);
    
   //                               Kompasswert        Magnetische Abweichung, Ablenkung im Boot
    
    SetN2kMagneticHeading(N2kMsg, 0, DegToRad(azimuth), DegToRad(2.6), DegToRad(0.0));
        
    NMEA2000.SendMsg(N2kMsg);
    

    //Heading+=0.1; if (Heading>=360.0 ) Heading-=360.0;

  }
}`

The nodes are sending the periodic data on the right way .... checked with CanOE.... they are also running stable... but only this identyfing makes trouble.

Pictures:
[http://dl2ksb.de/1.jpg](url)
[http://dl2ksb.de/2.jpg](url)
[http://dl2ksb.de/3.jpg](url)

someone a soulution?

ttlappalainen, 2020-03-20:
Each PGN has its defined period. On NMEA2000.cpp, where PGNs has been listed there is mentioned default period, which you should use. So use 2000 for temp and humidity as you have. For naming RapidData is not, good, since in NMEA2000 point rapid is <200 ms period. Magnetic heading you should send with 100 ms period.

Then one problem is OneWire temperature. As default it does it by blocking, which means 700 ms stop for your code. During that period some messages may not be served and may cause problems. If you use async mode with OneWire, you get stops down to 10 ms, which is OK, but still not good. With ESP32 it is possible to run OneWire on other core, but then you need to do something with ArduinoOTA, since that runs other core. Other solution would be to put NMEA2000.ParseMessages to own task to other core. If you use other core or tasking you should prevent data concurrent access with mutexes.

Also you should check how compass reading stops tasking. If there is any delays, it poisons the loop.

Try first to comment all device access - sensors, sht20 and compass - and just set fixed values to them. Then check does devices appear on bus as they should.

TyphoonSB, 2020-03-20:
Hi Timo,

thanks for your great project, and your support.

I tryed to comment everything out, without no result. In my opinion the OTA Stuff is interupting /slowing down the loop.
My problem is that I sealed the nodes with epoxy .... if I remove the OTA Stuff, I will not be able to reupload the software.
So I need to build a new test node first, to check if the OTA code is the problem.

btw: offtopic do you have a dbc file for the NMEA2k messages?!

ttlappalainen, 2020-03-23:
What do you meant with dbc file? I do not have any more information than available on network.

TyphoonSB, 2020-03-24:
dbc is a format which is used for CAN busses, to describe the messages, the signals and there values...

So I did some more tests today .... I commented out all the communication with sensors, with no result.
I also change the time value .... with no results.
strange is that if the device is displayed on the poltter the Node Adress and the manufacturer are shown correct, rest is -------

I also tested with an Arduino Mega ... same code (without OTA), same heading sensor.... works fine. But when I put a second node (one of the above) to the network, also the Arduino Mega is shown as ---- ----  and data of both is showed togeter under one device ....

TyphoonSB, 2020-03-24:
I think I solved the problem....

I found in your code following...

 NMEA2000.SendProductInformation();
 NMEA2000.SendIsoAddressClaim();

When I send them periodically it seems to work fine.

So TNAKS again for this great libary and your support!!!

ttlappalainen, 2020-03-25:
That may be cure, but does not solve actual problem. ISO address claim and product information should be sent on request and library does that automatically, if you just poll ParseMessages. You do not seem to have interrupt enabled for mcp_can. You should wire interrupt line to 8266 and add also
#define N2k_CAN_INT_PIN xx

If interrupt handling has not been enabled you may loose those requests, which will cause them to disappear. Check also that you do not have any delay(xx) calls anywhere on your code. Like I mentioned you should change one wire to async mode. Below is snap from my old code, where I used sensors with single line. With that code you have max about 10 ms delays, which is not bad, if you have interrupt receiving and sending. Now I normally add a DS2482-100 bus master to system.

In CanBoat and SignalK they have description file for messages. That has been designed on on big structure and does not compile on small Arduinos. That DB also does not include all important information. That is why I originally made it with functions. In my own systems I use a bit different method and have each PGN described on separated structure. In this way program takes memory only for selected PGNs. Then I also use class method, where each PGN has own class. Unfortunately it eats a bit more flash. For the beginners the function way is the easiest to understant - I think.

```
// *****************************************************************************
#define DataInstanceStart 0
#define InvalidInstance 0xff
tTemperatureSensorData SensorsData[] = {
  {DataInstanceStart+0,N2kts_FreezerTemperature,{0x28,0x50,0x0A,0xA1,0x07,0x00,0x00,0xD1,0x00},N2kDoubleNA,0.05,0},
  {DataInstanceStart+1,N2kts_RefridgerationTemperature,       {0x28,0xBA,0x22,0xA0,0x07,0x00,0x00,0x7E,0x00},N2kDoubleNA,-0.77,0},
  {DataInstanceStart+2,N2kts_HeatingSystemTemperature, {0x28,0x75,0xED,0x9F,0x07,0x00,0x00,0x5A,0x00},N2kDoubleNA,-0.87,0},
  {DataInstanceStart+3,N2kts_EngineRoomTemperature, {0x28,0x7B,0xDF,0x9F,0x07,0x00,0x00,0x4E,0x00},N2kDoubleNA,0.05,0},
  {InvalidInstance,N2kts_MainCabinTemperature,{},N2kDoubleNA,0,0}
};
void CheckTemperatures() { 
  static unsigned long lastTempRequest = 0;
  static int  resolution = 12;
  static unsigned long  delayInMillis = 0;
  static int LastSensorIndex = 0;
  double Temperature;

  if ( SensorsData[LastSensorIndex].Instance==InvalidInstance ) {
    LastSensorIndex=0;
    lastTempRequest=0;
    Serial.println("");
  }
  
  if ( lastTempRequest == 0 ) { // Initialization
    sensors.begin();
  
    sensors.setWaitForConversion(false);
    sensors.requestTemperatures();
    delayInMillis = 750 / (1 << (12 - resolution)); 
    lastTempRequest = millis(); 
  }

  if (millis() - lastTempRequest >= delayInMillis) // waited long enough??
  {
    Temperature = sensors.getTempC(SensorsData[LastSensorIndex].OneWireAddress);
    if (Temperature>-120) {
      SensorsData[LastSensorIndex].Temperature=CToKelvin(Temperature+SensorsData[LastSensorIndex].Offset);
      SensorsData[LastSensorIndex].FailCount=0;
      Serial.print(" Temperature ("); Serial.print(SensorsData[LastSensorIndex].Instance); Serial.print("):"); 
      Serial.println(KelvinToC(SensorsData[LastSensorIndex].Temperature), resolution - 8); 
    } else {
      SensorsData[LastSensorIndex].FailCount++;
      if (SensorsData[LastSensorIndex].FailCount++ > MaxSensorFailCount ) {
        SensorsData[LastSensorIndex].Temperature=N2kDoubleNA;
      }
    }
    LastSensorIndex++;
    // immediately after fetching the temperature we request a new sample in the async modus
    sensors.requestTemperatures(); 
    delayInMillis = 750 / (1 << (12 - resolution));
    lastTempRequest = millis(); 
  }
}

```

virtuvas, 2020-03-26:
Timo  & TyphoonSB, sorry to chip in,

I also had issues with my f.long code taking something like 3.5sec to run a full loop and maretron DSM250 display timing out and flashing the values in-out...
Following some testing I managed to drop that down to 800-1000ms but not down to 100 needed for some PGNs.
So using Teensy and P. Stoffregen OneWire lib (there's a fork specifically for ESP chips btw) I managed to get the times down to 100ms easily.

code extract:

```
//temp sensor includes
#include <OneWire.h>                      // using teensy Paul Stoffregen lib
#include <DallasTemperature.h>            // split requestTemperaturesByIndex() to getTempCByIndex() to avoid 750ms delay

// Gearbox oil Temp sensor (DS18B20) includes
// Setup a oneWire instance to communicate with any OneWire devices
const unsigned char GBoxOilTemp = A18;
OneWire gboiltemp(GBoxOilTemp);
// Pass our oneWire reference to Dallas Temperature
DallasTemperature gboilsensor(&gboiltemp);
float TransOilTemp;
float TransOilTempB;
```

setup:

```
    gboilsensor.begin();
    gboilsensor.requestTemperaturesByIndex(0);
```

and on the loop:
```
  TransOilTemp = gboilsensor.getTempCByIndex(0);
  Serial.println("Transmission Oil Temp: " + String(TransOilTemp));
```
considering it really works, anything fundamentally wrong with this Timo?

cheers

V.

ttlappalainen, 2020-03-26:
Something...

For 1-wire temperature sensors you need to request conversion and then wait for 750 ms (for 12 bit sensors) before you can read temperatures. The OneWire and DallasTemperature will as default do it in blocking way. Like you would have:
RequestTemperature();
delay(750);
ReadTemperature();
So your code will not do much during delay. In ESP it may do task switching, but not on Teensy or Arduinos. DallasTemperature also offers "async" way as I used in my code example. The idea is just send request and do other things until 750 ms has been spent and you can read data. With that method code will be blocked only about 10 ms at time, which is acceptable in simple temperature sender. In complex system I used the chip I mentioned. I also rewrote its handling with i2c_t3c, which gives interrupt, when transfer is ready. In this way code spends some us with 1-wire.

So read the code above carefully and modify it for your system so you will get much better performance.

TyphoonSB, 2020-03-26:
some short update from my site ....

in my test all cummunication is commented out .... no no interuption from these (till now)

Timo, the definition of interupt pin was a good topic.... so I defined it .... and then the uC runs into an error....

`
ISR not in IRAM!

User exception (panic/abort/assert)
Abort called

>>>stack>>>

ctx: cont
sp: 3ffffe60 end: 3fffffc0 offset: 0000
3ffffe60:  3ffee5bc afbdfa76 00000000 00000000  
3ffffe70:  000000fe 00000000 00000000 00000000  
3ffffe80:  00000000 00000000 00000000 00ff0000  
3ffffe90:  5ffffe00 5ffffe00 3ffee82c 00000000  
3ffffea0:  00000002 00000003 3ffef574 4020c8de  
3ffffeb0:  401005ce 0000001f 3ffee5bc 4020c8f4  
3ffffec0:  00989680 00000001 3ffef574 4020cdf9  
3ffffed0:  00000000 0000001f 3ffef5e0 40206d17  
3ffffee0:  0000000f 00000001 3ffee5bc 00000000  
3ffffef0:  3ffee5c4 3ffef5e0 3ffef574 4020cea8  
3fffff00:  3ffee5c4 3ffef5e0 3ffef574 40207ce9  
3fffff10:  00000000 00000000 00000010 00000001  
3fffff20:  00000000 3fff0c4c 3ffef574 40204bba  
3fffff30:  3fff06bc 00000000 00000000 00000000  
3fffff40:  00000098 00000001 3ffef574 00000000  
3fffff50:  3ffee5c5 3ffee5e0 3ffee5a0 402015c5  
3fffff60:  00000000 0000ffff 000000ff 00000000  
3fffff70:  40213a80 00000000 40201188 4020108c  
3fffff80:  00000002 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffee7cc  
3fffffa0:  3fffdad0 00000000 3ffee78c 4020c4d0  
3fffffb0:  feefeffe feefeffe 3ffe8500 40100ed9  
<<<stack<<<

`

so I changed in NMEA2000_mcp.cpp

`
void Can1Interrupt() {
  pNMEA2000_mcp1->InterruptHandler();
}
`

to 

`
ICACHE_RAM_ATTR void Can1Interrupt() {
  pNMEA2000_mcp1->InterruptHandler();
}
`

now it´s not running into an overflow.... but need to test it on the Boat, to say if my problem is solved now

ttlappalainen, 2020-03-26:
I could add that as define for ESP8266, so default library will compile.

TyphoonSB, 2020-03-26:
that would be fine .... but let me check if it´s work on the NMEA network on the Boat. I will check that later today.

ttlappalainen, 2020-03-26:
Please test with:
```
...
#if defined(ESP8266)
ICACHE_RAM_ATTR void Can1Interrupt();
#else 
void Can1Interrupt();
#endif
...
//*****************************************************************************
#if defined(ESP8266)
ICACHE_RAM_ATTR void Can1Interrupt() {
#else 
void Can1Interrupt() {
#endif
  pNMEA2000_mcp1->InterruptHandler();
}

```

virtuvas, 2020-03-26:
> Something...
>
> For 1-wire temperature sensors you need to request conversion and then wait for 750 ms (for 12 bit sensors) before you can read temperatures. The OneWire and DallasTemperature will as [...]

thanks Timo,
looks v.similar to the WaitForConversion2  asynchronous example that comes with DallasTemperature library. Will convert my code this way.

V.

TyphoonSB, 2020-03-27:
Hi Timo,

I tryed it out, and it seems to work now. Interupts are triggered, and the node is show as it should.

I created already a pullrequest aginst your mcp lib.

Thanks for solving this problem!

ttlappalainen, 2020-03-27:
Did you remove periodic?
NMEA2000.SendProductInformation();
NMEA2000.SendIsoAddressClaim();

TyphoonSB, 2020-03-27:
Yes I did!