ttlappalainen/NMEA2000 #166: NMEA 2000 Ac Bus  input PGN127503

AntonisPapa, 2020-03-25:
Hello Timo,

I would like to ask you if you have any experience with the AC inputs?
I am trying to make a program to run the information of AC Voltage and Current to the Nmea 2000 network.
Thanks in advance.

ttlappalainen, 2020-03-25:
Would you like to have ready module with ESP32? I have contact. They may sell it as blank module, but then you need also small programmer. The same programmer works with their othe modules. Other option is ready module.

AntonisPapa, 2020-03-25:
I already make a dc battery monitor with arduino due .And now I would like to make an ac monitor only voltage and current in the same board.If you have any idea please let me know.

ttlappalainen, 2020-03-25:
Check the module PZEM-004T-100A

AntonisPapa, 2020-03-25:
Thank you very much for your help.
I would like to ask you something else,are you planing to add the pgn127503 sentences in your library or else you have any idea to start to make a programme for that pgn 127503?

AntonisPapa, 2020-04-23:
Timo please let me know how i could add to your library the pgn127503?
Thanks

Tarakona2850, 2022-01-03:
i want to use this pgn too. 
How could this be solved?
What should be changed in the description ( N2kMessages.h ,  N2kMessages.h , N2kTypes.h ?
Please help me.
Thanks.

Csaba

Tarakona2850, 2022-01-03:
Hello Timo!

We tried but not perfect 😭  :

N2kMessages.h :
						  
//*****************************************************************************

void SetN2kPGN127503(tN2kMsg &N2kMsg, unsigned char Instance, unsigned char NrLines, tN2kACLine Line, tN2kACAcceptability Acceptability, double Voltage, double Current, double Frequency, double BreakerSize, uint32_t RealPower, uint32_t ReactivePower, int8_t PowerFactor);

inline void SetN2kACInputStatus(tN2kMsg &N2kMsg, unsigned char Instance, unsigned char NrLines, tN2kACLine Line, tN2kACAcceptability Acceptability, double Voltage, double Current, double Frequency, double BreakerSize, uint32_t RealPower, uint32_t ReactivePower, int8_t PowerFactor) {
  SetN2kPGN127503(N2kMsg, Instance, NrLines, Line, Acceptability, Voltage, Current, Frequency, BreakerSize, RealPower, ReactivePower, PowerFactor);
}

‐-------------------------------------------------------------------------------
N2kMessages.cpp : 

//*****************************************************************************
// AC Input Status
void SetN2kPGN127503(tN2kMsg &N2kMsg, unsigned char Instance, unsigned char NrLines, tN2kACLine Line, tN2kACAcceptability Acceptability, double Voltage, double Current, double Frequency, double BreakerSize, uint32_t RealPower, uint32_t ReactivePower, int8_t PowerFactor) {
    N2kMsg.SetPGN(127503L);
    N2kMsg.Priority=6;
    N2kMsg.AddByte(Instance);
    N2kMsg.AddByte(NrLines);
    N2kMsg.AddByte((Line&0x03) | ((Acceptability&0x03)<<2) | (0x0f<<4));
    N2kMsg.Add2ByteDouble(Voltage,99);
    N2kMsg.Add2ByteDouble(Frequency,99);
    N2kMsg.Add2ByteDouble(BreakerSize,99);
    N2kMsg.Add4ByteUDouble(RealPower,99);
    N2kMsg.Add4ByteUDouble(ReactivePower,99);
    N2kMsg.AddByte(PowerFactor);
}

‐-------------------------------------------------------------------------

N2KTypes.h :    

enum tN2kACLine {
                            N2kacl_Line1=0,
                            N2kacl_Line2=1,
                            N2kacl_Line3=2,
                            N2kacl_reserved=3
                          };

enum tN2kACAcceptability {
                            N2aca_Bad_Level=0,
                            N2aca_Bad_Frequency=1,
                            N2aca_Being_Qualified=2,
                            N2aca_Good=3
                          };
--------------------------------------------------------------------------------

Unfortunately only the first two fields go through .. then no data. What do I do wrong?

[image]

Happy New Year

Csabi

ttlappalainen, 2022-01-03:
First on parameters provide all physically real values as double. E.g. RealPower is real not integer.

Then why you have 99 as precision? Voltage precision is 0.01, current 0.1, Frequency 0.01, breaker 0.1, power 1,  and power factor 0.01.

You also seem to miss current. It should be:
N2kMsg.Add2ByteUDouble(Voltage,0.01);
N2kMsg.Add2ByteUDouble(Current,0.1);
N2kMsg.Add2ByteUDouble(Frequency,0.01);
N2kMsg.Add2ByteUDouble(BreakerSize,0.1);
N2kMsg.Add4ByteUDouble(RealPower,1);
N2kMsg.Add4ByteUDouble(ReactivePower,1);
N2kMsg.Add1ByteDouble(PowerFactor,0.01);

ttlappalainen, 2022-01-03:
Note also that this is deprecated and should not be used in new desings. Instead J1939-75 AC parameters are recommended.

For data output your picture does not prove anything. It may be that the reader you use does not fully support this PGN. I have it supported on my NMEA Simulator commercial version.

Tarakona2850, 2022-01-03:
Thanks for the help!
I'll test it on a B&G Vulcan plotter tomorrow.

Best regards!
Csabi

Tarakona2850, 2022-01-05:
Thanks!
Working perfectly.
[image]