ttlappalainen/NMEA2000 #167: Small Craft Status PGN frequency?

virtuvas, 2020-03-26:
Timo,

any reason this PGN has NA in period?
how often should I sent that when tabs are moving, any idea?
Do I also have to send it periodically even when tabs are stable and not moving?

tried 2000, 1000, 500, 200 all seem to work on the Maretron DSM250 and the Garmin GMI10, so no clues from there :D
Obviously looks smoother by far on 200 and 500 and 2000 is really slow. Tempted to leave it at 500!

cheers

V.

ttlappalainen, 2020-03-27:
Its period should be 200 ms. If you provide that message, you should send it periodically even you do not have information. If you do not have information, you send N2kInt8NA

virtuvas, 2020-03-27:
thanks, that's what I was doing anyway for the no info.
Maybe you should add the 200 on line 296 of the NMEA2000.cpp?

Two more PGN broadcast frequency questions then Timo:

A. How often can I sent my custom PGNs? (they are short ones)

B. Maretron literature mentions that the EGT custom PGN130823 needs to be sent every 2000. Do you have to add that on your NMEA2000.cpp?

thanks

V.

ttlappalainen, 2020-03-27:
Yes, I noticed that information was missing. There were also some priorities wrong on that list, but they are right on N2kMessages.cpp.

A. There is no definition for custom PGN period. So you use what you need.

B. No information for PGN130823. But you could use 130316 with source N2kts_ExhaustGasTemperature (14)

virtuvas, 2020-03-27:
thanks Timo!

closed 2020-03-27 by virtuvas