ttlappalainen/NMEA2000 #168: How to calibrate any device on N2K bus?

Nanfter, 2020-03-28:
**Hello Mr ttlappalainen.**

How are you?  Thanks very much for making the library. I have been working with your NMEA2000 Library is awesome, congratulations. I want to made a device "tank level" e.g  maretron's tank level adapter.  

The Maretron TLA100 has the following features.
• NMEA 2000® Interface
• Adapts American standard (240-30 ohm) resistive senders to NMEA 2000® Network
• Adapts European standard (10-180 ohm) resistive senders to NMEA 2000® Network
• Can be Calibrated for any Resistive Sender Ranging from 0-300 Ohms or 300-0 Ohms
• Accommodates Irregularly Shaped Tanks with 16 Point Calibration
• 16 Programmable Tank Types Including Fuel, Fresh Water, Waste Water, Live Well
• Programmable Tank Number(s) Up to 16 per Tank Type
• Programmable Tank Capacity

Periodic Data PGNs                  127250      Fluid Level 0.4 Times/Second
Response to Requested PGNs   126464     PGN List (Transmit and Receive) N/A
                                                126996     Product Information N/A
                                                126998     Configuration Information N/A
Protocol PGNs                          059392     ISO Acknowledge N/A
                                                059904     ISO Request N/A
                                                060928     ISO Address Claim N/A
                                                065240     ISO Address Command N/A
                                                126206     NMEA N/A 

My quetion is: 
**is possible do that with your library?  Is there a PGN for calibrate any device?**

ttlappalainen, 2020-03-28:
127250 is Vessel Heading, not tank level. You probably meant 127505.

It is posible to make tank level sender with library. You do not need to follow TLA100 at all. Just make hw to measure tank level and convert it to % of tank capasity and send 127505 periodically every 2500 ms. If you make this for your own, it is simple, since you can measure calibration and hardcode it with table. I have not tested how accurate Arduinos ADC converter is, but with Teensy you can get accurate device with simple hw. 

There is no standard PGN for calibration. Each manufacturer uses their own proprietary messages for calibration and there is not much info about them. The only way to get info is to find if someone has done reverse engineering for them or do it by yourself. But for making reverse engineering you need the real devices.

If you think to sell it, then thinks gets complex. There are some more requirements to fullfill for commercial than to get it running on own of friend boat. Nothing impossible, causes costs.

Nanfter, 2020-03-28:
Hello Mr Timo.
Thank very much for your quickly response.

All information was taken from the manual,  I have tested a sketch of Fluid level is working fine, I did the tank table.

[image]

Thanks you very much for the informations.