ttlappalainen/NMEA2000 #171: How to make a new PGN or add a new PGN to the code?

Nanfter, 2020-04-16:
Hello Timo, 

How are you?

Timo is possible to do a tutorial, how to add new PGN at code?

ttlappalainen, 2020-04-16:
Not really. What PGN you are missing?

There is lot of samples on N2kMessages module. If PGN is some of the common, I can add it to there. If it is proprietary, than best is to make own module for it.

Nanfter, 2020-04-17:
Hello Timo.

I am trying to make my own pgn to configure my devices and I am also studying czone pgn and I want to make it included in the code and be able to send them.

ttlappalainen, 2020-04-17:
For your own PGN:s you can use any PGN on proprietary PGN range. Remember that each proprietary message has at the beginning fields manufacturer code and industry field. When you send messages, you have to provide your manufacturer code and when you read messages, you accept only those, which has your manufacturer code. After those field you can use data how ever you want. See example N2kMaretron module.

Proprietary message ranges are:
61184, single fram addressed
65280U-,65535U single frame global
126720, fast packet addressed
130816-131071, fast packed global

virtuvas, 2020-04-20:
hi,
the following is code that Timo helped me built in order to communicate info between two devices that I've designed and built:

the gvd.h file:

```
#ifndef _GVD_H_
#define _GVD_H_

#include "N2kMsg.h"
#include <stdint.h>

// *****************************************************************************
// Vosper Fin Sender MAIN params
// Input:
//  - PortFinPosition	   in int8_t
//  - StbrdFinPosition	   in int8_t
//  - PitchAngle	       in radians [double]
//  - RollAngle		       in radians [double]
// Output:
//  - N2kMsg NMEA2000 message ready to be send.

void SetN2kPGN65521(tN2kMsg &N2kMsg,  int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle);

inline void SetN2kFinSenderMAIN(tN2kMsg &N2kMsg,  int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle) {
  SetN2kPGN65521(N2kMsg, PortFinPosition, StbrdFinPosition, PitchAngle, RollAngle);
}

bool ParseN2kPGN65521(const tN2kMsg &N2kMsg,  int8_t &PortFinPosition,  int8_t &StbrdFinPosition, double &PitchAngle, double &RollAngle);
inline bool ParseN2kFinSenderMAIN(const tN2kMsg &N2kMsg,  int8_t &PortFinPosition,  int8_t &StbrdFinPosition, double &PitchAngle, double &RollAngle) {
  return ParseN2kPGN65521(N2kMsg, PortFinPosition, StbrdFinPosition, PitchAngle, RollAngle);
}
#endif
```

and the gvd.cpp file:

```
#include "GVD.h"
#include <string.h>

uint16_t MfgCode=1210; // set here same code used on SetDeviceInformation _ManufacturerCode

//*****************************************************************************
// Vosper Fin Sender MAIN params
// 
void SetN2kPGN65521(tN2kMsg &N2kMsg,  int8_t PortFinPosition,  int8_t StbrdFinPosition, double PitchAngle, double RollAngle) {
    N2kMsg.SetPGN(65521L);
    N2kMsg.Priority=6;
    MfgCode&=0x7ff; // Just for sure filter over 11 bits away.
    MfgCode|=0x9800; // Add industry group =4 and 2 reserved bits [and the 11bits after it...].
    N2kMsg.Add2ByteUInt(MfgCode);
    N2kMsg.AddByte(PortFinPosition);
    N2kMsg.AddByte(StbrdFinPosition);
    N2kMsg.Add2ByteDouble(PitchAngle, 0.001);
    N2kMsg.Add2ByteDouble(RollAngle, 0.001);

}
//*****************************************************************************
bool ParseN2kPGN65521(const tN2kMsg &N2kMsg,  int8_t &PortFinPosition,  int8_t &StbrdFinPosition, double &PitchAngle, double &RollAngle) {
  if (N2kMsg.PGN!=65521L) return false;
  int Index=0;
  uint16_t MsgMfgCode;
  
  MsgMfgCode=N2kMsg.Get2ByteInt(Index); 
  if ( MsgMfgCode & (0x7ff != MfgCode) ) return false; // Not from Vosper device
  if ( MsgMfgCode & (0xe000 != 0x8000) ) return false; // Check industry code is 4
  PortFinPosition  = N2kMsg.GetByte(Index);
  StbrdFinPosition = N2kMsg.GetByte(Index);
  PitchAngle       = N2kMsg.Get2ByteDouble(0.001, Index);
  RollAngle        = N2kMsg.Get2ByteDouble(0.001, Index);

  return true;
}
```

that's a single frame global setup, enabling 6bytes, in my case two int8 and two doubles

hope it helps!

cheers

V.