ttlappalainen/NMEA2000 #175: Device only showing up on MFD only on boot up

TryingToGITthis, 2020-05-23:
I have a Simrad EVO2 and while I reboot or upload new code, the Simrad sees it in the list. However, if after a minute or two I refresh the list, my device disappears. Speeding up heartbeat interval didn't resolve it either. Other devices on the network show up quickly.

As long as I don't refresh the device list on the Simrad, I do see the data I'm transmitting. 

What do I need to do to refresh the product info in the device list? I do call NMEA2000.SendProductInformation() every two seconds but that didn't help.

virtuvas, 2020-05-24:
had problems like that, ended up using:

NMEA2000.SendIsoAddressClaim();
NMEA2000.SendProductInformation();
NMEA2000.ParseMessages();

every 20secs (you need the ParseMessages()!)
lately and having sorting all varying issues with my code, I noticed I don't really need that anymore!
So as Timo explained, check v.carefully for bits of code that delay loops, examine if messages are indeed sent at right intervals, and go from it

good luck!

V.

ttlappalainen, 2020-05-24:
This is not correct!

You should call NMEA2000.ParseMessages(); in loop as often as possible and avoid longer than 40 ms delays between calls. ParseMessages is like a "task", which should run continuosly without delays. Arduino systems except ESP32 does not have tasks, so one should make them. Any other "task" in loop() should be designed so that it does not cause long delay. E.g. default example of DallasTemperature is good example of bad "task", since it stops loop for 700 ms. Instead it should be interrupt driven (possible with Teensy and 1-wire control chip) or at least asynchronous, which lowers delay down to 10 ms.

In your problem first check that your loop runs without delays and calls ParseMessages every time on loop. Make a timer to measure maximum delay time inside your loop.

When you call ParseMessages right in loop, library takes care of right responces to basic bus requests like product information, address claiming etc. If you do not call ParseMessages often enough, library will loose those and cause other systems to loose your device. You need to call SendIsoAddressClaim and SendProductInformation periodically in every e.g. 10 sec ONLY, IF YOU HAVE DEVICES (normally older ones), which does NOT WORK RIGHT on the bus. Either you should not change heartbeat interval.

When you have measured your maximum delay in loop e.g. 20 ms, you need to define receive buffer size. In worst case burst you may get 2.5 frames/ms, so for 20 ms delay you need at least 50 frame receive buffer. I prefer to avoid continuously more than 40 ms delays, since if device does not respond to request fast enough, caller may give up.

Also check that you have 2 120 ohme termination resistor on your bus.

virtuvas, 2020-05-24:
apologies Timo, my mistake, yes, 
I run ParseMessages() in a tight loop, just have the other two in 20sec loop, sorry for confusion!

V.

TryingToGITthis, 2020-05-24:
Thank you. I did check I my loop was VERY slow. It's now under 20ms. I will test it on the boat in a little bit.

I have one resistor, do I need two?
[image]

ttlappalainen, 2020-05-24:
One resistor is OK for testing. In final installation on boat there must be two.

Remember that if you dor not use isolated tranceiver, you should not have your device far from NMEA 2000 power feed point. Otherwise you may cause ground loop and get fancy effecs.

TryingToGITthis, 2020-05-24:
In series or parrallel?

ttlappalainen, 2020-05-24:
Parallel. Between CAN H and CAN L in both end of the main bus.

TryingToGITthis, 2020-05-24:
This good?
[image]

ttlappalainen, 2020-05-24:
Yes, but you must have one resistor on both end of main bus. This means that if your N2k bus goes from aft to bow, you have one resostor on aft and one on bow. Nothing between.

TryingToGITthis, 2020-05-24:
I'm plugging this into an existing network with other devices. There are terminators on all ends. So for a single device, I only need a single resistor, correct?

ttlappalainen, 2020-05-25:
No. On the whole bus system there should be exactly 2 terminator resistors on one on each end of main bus. There shold not be resistors at the end of stubs. See e.g. https://www.wikiwand.com/en/NMEA_2000