ttlappalainen/NMEA2000 #176: Compiling example code in docs for Raspberry PI fails

rcojoe, 2020-06-11:
I'm having trouble compiling the cpp example in the documents. 
The issue I see is undefined references to tNMEA2000, with specific examples of tNMEA2000::Open() and tNMEA2000::ParseMessages() being referenced.
 The code I'm trying to compile is here: 

`/*
* File: main.cpp
* Author: al
*
* Testing for CAN and RPI
*
* See: https://github.com/thomasonw/NMEA2000_socketCAN
*
* Created on February 12, 2017, 2:37 PM
*/
#include <cstdlib>
#include <stdio.h>
#include <iostream>
#include "NMEA2000_CAN.h"
using namespace std;
int main(void)
{
cout << "Starting CAN watching" << endl;
setvbuf (stdout, NULL, _IONBF, 0); // No buffering on stdout, just send chars as they come.
NMEA2000.SetCANPort("can0"); // Select the CAN port @ can0 (default)
// This is new (optional) call to allow you to use a different socketCAN port
NMEA2000.SetForwardStream(&serStream); // Connect bridge function for streaming output.
NMEA2000.SetDebugMode(tNMEA2000::dm_ClearText); // Lets us see the debug messages on the terminal
NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text (for now)
if (!NMEA2000.Open()) {
cout << "Failed to open CAN port" << endl;
return 1;
}
cout << endl << "CAN started, going to watch it now" << endl;
while(1) {
NMEA2000.ParseMessages(); // Will send out CAN messages in open text
}
return 0;
}`

My compiler output is below: 
`g++ main.cpp -I./NMEA2000_socketCAN/ -I./NMEA2000/src -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-6+rpi1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
COLLECT_GCC_OPTIONS='-I' './NMEA2000_socketCAN/' '-I' './NMEA2000/src' '-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=vfp' '-mtls-dialect=gnu' '-marm' '-march=armv6+fp'
 /usr/lib/gcc/arm-linux-gnueabihf/8/cc1plus -quiet -v -I ./NMEA2000_socketCAN/ -I ./NMEA2000/src -imultilib . -imultiarch arm-linux-gnueabihf -D_GNU_SOURCE main.cpp -quiet -dumpbase main.cpp -mfloat-abi=hard -mfpu=vfp -mtls-dialect=gnu -marm -march=armv6+fp -auxbase main -version -o /tmp/ccYrCMeF.s
GNU C++14 (Raspbian 8.3.0-6+rpi1) version 8.3.0 (arm-linux-gnueabihf)
        compiled by GNU C version 8.3.0, GMP version 6.1.2, MPFR version 4.0.2, MPC version 1.1.0, isl version isl-0.20-GMP

GGC heuristics: --param ggc-min-expand=93 --param ggc-min-heapsize=118537
ignoring duplicate directory "/usr/include/arm-linux-gnueabihf/c++/8"
ignoring nonexistent directory "/usr/local/include/arm-linux-gnueabihf"
ignoring nonexistent directory "/usr/lib/gcc/arm-linux-gnueabihf/8/../../../../arm-linux-gnueabihf/include"
#include "..." search starts here:
#include <...> search starts here:
 ./NMEA2000_socketCAN/
 ./NMEA2000/src
 /usr/include/c++/8
 /usr/include/arm-linux-gnueabihf/c++/8
 /usr/include/c++/8/backward
 /usr/lib/gcc/arm-linux-gnueabihf/8/include
 /usr/local/include
 /usr/lib/gcc/arm-linux-gnueabihf/8/include-fixed
 /usr/include/arm-linux-gnueabihf
 /usr/include
End of search list.
GNU C++14 (Raspbian 8.3.0-6+rpi1) version 8.3.0 (arm-linux-gnueabihf)
        compiled by GNU C version 8.3.0, GMP version 6.1.2, MPFR version 4.0.2, MPC version 1.1.0, isl version isl-0.20-GMP

GGC heuristics: --param ggc-min-expand=93 --param ggc-min-heapsize=118537
Compiler executable checksum: 7defdc925cf5fede452fc531d54623d1
COLLECT_GCC_OPTIONS='-I' './NMEA2000_socketCAN/' '-I' './NMEA2000/src' '-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=vfp' '-mtls-dialect=gnu' '-marm' '-march=armv6+fp'
 as -v -I ./NMEA2000_socketCAN/ -I ./NMEA2000/src -march=armv6 -mfloat-abi=hard -mfpu=vfp -meabi=5 -o /tmp/cc3HbZeA.o /tmp/ccYrCMeF.s
GNU assembler version 2.31.1 (arm-linux-gnueabihf) using BFD version (GNU Binutils for Raspbian) 2.31.1
COMPILER_PATH=/usr/lib/gcc/arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/
LIBRARY_PATH=/usr/lib/gcc/arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../:/lib/arm-linux-gnueabihf/:/lib/:/usr/lib/arm-linux-gnueabihf/:/usr/lib/
COLLECT_GCC_OPTIONS='-I' './NMEA2000_socketCAN/' '-I' './NMEA2000/src' '-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=vfp' '-mtls-dialect=gnu' '-marm' '-march=armv6+fp'
 /usr/lib/gcc/arm-linux-gnueabihf/8/collect2 -plugin /usr/lib/gcc/arm-linux-gnueabihf/8/liblto_plugin.so -plugin-opt=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper -plugin-opt=-fresolution=/tmp/ccCFJ5Fy.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc --build-id --eh-frame-hdr -dynamic-linker /lib/ld-linux-armhf.so.3 -X --hash-style=gnu -m armelf_linux_eabi /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/crt1.o /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/crti.o /usr/lib/gcc/arm-linux-gnueabihf/8/crtbegin.o -L/usr/lib/gcc/arm-linux-gnueabihf/8 -L/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf -L/usr/lib/gcc/arm-linux-gnueabihf/8/../../.. -L/lib/arm-linux-gnueabihf -L/usr/lib/arm-linux-gnueabihf /tmp/cc3HbZeA.o -lstdc++ -lm -lgcc_s -lgcc -lc -lgcc_s -lgcc /usr/lib/gcc/arm-linux-gnueabihf/8/crtend.o /usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/crtn.o
/usr/bin/ld: /tmp/cc3HbZeA.o: in function `main':
main.cpp:(.text+0x60): undefined reference to `tNMEA2000::SetDebugMode(tNMEA2000::tDebugMode)'
/usr/bin/ld: main.cpp:(.text+0x84): undefined reference to `tNMEA2000::Open()'
/usr/bin/ld: main.cpp:(.text+0xf8): undefined reference to `tNMEA2000::ParseMessages()'
/usr/bin/ld: /tmp/cc3HbZeA.o: in function `__static_initialization_and_destruction_0(int, int)':
main.cpp:(.text+0x188): undefined reference to `tNMEA2000_SocketCAN::tNMEA2000_SocketCAN(char*)'
/usr/bin/ld: main.cpp:(.text+0x1a0): undefined reference to `tSocketStream::tSocketStream(char const*)'
/usr/bin/ld: main.cpp:(.text+0x1e8): undefined reference to `tSocketStream::~tSocketStream()'
collect2: error: ld returned 1 exit status
`

rcojoe, 2020-06-12:
I understand now this is beyond a simple command line g++ command but it would be nice if there was a "getting started" script to compile one of the examples using g++

Joe

ttlappalainen, 2020-06-12:
I am sorry, but my current interest is mainly on MCU systems like Teensy, ESP32 etc. I tested that library works on RPi by using CodeBlock. Unfortunately I do not know how MCP2515 driver for RPi has been written. The most common problem is that drivers sends all frames to any available "mailbox" on controller , which causes problem that sending NMEA2000 fast packet messages does not always work, since frame order may be wrong.

Getting "undefined reference" is basic linker error, which you get if you have not defined right the libraries.

titio72, 2020-06-13:
Hi Joe, before building the library try to modify the CMakeList of the library adding all the cpp files.
Here's what it looks for me (as I'm lazy I've copied also the NMEA_SocketCAN sources in the same folder):

add_library(nmea2000 STATIC
  N2kMsg.cpp
  N2kStream.cpp
  N2kMessages.cpp
  Seasmart.cpp
  N2kDeviceList.cpp
  N2kGroupFunction.cpp
  N2kGroupFunctionDefaultHandlers.cpp
  N2kMaretron.cpp
  NMEA2000.cpp
  NMEA2000_SocketCAN.cpp
)

ttlappalainen, 2020-06-13:
There is also small problem on using examples directly on RPi. Linux is multitasking system. So when you call NMEA2000.ParseMessages(); in loop, it will eat 25% of RPi resources. pthread_yield() at end of the loop may do the thing, but may just cause extra  by changing between other threads. In example https://github.com/ttlappalainen/NMEA2000/blob/master/Examples/NMEA2000ToNMEA0183/main.cpp I used usleep(100), which radically dropped power usage for that example.

rcojoe, 2020-06-13:
@titio72,  I see that you are using the cmake setup to compile your entire program.  I'm following you on that, and it seems like the path to take to get all the libraries included for the linker.  I'll would give this a try but after reading @ttlappalainen last comment,  I am going to try to make some progress with an esp32 vs additional issues that may come as a result of using a linux system.  I will revisit the linux based system once I get all my prototyping done.
@ttlappalainen After seeing your last comment about the loop issue,  I moved my project over to arduino esp32's and a bare mcp2515 which I had working in about 15 minutes without any additional components, the library works beautifully.  My goal is to make a binary switch example using the code from #128 reference.  Thank you

closed 2020-06-13 by rcojoe