ttlappalainen/NMEA2000 #177: NMEA2000ToWiFiAsSeaSmart

AAKFISHERMANN, 2020-06-23:
Hi,
I´m trying to compile NMEA2000ToWiFiAsSeaSmart code on Atom PlatformIO. Board is ESP32 DEVKITv1.
Buit each time i compile it, i have errors:

- undefined reference to `vtable for tN2kDataToNMEA0183'

- undefined reference to `tN2kDataToNMEA0183::Update()'

These errors occurs in any version os ESPRESSIF32 (1.7.0, 1.8.0 and the last 1.12.4)
Other codes using NMEA2000 library, works fine; But i have a issue with this one: N2kDataToNMEA0183.h

Thanks for any help
regards

ttlappalainen, 2020-06-24:
You are missong file N2kDataToNMEA0183.cpp from your project.

AAKFISHERMANN, 2020-06-24:
Thanks. I changed declaration in PlatformIO from #include "N2kDataToNMEA0183.h" to #include "N2kDataToNMEA0183.cpp" and now it compiles. 
Thanks for your attention.
regards

ttlappalainen, 2020-06-25:
It is not right way to include .cpp.

AAKFISHERMANN, 2020-06-26:
Hi, You are right. I did that only to test. Now, i put files in correct directory (src) and are working well. Thanks.