ttlappalainen/NMEA2000 #186: PGN Conflict

tmsio, 2020-08-06:
Hello Timo, i have a problem,
My  Yamaha outboard, send fuel flow in NMEA network, i have my esp32 that send the same data.
Raymarine Axiom, read yamaha data and not esp32 data.
Axiom don t allow me to choice wich data to display.
There is a way to allow Axiom to read esp32 data?

Another question i want to recive trip fuel used
It correct this:

[CODE]
double litri_trip_nmea() {
  return litri_trip;
  SetN2kEngineTripParameters(N2kMsg, 0, litri_trip_nmea(), N2kDoubleNA, N2kDoubleNA, N2kDoubleNA);
  NMEA2000.SendMsg(N2kMsg);
[/CODE]

Thi is my code:

[CODE]
double Engine0Flow() {
  return lh1;
}

double tank_nmea() {

  return tank_perc;
}

double cap_tank_nmea() {
  return cap_tank;

}
double litri_trip_nmea() {
  return litri_trip;

}

void SendN2kFlow() {

  tN2kMsg N2kMsg;
  SetN2kEngineTripParameters(N2kMsg, 0, litri_trip_nmea(), N2kDoubleNA, N2kDoubleNA, N2kDoubleNA);
  NMEA2000.SendMsg(N2kMsg);
  SetN2kFluidLevel(N2kMsg, 0, N2kft_Fuel, tank_nmea(), cap_serbatoio_nmea());
  NMEA2000.SendMsg(N2kMsg);
  SetN2kEngineDynamicParam(N2kMsg, 0, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, Engine0Flow(), contatore_1, N2kDoubleNA, N2kDoubleNA, N2kInt8NA, N2kInt8NA, false);
  NMEA2000.SendMsg(N2kMsg);
[/CODE]

Thank You regards.
Emiliano

ttlappalainen, 2020-08-07:
Can Axiom read data, you you disconnect yamaha?

tmsio, 2020-08-07:
Hi Timo, Yes it did.
Regards

ttlappalainen, 2020-08-07:
Then I am afraid that there is not much to do. If Axiom does not offer selection possibility for data sources, only way is to disconnect sources. Or you could make a gateway, which has Yamaha on one side and other on other side. Just collect data you need and forward it to other bus. It is possible with library, if you create manually at least for other NMEA2000 object. Now there is easy way for Arduino users just include NMEA2000_CAN.h, which just internally selects "driver" according to MCU and creates NMEA2000 object. For ESP32 it should be possible to add MCP2515 with MCP2551 and make other NMEA2000_Yamaha by using mcp_can library.

tmsio, 2020-08-13:
Think you for you support and quick answerTimo.
Regards