ttlappalainen/NMEA2000 #188: Help on Trip Fuel Used

tmsio, 2020-08-18:
Hello Timo, i have a problem sending Fule trip Used.
At moment i send l/h and fuel remaning in the tank without problem, i would you like also send trip used on my raymarine.

I see data on my actisense but no on Raymarine (i send data every one second).

This is the part of code:

`double Engine0Flow() {
  return lh1;
}

double serbatoio_nmea() {

  return serbatoio_perc;
}

double cap_serbatoio_nmea() {
  return cap_serbatoio;

}
double litri_trip_nmea() {
  return litri_trip;

}

void SendN2kFlow() {

  tN2kMsg N2kMsg;
  SetN2kEngineTripParameters(N2kMsg, 0, litri_trip_nmea(), N2kDoubleNA, N2kDoubleNA, N2kDoubleNA);
  NMEA2000.SendMsg(N2kMsg);
  SetN2kFluidLevel(N2kMsg, 0, N2kft_Fuel, serbatoio_nmea(), cap_serbatoio_nmea());
  NMEA2000.SendMsg(N2kMsg);
  SetN2kEngineDynamicParam(N2kMsg, 0, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, Engine0Flow(), 
 contatore_1, N2kDoubleNA, N2kDoubleNA, N2kInt8NA, N2kInt8NA, false);
  NMEA2000.SendMsg(N2kMsg);
}`

Thank You.

Regards.

Emiliano

ttlappalainen, 2020-08-18:
Hard to say. Are you sure your Raymarine support PGN 127497? Do you see right period on Actisense or is it variating? 

It should not matter, but all your messages has different period. You should have:
- SetN2kEngineTripParameters 1000 ms
- SetN2kFluidLevel 2500 ms
- SetN2kEngineDynamicParam 500 ms

tmsio, 2020-08-19:
Hello Timo, i modified my code:
I send:
Fluel rate in L/H (work whitout problem)
Hour engine  (work whitout problem)
% tank          (work whitout problem)
Trip               (i have to test after this modification)

NMEA Conf:

  NMEA2000.SetN2kCANSendFrameBufSize(250);
  NMEA2000.SetProductInformation(seriale.c_str(), // Manufacturer's Model serial code
                                 100, // Manufacturer's product code
                                 "My Boat",  // Manufacturer's Model ID
                                 "300(30/08/2020)",  // Manufacturer's Software version code
                                 s_ver.c_str() // Manufacturer's Model version
                                );
  // Set device information
  NMEA2000.SetDeviceInformation(password, // Unique number. Use e.g. Serial number.
                                140, 
                                50, 
                                1818 
                               );
  NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly, 22);
  NMEA2000.EnableForward(false); // Disable all msg forwarding to USB (=Serial)
  NMEA2000.ExtendTransmitMessages(TransmitMessages);
  NMEA2000.Open();

main loop:

`if (millis() - time05s >= 500) 
  {
    SendN2kFlow();
    time05s = millis();
  }

  if (millis() - time1s >= tempo)
  {
      SendN2kTrip();
      time1s = millis();
   
  }

  if (millis() - time25s >= 2500) 
  {
    SendN2kTank();
    time25s = millis();
  }

}`

Void;

`double Engine0Flow() {
  return lh1;
}

double serbatoio_nmea() {
  return serbatoio_perc;
}

double cap_serbatoio_nmea() {
  return cap_serbatoio;

}
double litri_trip_nmea() {
  return litri_trip;

}

void SendN2kFlow() {

  tN2kMsg N2kMsg;
  SetN2kEngineDynamicParam(N2kMsg, 0, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, N2kDoubleNA, Engine0Flow(), 
  contatore_1, N2kDoubleNA, N2kDoubleNA, N2kInt8NA, N2kInt8NA, false);
  NMEA2000.SendMsg(N2kMsg);
}

void SendN2kTrip() {
  tN2kMsg N2kMsg;
  SetN2kEngineTripParameters(N2kMsg, 0, litri_trip_nmea(), N2kDoubleNA, N2kDoubleNA, N2kDoubleNA);
  NMEA2000.SendMsg(N2kMsg);
  NMEA2000.SendProductInformation();
  NMEA2000.SendIsoAddressClaim();
}

void SendN2kTank()
{
  tN2kMsg N2kMsg;
  SetN2kFluidLevel(N2kMsg, 0, N2kft_Fuel, serbatoio_nmea(), cap_serbatoio_nmea());
  NMEA2000.SendMsg(N2kMsg);
}

This is Actisense Screen:

[image]

Product info:

[image]
.

It s all correct?

Just another questions, how can set "Manufacturer" in Network View?
It s important to send "Configuration Information"?
If "YES" how i cand do?

Thank You.
Regards.

Emiliano
`

ttlappalainen, 2020-08-25:
Looks OK.

You can not set manufacturer. Unfortunately NMEA2000 is stupid in that point (also in that). You set manufacturer code on SetDeviceInformation and  each registered code has to be hardcoded to all devices. Which also means that if some new manufacturer registers to NMEA organization, it will not appear to old devices, if they fw will not be upgraded.

Small hint. You save a bit on speed, if you have on your loop:
unsigned long nextTime05s=5000; // start sending after 5000 ms
...
```
if ( nextTime05s<millis() ) {
  nextTime05s+=500;
...
}
```
- You save one one calculation. Just have compare.
- By incrementing with period, you will not have cumulative error on send time.

Since millis runs over after 72 days you will have problem then. millis() will then result e.g. 25, so 25-500 > 500 results true and your loop will start to send data in every round until millis() returns 500. It is worse with 2500 period. By using
```
if ( millis()-nextTime05s<LONG_MAX ) {
  nextTime05s+=500;
 ...
}
```
Fixes the problem.

tmsio, 2020-08-25:
Thank You Timo.  :)))))
Regards

Emiliano