ttlappalainen/NMEA2000 #189: prefilter vacuum: tN2kPressureSource fields, support for Vacuum in PGN130314?

virtuvas, 2020-08-25:
good afternoon Timo,

thinking of adding support and/or warnings for clogged prefilters.
My Separs can accept a vacuum gauge, but being in the engine room means I have to get there to monitor them...
So instead thought I could fit a vacuum/pressure sender and pass the data to a teensy.

Now, it seems that I should be able to use PGN130314. Only:
a. supported pressuresources do not include fuel lines, whereas someone on ybw mentioned that option 5. is fuel but you don't seem to list it in N2kTypes.h. Is it something that can be added?
b. and most important, does 130314 support vacuum? Value is unsigned, am I missing something?

I guess same applies to PGN127489, would it be useable for this?

If the answer to the above is NO, I'll have a go at reversing the Maretron FPM100 custom PGN127505 which does support vacuum for filter monitoring (according to the manual at least!).

cheers

V.

ttlappalainen, 2020-08-25:
Now I do not understand? There is N2kps_Filter = 5 and N2kMsg.Add4ByteDouble(ActualPressure,0.1); is signed.
```
enum tN2kPressureSource {
                            N2kps_Atmospheric = 0,
                            N2kps_Water = 1,
                            N2kps_Steam = 2,
                            N2kps_CompressedAir = 3,
                            N2kps_Hydraulic = 4,
                            N2kps_Filter = 5,
                            N2kps_AltimeterSetting = 6,
                            N2kps_Oil = 7,
                            N2kps_Fuel = 8,
                            N2kps_Reserved = 253,
                            N2kps_Error = 254,
                            N2kps_Unavailable = 255
                          };

void SetN2kPGN130314(tN2kMsg &N2kMsg, unsigned char SID, unsigned char PressureInstance,
                     tN2kPressureSource PressureSource, double ActualPressure) {
  N2kMsg.SetPGN(130314L);
  N2kMsg.Priority = 5;
  N2kMsg.AddByte(SID);
  N2kMsg.AddByte((unsigned char) PressureInstance);
  N2kMsg.AddByte((unsigned char) PressureSource);
  N2kMsg.Add4ByteDouble(ActualPressure,0.1);
  N2kMsg.AddByte(0xff); // reserved
}

```

virtuvas, 2020-08-25:
oops!

sorry you don't understand because I obviously have old versions of the lib (Oct 2019!) :
N2kTypes.h with only 5entries and not 12 and 
N2kMessages.cpp where I had ActualPressure=N2kMsg.Get4ByteUDouble(0.1, Index);

will update all 3 computers I'm working with latest libs, sorry!

V.

closed 2020-08-25 by virtuvas