ttlappalainen/NMEA2000 #193: Can't receive CAN frames on Teensy 4.0

skpang, 2020-11-11:
I'm using a Teensy 4.0 with NMEA2000_Teensyx.cpp and NMEA2000_Teensyx.h files.

The example WindMonitor.ino is working and I can see new CAN frames on the CAN analyser once every second.

With DeviceAnalyzer.ino at start up I can see a PGN EEFF and a Heartbeat frame sent out on the analyser.
On the serial debug I see these messages:
Device analyzer is starting up...
  1500 - tNMEA2000_Teensyx::CANSendFrame pgn: 60928, prio:6, src:50, dst:255, data:02, mb:62
Device analyzer started
  1500 - Tx inactive mb:62, prio:6
  1751 - tNMEA2000_Teensyx::CANSendFrame pgn:126993, prio:7, src:50, dst:255, data:60, mb:63
  1751 - Tx inactive mb:63, prio:7

but nothing else. My NMEA2000 has a windspeed sensor, GPS and temperature sensor so I expect these to show up.

I thing tried the same sketch on a Teensy 3.2  and it shows up a list of devices.

Something not right with Teensy 4.0 receiving CAN frames ?

ttlappalainen, 2020-11-12:
Nothing wrong with Teensy 4.0 receiving CAN frames.

I pushed DeviceAnalyzer example fix. Serial.println(pDevice->GetManufacturerInformation()); crashed, when received char pointer is 0. I wonder has that changed on Teensy4 println or was it due to faster process.

skpang, 2020-11-12:
With the fix I can now see the device list.

Thank you for your help.

closed 2020-11-12 by skpang