ttlappalainen/NMEA2000 #195: SetN2kDCBatStatus Ampere decimals

anhorbc, 2020-12-19:
Hi Timo!
First of all I would like to thank you for all your excellent work! I cannot imagine the number of hours you have spent on this. 
I have a question regarding SetN2kDCBatStatus. BatteryCurrent seems to be sent with only one decimal. Do you know if it is a limitation due to the NMEA 2000 standard? I was thinking of modifying N2kMessages.cpp to send more decimals, but I realize that this might not be according to the NMEA 2000 specifications..

/A

ttlappalainen, 2020-12-19:
Yes it is. You can not change N2kMessages.cpp, if you do not find special error there. NMEA2000 data structure is fixed by specifications.

anhorbc, 2020-12-19:
Ok! Kind of figured.. Thanks a lot!