ttlappalainen/NMEA2000 #199: PGN127489 parsing warning flags?

virtuvas, 2021-01-05:
happy new year all,

have managed successfully to set warning flags for low oil pressure, overheat, whatever. That's fine, Garmins all come up with the warning screen saying which engine and what type of error is there and I can turn it off from each device/MFD. 
For example the one below is low oil pressure warning:

`     SetN2kEngineDynamicParam(N2kMsg, EngineInst, OilPressureB, EngineOilTempB, CoolantTempB, AltVoltB, N2kDoubleNA, 
                               EngineHoursB, SeawaterPressureB, N2kDoubleNA, N2kInt8NA, N2kInt8NA, 0, 0, 1);
`

and the following is coolant flow warning:

`      SetN2kEngineDynamicParam(N2kMsg, EngineInst, OilPressureB, EngineOilTempB, CoolantTempB, AltVoltB, N2kDoubleNA, 
                               EngineHoursB, SeawaterPressureB, N2kDoubleNA, N2kInt8NA, N2kInt8NA, 0,0,0,0,0,0,0,1);`

maybe rough, but works :-)

Now the problem I have is that I cannot figure out how to parse that!
there are 24 bool options btw, how do I reach them when analysing the PGN??
I need overheat, low oil press, coolant flow and general error.

ideas?

cheers

V.

ttlappalainen, 2021-01-05:
I updated the library. Now you can do easily:
```
    tN2kEngineDiscreteStatus1 Status1;
    Status1.Bits.LowOilLevel=1;
    Status1.Bits.OverTemperature=1;
    SetN2kEngineDynamicParam(N2kMsg,0,656000,CToKelvin(86.3),CToKelvin(82.1),14.21,5.67,hToSeconds(2137.55),N2kDoubleNA,N2kDoubleNA,N2kInt8NA,N2kInt8NA,Status1);
```

and

```
   ...
  tN2kEngineDiscreteStatus1 Status1;
  tN2kEngineDiscreteStatus2 Status2;
  if (ParseN2kPGN127489(N2kMsg,EngineInstance,EngineOilPress,EngineOilTemp,EngineCoolantTemp,
                                 AltenatorVoltage,FuelRate,EngineHours,
                                 EngineCoolantPress,EngineFuelPress,
                                 EngineLoad,EngineTorque,Status1,Status2) ) {
...
   if ( Status1.Bits.OverTemperature ) ...

```
I forgot alias for ParseN2kPGN127489

virtuvas, 2021-01-05:
thanks Timo,

there are 3 sets of bools though! best if you do the last one as well!

V.

virtuvas, 2021-01-05:
and I guess means that lib will fail now unless you add the parameters in both Setting and Parsing, right?

ttlappalainen, 2021-01-05:
No, there are compatible alias also.

closed 2021-01-23 by virtuvas