ttlappalainen/NMEA2000 #200: 129540 - GNSS Sats in View

PeerD, 2021-01-09:
Hey

First and foremost, thank you so much for this awesome library!

I wanted to ask if it would be possible to support the 129540 - "GNSS Sats in View" NMEA 2000 Packet. Or is it still limited by #103?

I thought about building the code to parse the message myself (to safe you some time) but I wasn't able to find a reference that told me what values (what value types to be precise) to expect.

Thanks!
Peer

ttlappalainen, 2021-01-11:
I can add it, but for now it is still limited because of normal fastpacket size.

PeerD, 2021-01-12:
That would be great! And the limit is fine.

I also found some other missing parsers (all AIS related). Do you want me to just post them here or open a new issue?

MrClowny666, 2021-01-21:
Please post a new issue, ours might be overlapping

closed 2021-02-21 by ttlappalainen