ttlappalainen/NMEA2000 #201: Alert implementation

dmarc1234, 2021-01-18:
Hello

I am working on a project that needs to use the NMEA2000 Alert system on my Raymarine plotter, PGN's 126983 ~ 126988, but I cannot seem find any functionality that process these PGN's in your implementation. Is this alert system supported and if not is it something you would consider adding?

David

ttlappalainen, 2021-01-18:
The base library supports message transporting and so also alert messages. But there is no message creation/parsing functions, which you probably mean. Also alert handling is a small library itself. It is not just like reading or sending RPM information.

Adding support and whole logic for alert messaging is several hours work, so I'll not do it for free.

dmarc1234, 2021-01-18:
Hi Timo

I understand and I would be interested in paying to have this work done as I think it is a good feature for everyone to share. What would be the costs ?

David

ttlappalainen, 2021-01-19:
If you are making this as private and do not drive with Ferrari, forget it. I do not want to do halfway solution. Full support for all alert PGNs would also require full support of 126208 PGN for them. So we would talk about 3k-5k € project.

dmarc1234, 2021-01-19:
Its only part of the group functionality that is required for the alert implementation. I have attached a spreadsheet supplied by Raymarine that outlines the alert process I am looking for. As indicated I am interested in paying for this to be implemented but my budget would not extend as far as 3~5K Euro. Let me know what can be done ?

David
[Raymarine alert implementation.xlsx](https://github.com/ttlappalainen/NMEA2000/files/5834872/Raymarine.alert.implementation.xlsx)

cbitterfield, 2021-11-23:
I am working on a similar project for garmin chartplotters. When I get the hardware all setup, I am going to take a look at this as well

ttlappalainen, 2021-11-23:
Are you familiar with c++ and classes? In library I wrote N2kMessages with simple functional way, which is easier for beginners and works with messages having few parameters. But e.g. with engine dynamic, own class would be much easier to use. Or structure as parameter instead of single values. The class or structure can have default values so that it it would be enough to set/read only one value.

I use library in different way. I have message description database so that I can just take any PGN and set/read any field on it.

dpa11111, 2022-01-11:
This question is related to this thread, but somewhat different.
I am implementing a small embedded device for NMEA2000 bus, that will have a sensor that needs to report an ALERT to the Garmin MFD like GPS MAP 8212 etc. When this MFD gets an alert, it opens a window at the bottom of display, and has an acknowledge pushbutton. It also sounds a tone.
So I am trying to find more detail on the use of ALERT PGN 126983 thru 126988, and messaging support using 126208.
Any details on how this behavior, or communications and procedural protocol steps would proceed will be appreciated.

ttlappalainen, 2022-01-20:
Sorry for delay with this.

I should implement alerts to my system too and then will add them to library. But can not say for sure is it ready on end of feb.

BitterAndReal, 2022-10-23:
Is there any progress with the alert implementation?

cbitterfield, 2022-10-23:
Not yet

virtuvas, 2023-06-02:
hi all,

finishing a v.crude bilge monitoring system and would like to get some alerts for it as well, any news/updates on that?
ATM I'll do a custom sentence for the three bilge pumps and pass the data to signalk (node red even) and get some alarm there (which is a convoluted way of doing things)

thanks

V

ttlappalainen, 2023-06-04:
No progress on that side.

ttlappalainen, 2023-08-28:
I am implementing alerts to Simulator. Do you have some device supporting alert message 126983 and response 126984? I did not found support at least from Simrad NSSEvo3. If you some hw for Simulator, you can test does your device response anything to those messages.

jrogby, 2023-10-04:
> I am implementing alerts to Simulator. Do you have some device supporting alert message 126983 and response 126984? I did not found support at least from Simrad NSSEvo3. If you some hw for Simulator, you can test does your device response anything to those messages.

Hi!
I will do some tests with Garmin 74xx/76xx and 84xx/86xx. The manual state that the PGNs are supported.. :)

ttlappalainen, 2023-10-05:
Do you have some source sending alerts? If you have some hw for Simulator, you can currently use Tools-NMEA2000 Commands-Any message and send e.g., message below. That should popup (first data row) technical (1) warning (5) of propulsion gas turbine plant system (08) sub-system (06) vacuum at compressor inlet -high (ID=0003)

Command support is combination of 3 bits silence, acknowledge, escalation. Status is combination of those status bits.

You can also test below by using two Simulators with null modem cable (true or virtual).

I thought to add way to define alerts for sources on Simulator, but did not continue it due to rare support for alert messaging.

// Alert
PGN:126983, Source:100,
15,08,06,03,00,                // type (1) category (5) system (08) sub-system (06) ID (0003 remember little endian)
FE,FF,9F,8E,00,91,78,C0,   // sender NAME
00,01,                               // source instance (00), source index (01)
07,38,                               // occurance number (07), command support (3), status (0 << 3), reserved (3 << 6)
FF,BB,FF,AA,FF,00,FF,FF,   // acknowledge source NAME
11,                                   // trigger (1 = auto), treshold status (1 << 4 = treshold exceeded)
00,                                   // priority (00 = highest)
02                                    // state (02 = active)

frank7171, 2025-10-31:
It looks like N2kAlerts.h was never completed, nor added to NMEA2000.h?

ttlappalainen, 2025-11-01:
No. I have not found unprotected documentation of alert handling.