ttlappalainen/NMEA2000 #203: Example MessagSender not working (DCStatus parameter missing) and Strange valuses for Rudder and Variation

GWS65, 2021-02-02:
Hello Timo,

First of all thank you very much for all the hard work you did on the NMEA2000 library! 

I coupled 2 ESP32 devices to the bus with different IDs and 'was trying tot run the MessageSender example. 
First issue was that I got a compile error on line 165 
SetN2kDCStatus(N2kMsg,1,1,N2kDCt_Alternator,86,91,1420,0.215);
Apparently there is a parameter missing. Just a minor thing but just to let you know.

Another issue is much more complex so it seems. I sent the data to the bus from some RayMarine Wind and Pilot devices and your MessageSender on another ESP32. There si no GPS connected to the bus. I do get most of the data as expected except for Rudder and Variation. it sometimes looks valid but most of the time the value that I pick up via JSON (boatdata.h) is 

RudderPosition : -57295780000
Variation : -57295780000

If I print the value to the Output Stream it shows -100000000L and or  ovf (overflow) 
but every now and than when I refresh the screen it shows the expected data. (maybe 1 out of 100 refreshes).

For Rudder I could think of my pilot that does not have a ruddersensor connected messing something up with two devices that could send that data. Variation should come from the HDG message and heading lat/lon show ok always but not variation.

If you have any idea on the cause of this I would very much appreciate you sharing it.

Thanks,
GWS65

ttlappalainen, 2021-02-02:
I defined -100000000L as "not available" values. When you read data you should always use function:
If ( !N2kIsNA(val) ) { // values is ok
That function has been defined for all types.
On some coming version you have to use N2kIsNotValid, since there is also "overflow" definition on NMEA 2000.

Anyway it means that on that specific field data has been marked as "not available". It can be same with any field on messages. E.g. Some device may send Engine dynamic data, but only oil pressure and oil temperature, whil all other fiedls are  "not available"

GWS65, 2021-02-05:
Hello Timo,

I think I am closer to the cause of my issue but don't fully understand yet
how to solve.

I have two ESP32 devices with can bus adapter connected to a RayMarine
SeatalkNG network. One is sending the simulation messages from example
MesageSender the other is using DataDisplay2.ino to receive. I only changed
the error on line 165 in MessageSender.ino as explained before, missing
parameter in some DC message function. (commented it out with line 166).

The Raymarine network consists of:
EVO 100 sensor and controller for the Wheel drive motor
   The EV-100 sensor also gives heading and attitude (Yaw Pitch & Roll)
messages
p70 control panel
IC-5 connect box with Wind sensor (speed / direction)
i70 display

You are using the statement if (!N2kIsNA(val)) { some routine }  in a
template function. First had to learn more about this to understand the
"trick"  of template routine's but I think I got it sort of. At least in
DataDisplay2 most of the output is using it.
Therefore I assumed that this was working OK.

It looks like the rudders in the network (Rudder: 252 and  Rudder: 1
(MessageSender) and I did see Rudder: 0 or Not available ) are fighting for
the right to be heard. Most of the time :252 wins (252 could that be
RayMarine?) or it is the simulator in MessageSender.
How to find out what number belongs to which source except switching it
off/on ? Is there a standard number per sender and what will happen when
two devices claim same ID or receiver is not considering ID?

When I restart the ESP32 I do see rudder 1 for a while but after some time
(up to half an hour) Rudder 1 does not seem to be send from the simulator
or not received but I do see messages from Rudder 0 and 252 both with no
position or direction order. Is there a limit on the number of similar
devices on the NMEA2k bus?

The reason the issue came up is actually because I was using the receiver
from AK-Homberger to send JSON at first (
https://github.com/AK-Homberger/NMEA2000WifiGateway-with-ESP32/blob/master/Wifi-Serial-AP-AIS-CAN/Wifi-Serial-AP-AIS-CAN.ino
)
Looks like this one does not take care of some necessary checks like you
described. Might work if you have a ListenOnlyMode but in my case I was
building my own Rudder sensor to send a value to the bus and wanted to send
Rudder data. Do you think that code needs to be enhanced and use the
N2kIsNA(val) function too?

Thanks,
GWS65