ttlappalainen/NMEA2000 #204: Missing AIS & DSC PGNs

PeerD, 2021-02-22:
Hi

Thank you so much for this library and the support of it. I found some more official PGNs I'd like to use but where the parsers are missing from this library. It would be awesome if they could be integrated aswell.

Also, is there some way to support you? Like the possibility to send you a donation? I'd really like to express my gratitude for your awesome work here!

In my eye important messages (aka the ones I'd like to use):
- 129040 AIS Class B Extended Position Report
- 129041 AIS Aids to Navigation (AtoN) Report
- 129794 AIS Class A Static and Voyage Related Data
- 129808 DSC Call Information

Messages I'd like to use but are not really that important:
- 129793 AIS UTC and Date Report (Base Station)
- 129799 Radio Frequency/Mode/Power

Just for sake of completeness, the following messages are not supported aswell (but I have no personal interest in them): 
- 129797 AIS Binary Broadcast Message
- 129798 AIS SAR Aircraft Position Report
- 129801 AIS Addressed Safety Related Message
- 129802 AIS Safety Related Broadcast Message
- 129809 AIS Class B “CS” Static Data Report, Part A
- 129810 AIS Class B “CS” Static Data Report, Part B

ttlappalainen, 2021-02-28:
For those of you interest they would require about 2-4 h work/PGN execpt 129808, which seem to be more complex.

closed 2021-04-16 by ttlappalainen

pk1965, 2021-05-19:
Hi Tim,
I've been offline for about 6 months doing other stuff.  I've started to develop an hybrid AIS transmitter for education based on your library and a RPI.  Stage one is getting the code to work over NMEA2000 to the software program expedition which i've kind of completed..  well more like hacked

I noticed that there was some PGN's missing and a quick search I found this issue ticket.  It appears since the ticket was originally raised you had added some of the PGN's. ie.. 129794, 129809,129810.

Would it be possible to have a relook at the others in this list specifically the ones that deal with messaging and SAR. ie,

- 129798 AIS SAR Aircraft Position Report
- 129801 AIS Addressed Safety Related Message
- 129802 AIS Safety Related Broadcast Message

What I'm trying to do demonstrate how systems react when these messages come in.