ttlappalainen/NMEA2000 #205: actisense bi-directional?

gregsyoung, 2021-02-28:
i realise this is not really an issue, but didnt know how to raise the question..

 i built a teensy 3.x based device 4 years ago, using the instructions and sw at the time (Teensy with ActisenListenerSender) all has worked fine - but i never had need make a change on n2k using it (only for listening)
i(i also at the time built the second port using the optional hw. .. tested it ok, but never used it .. )

Am i correct that the Actinsense lisenersender will only allow thw actisense to recieve? (and the second usb port can be used for sending.. albeit  not from the actisense software?)
pwrhaps i have the wrong? version of software loaded in teensy to allow it to transmit from actisense?

during the intervening time i note a lot of updated sw on your github, buut uncertain if any of these will help in my task above?

q; is there a version of the firmware for teensy that will allow it to work with actisense and BOTH recieve and transmit onto n2k bus? 

thanks in anticipation of any guidance

ttlappalainen, 2021-02-28:
If you use my NMEA Simulator, it should work for both direction.

gregsyoung, 2021-02-28:
thanks for response, but not sure if i correctly understand..

Im trying to use the actisense reader sw - to modify some data instances of devices on the n2k bus.

when i try to make a change in actisense sw.. nothing happens ... and i suspected this was because if was’operating in a “recieve only” mode (with the actisenselistenersender firmware in the teensy) 

if i understand ... your “nmea simulator” can  transmit PGns onto the bus?  (via the second usb port on my teensy hw with pololu module) 

does it have capability to make a data instance change to a PGN of a device on the bus?

ttlappalainen, 2021-02-28:
If you have on actisenselistenersender as current default both read and forward stream pointing to Serial, NMEA Simulator can read and send data to the bus. I just tested that it works also with Actisense Reader and I could change e.g. "Installation description 1" string on device. I just had to turn devie on, while Reader was active so that device notifies itself to network with ISO address claim. Then I selected from Reader "Network View" device, wrote text and pressed enter and setting was changed.

Also NMEA Simulator can change settings or do nearly anything with bus. There is Tools-"NMEA 2000 Commands", which can be used to send anything to the bus. Unfortunately you need to know rather much about NMEA 2000 to use it efficiently. On "Any message" tab you can build you message in hex and so test anything. Changing device/system instance can be done also with "Group function"-"Command" tab. If you want to change "lower device instance" for some device, you set 
"PGN to command" 60928
"Destination" <your device address>
"Priority setting" 6
"Pair" Field=3, Value=<new lower device instance>, Multiplier=1, Bytes=1
On all other pairs Field=0
Then press "Send command".

Or naturally you can set upper and lower instance value with one command by setting to next pair Field=4 and others as on first pair.

I have on plans to make common settings easier. Unfortunately I do not have time for that, since I am trying to find ways to spend all those donations (about 200 €) I have got for NMEA Simulator.

gregsyoung, 2021-03-01:
timo, many thanks for detailed  response, this gave me confidence to find the problem.
I ended up reprogramming the teensy with the most recent “actisenselistenersender.ino”

and bingo! it all worked exactly as you describe. 
im now suspicious that perhaps i had only loaded actisenselistener (way back 4 years ago... and didnt keep a proper record). 

issue closed, with my thanks for your assistance, and ongoing great support  of the community

closed 2021-03-01 by gregsyoung