ttlappalainen/NMEA2000 #206: PGN130312 SID & TempInstance Q

virtuvas, 2021-03-02:
Timo,

quick questions

I'm generally using 1 for SID, not sure if I should change it for something, is that right?
I'm asking as I want to record temps from two different fridges and OK, I'll get the TempSource right, but how do I differentiate between the galley and salon fridge? 
I'm thinking of changing their respective TempInstance numbers (0 and 1 or 1 and 2) and use that on my custom display or signalK output. Is that OK? is there another "proper" way of doing this?

thanks

V

ttlappalainen, 2021-03-03:
SID is meant to glue data to same measurement. So if your device e.g. sends boat speed and wind data and they will be measured at same time, you can use same SID for each message send for measured at same time. Then you increment it for next measurement and roll to 0 after 252. I think it is not very important that you fridge temperatures are measured at same time so it is best to use just NA = 0xff instead of any value.

Yes, you separate different fridge temperatures with instance. Note that you may need to even go higher instances. Some MFD:s does not accept same instances for same PGN:s even they come from different device. In principle devices should separate devices with NAME (data on 60928) so that different devices can use same instance.

virtuvas, 2021-03-03:
thanks Timo,

learn something new every day - should stop asking!
OK, I've just checked and I'm only sending the following PGNs that include SID:
PGN130306
PGN130310
PGN130311
PGN130312
and I've been sending 0 or 1 all along for all of them. Since it really makes no difference if they were taken at the same time I'll change that to 0xff

Now, you do mention PGN60928, do we really need to send that as well? 
I don't recall anything in your work mentioning it, am I missing something?
V.

ttlappalainen, 2021-03-03:
60928 - ISO Address claim is automatic and handled by library. Normally you do not need to think about it.

Hint! If you make a device, which reads data from bus, then you should use device NAME to lock message you are reading. Due to address claiming device source (aka address) may change and you have to follow those changes. There is N2kDeviceList module, which takes care of reading devices on your bus and checking possible address changes and that makes your life much easier. Naturally on your own boat there is so few device additions that address claiming happens very rarely and so using just source may work for your lifetime.

Just for note. 0xff is right for SID, if you do not use it. Anyway who knows, if there would be MFD or something, which does not accept data if SID=0xff. Should not be, but never know.

virtuvas, 2021-03-06:
well,

works fine onboard, GMIs happily let you select which TempInstance you want and show them properly, example of the call:
`
SetN2kTemperature(N2kMsg, 0xff, 150, N2kts_RefridgerationTemperature, CToKelvin(6))
`
`
SetN2kTemperature(N2kMsg, 0xff, 151, N2kts_RefridgerationTemperature, CToKelvin(3))
`

Problem now is that the rpi with signalK picking up PGNs and sending them to influxdb messes them as one with either sending 3 or 6C...
will try also (tomorrow) with different values (may actually use again 150 and 150 instead of 0xff) and see what happens and report.

cheers

V.

ttlappalainen, 2021-03-07:
Maybe SignalK does not handle SID right. If you provide 0xff, it means that it is not available and each message will have separate timing for data.

If you need to provide something your can simply increment SID for each message or if you get temp at same time, you send same SID for each masurement. If you e.g. have 1-wire sensors and you send  one "start conversion" message to 1-wire bus, they will be measured at same time. Then for those temperatures you could use same SID. But then handling get bit more complex. You should send temperature messages periodically 2000 ms independ of your temperature reading. So if you start reading temperatures for one time, it may happen that you have read temperature for sensor one, but then comes time to send messages and you send them before you have read sensor 2 temp. Then those can not have same SID. This only means that you should buffer readings and move them for ready to send with same SID. And even more complex, your device sends other messages having SID, which are not syncroniced with temperatures. You must use single incrementing SID per device - or 0xff

virtuvas, 2021-03-10:
well,

for the time being and since signalk wont be able to differentiate  N2kts_RefridgerationTemperature values from different tempinstances, I just changed the one to N2kts_FreezerTemperature and it works fine ;-)

will now have a go at dealing with relay states and sending messages to turn on and off loads. See how that goes. I've got a Maretron DSM250 for testing the messages.

cheers

V.

closed 2021-03-10 by virtuvas