ttlappalainen/NMEA2000 #207: Multiple CAN ports

ba05, 2021-03-03:
Can I run two separate instances of the library on a Teensy 3.6 or similar - one instance on each CAN port?

ttlappalainen, 2021-03-04:
Yes. If you use Teensy you simply create an other instance of tNMEA2000 object:
```
...
#include <NMEA2000_CAN.h>
...
tNMEA2000 &NMEA2000_2=*(new tNMEA2000_teensy(4,1)); // 4 is timeout not used on newest library
...
```
If you use e.g. ESP32 and second can is build with MCP2515 (mcp_can) then you do:
```
...
#include <NMEA2000_CAN.h>
#include <NMEA2000_mcp.h>
...
#define N2k_SPI_CS_PIN 53
#define N2k_CAN_INT_PIN 21
tNMEA2000 &NMEA2000_2=*(new tNMEA2000_mcp(N2k_SPI_CS_PIN,MCP_16MHz,N2k_CAN_INT_PIN));
...
```
Naturally you need to do setup for both tNMEA2000 objects and also call ParseMessages for both objects in loop.

ba05, 2021-03-04:
Thanks! Great library!

closed 2021-03-04 by ba05