ttlappalainen/NMEA2000 #208: Fast Packets Not Working

mikeh9, 2021-03-10:
NMEA2000.cpp -> inside function GetSequenceCounter(unsigned long PGN, int iDev)

There are some code mistakes:
if ( Devices[iDev].PGNSequenceCounters==0) { 

you are comparing a pointer address to 0, did you mean Devices[iDev].PGNSequenceCounters[0]==0?

This causes my AT90CAN128 to hang every time product information is sent via fast packets.

The same code mistake is repeated 5 lines down:
if ( Devices[iDev].PGNSequenceCounters==0 ) return 0;

did you mean if ( Devices[iDev].PGNSequenceCounters[0]==0 ) return 0;

ttlappalainen, 2021-03-10:
No. It is dynamic vector. On NMEA2000.h line 298 on constructor device PGNSequenceCounters will be initialized to 0. On line you point, it will be tested has it been created. If not, two lines below there is allocation:
Devices[iDev].PGNSequenceCounters=new unsigned long[Devices[iDev].MaxPGNSequenceCounters];
and then initialization.
It will be tested again, that it has been created.

I expect that problem is with the small device you try to use. Arduino Mega is minimum, which runs without reducing memory by disabling features. So when it starts to send product information, library creates those sequence counters and tries to allocate memory, which you do not have.

mikeh9, 2021-03-12:
Thanks! You were correct. My device only has 4kb of ram. The line that actually crashed was:
Devices[iDev].PGNSequenceCounters=new unsigned long[Devices[iDev].MaxPGNSequenceCounters];

which was trying to allocate more ram.

I really appreciate your help.

closed 2021-03-12 by mikeh9