ttlappalainen/NMEA2000 #211: PGN127489 Engine fault list too short ?

klemenspehar, 2021-04-13:
Hi,

firstly I appreciate so much your NMEA libraries. I was quickly figuring all out, so I can manipulate messages with ease. 
Nevertheless I'm working on transmitting Engine Dynamic parameters (PGN127489) and I have faced flood of unknown faults on my Raymarine Axiom MFD (OK, actually just 8 of them). After investigating raw CAN data, I found, that 4th CAN frame of PGN127489 message always sends FF on Byte position 4.
Even earlier during coding I noted, you defined only 24 error flags in N2kMessages.h file. I was not digging further, but somehow the code is pushing FF to last 8 bits. I tried to pull the whole Status2 to zero value (actually I don't need no flag from Status2), but it doesn't work.

My opinion is, all of these statuses (if not specifically set) should be zero by default.

I have modified N2kMessages.h as follows:

I added 8 bool flags into: inline void SetN2kPGN127489(...  and inline void SetN2kEngineDynamicParam(...
bool flagBlind0=false, bool flagBlind1=false,  ... ,  bool flagBlind7=false)

and added bits into Status2:
Status2.Bits.Blind0 = flagBlind0;
Status2.Bits.Blind1 = flagBlind1;
...
Status2.Bits.Blind7 = flagBlind7;

The flags should be added inot file NMEA2000StdTypes.h under structure union tN2kDD223 as follows:
uint16_t Blind0 : 1;
uint16_t Blind1 : 1;
...
uint16_t Blind7 : 1;

I'm not sure this by the NMEA book, but it works.

Thanx.

ttlappalainen, 2021-04-14:
Please check that you have latest version having line 146   tN2kDD223(uint16_t _Status=0): Status(_Status) {};
That sets all flags to 0 as default.

klemenspehar, 2021-04-14:
My line 146 contains:
tN2kDD223(uint16_t _Status=0): Status(_Status & 0x00ff) {};

I checked your Git:
https://github.com/ttlappalainen/NMEA2000/blob/master/src/NMEA2000StdTypes.h
and seems the same.

ttlappalainen, 2021-04-14:
Ah, I have not updated the newest version with Manufacturer specific bits. But that version should anyvay reset high unspecified bits.