ttlappalainen/NMEA2000 #212: Several NMEA2000_CAN.h issues

lsoltero, 2021-04-15:
1. tNMEA2000 &NMEA2000=*(new tNMEA2000_SocketCAN(SOCKET_CAN_PORT));

really should be defined as
tNMEA2000_SocketCAN &NMEA2000=*(new tNMEA2000_SocketCAN(SOCKET_CAN_PORT));

otherwise, you have issues when you define additional tNMEA2000_SocketCAN objects and then do a
NMEA2000 = the_new_NMEA200 object.  since the types don't match.   The compiler will let this through but you get a run time EXECV under Linux.

2. NMEA2000.h as written can only in included in one cpp file in a project.  Since it instantiates a global variable yoou get multiple instances of this and the compile complains of collisions...  a better approach would be 

#elif USE_N2K_CAN == USE_N2K_SOCKET_CAN
// Use socketCAN devicethus
#ifndef __ USE_N2K_SOCKET_CAN_DEFINED
#ifndef SOCKET_CAN_PORT
  #define SOCKET_CAN_PORT NULL
  #endif
#include <NMEA2000_SocketCAN.h>       // https://github.com/thomasonw/NMEA2000_socketCAN
tNMEA2000_SocketCAN  &NMEA2000 =*(new tNMEA2000_SocketCAN(SOCKET_CAN_PORT));
tSocketStream serStream;
#define __USE_N2K_SOCKET_CAN_DEFINED
`#else `
extern tNMEA2000_SocketCAN &NMEA2000
`#endif`

or better yet... let the user define the object in his code.

items 1 and 2 apply to all other CAN types as well but since I am only working on Linux I just illustrate the issues for this case.

ttlappalainen, 2021-04-16:
1. No. With current definition you can use:
`extern tNMEA2000 &NMEA2000;`
In other modules. If you use driver specific definition, you have to do that also on all other modules.
Change of this would cause version incompatibility and will not be done.
If you define additional object you use:
`tNMEA2000_SocketCAN &NMEA2000_2=*(new tNMEA2000_SocketCAN(SOCKET_CAN_PORT));`
or
`tNMEA2000_SocketCAN NMEA2000_2(SOCKET_CAN_PORT);`

2. It could have ifdef, but still definition has to be:
 `extern tNMEA2000 &NMEA2000;`

Note that NMEA2000_CAN.h is designed for easy start for beginners. I also use it in test programs, which compiles on different boards. In projects designed only for specific boards I do not include NMEA2000_CAN.h at all. Instead I use e.g.
```
#include <NMEA2000.h>
#include <NMEA2000_esp32.h>       // https://github.com/ttlappalainen/NMEA2000_esp32
```
and define object by my own. But mostly I anyway use  `extern tNMEA2000 &NMEA2000;`, since then different modules can be used on different projects.

lsoltero, 2021-04-16:
Thank you for your comments.

I agree that not using NMEA2000_CAN.h and doing the includes directly of NMEA2000.h and NMEA2000_SocketCAN.h (in my case) is much better since then you don't have a global object allocated in an include file which is usually considered bad practice.

Having said that, however... the file is still there an users are being encouraged to use it in the instructions and examples.  

The following code causes a CORE dump with a segmentation error on my ubuntu development machine.
#define SOCKET_CAN_PORT (char *)"can1"
#define USE_N2K_CAN USE_N2K_SOCKET_CAN  // for use with socketCAN (linux, etc) systems
#include <NMEA2000_CAN.h>  // must only be defined in one place since defines globalVariable NMEA2000 and serStream;

char *candev = "can0"
main()
{
     tNMEA2000_SocketCAN *t = new tNMEA2000_SocketCAN(candev);
    NMEA2000=*(t);

when using gdb to debug the segmentation occurs in the NMEA2000 = operator during the assignment.  printing out the variables you see that the two classes are NOT the same... NMEA2000 is of type tNMEA2000 and the other is of type tNMEA2000_SocketCAN which includes some membery variables not included in the the base clase... Specifically 

    int   skt;
    char*  _CANport;

given that tNMEA2000_SocketCAN inheits from tNMEA2000 there is really no problem declaring NMEA2000 as type tNMEA2000_SocketCAN.  That is done inside the NMEA2000_SocketCAN.h and is totally consistent.  
extern NMEA2000 ... should also be declared there since this include file allocates the object.  I think that we agree that this is bad stile anyway... but just tyring to make things work for that first time user inadvertently including this file.

Take care.

closed 2021-04-16 by ttlappalainen